ratgdo/esphome-ratgdo #158: Light, motion and lock remotes status not updating.

jschafermo, 2023-12-25:
I received two Ratgdo boards and installed them on my two Chamberlain openers a couple days ago.  All hardware and wiring are identical on both openers.  One Ratgdo works perfectly with instant status updates, the other one does not.  My problem Ratgdo will update the light status after approximately five minutes which I believe the opener sends updates at this interval.  I can toggle the light from the web interface or Esphome and the light will toggle instantly with a status update.  If I toggle the light from the wall controller, the light will toggle, but the status will not change for a few minutes.  The light status updates instantly when the door is opened or closed.  No status update happens with motion detection and lock remotes while using the wall controller.  It appears anything triggered by the wall controller is not being captured by the Ratgdo board.  Looking at the logs, events are logged for everything except when triggered by the wall controller.  I log the following every now and then:

```
10:11:42 | [W] | [component:214] | Component preferences took a long time for an operation (0.06 s).
10:11:42 | [W] | [component:215] | Components should block for at most 20-30ms.
```

As a quick test, I swapped Ratgdo boards on the openers.  The problem transferred with the board swap.  Currently, both Ratgdo boards are wired using red, white and black connections and powered by USB.  Not using passthrough or dry contacts.  

Tested obstruction detection, status will update instantly on both boards.  While testing the obstruction sensor, I discovered if an obstruction was detected, the light status would update when using the wall controller.  The status update takes a couple seconds, not instantly as the working Ratgdo.  

I re-flashed the Ratgdo board with the problem, but this didn't resolve my issue.  

Not knowing how light, motion and lock status change works, not sure what to try next.

Any suggestion would be appreciated.

Thanks

mulcmu, 2023-12-26:
The component blocking warning isn't an issue.  What happens if you leave the ratgdo hardware installed but just swap the ESP board between setups?

jschafermo, 2023-12-26:
> The component blocking warning isn't an issue. What happens if you leave the ratgdo hardware installed but just swap the ESP board between setups?

My version of Ratgdo have ESP boards which can't be removed.  I can only swap the entire Ratgdo and ESP.

Thanks

mulcmu, 2023-12-26:
I didn't realize rev 2.5 had ESP integrated, that would make it a bit difficult to swap.  Based on the logs from [here](https://community.home-assistant.io/t/ratgdo-light-control/660780/2) it looks to me like the garage door is pulling the red line to ground low enough to trigger a logic low on the board with the issue, however the wall control transmissions are not pulling the red line down low enough. 

I'd visually inspect the board to see if there is any crust or poor solder joints that might be the culprit.  It seems to me you've isolated the problem down to a hardware problem on the ratgdo board.

jschafermo, 2023-12-26:
> I didn't realize rev 2.5 had ESP integrated, that would make it a bit difficult to swap. Based on the logs from [here](https://community.home-assistant.io/t/ratgdo-light-control/660780/2) it looks [...]

I will check the ratgdo closer to see if I spot any problems.  From everything I've been able to do, I believe I have a defective ratdgo board.  I wanted to check to see if there was anything else I could try.  My other ratgdo board works great on both openers.

I appreciate your help!

Thanks

mariusmuja, 2023-12-26:
Yeah, sounds like the data line is not pulled low enough to reliably triggering the mosfet on the board that's not working. 

A mosfet threshold voltage decreases a bit with the rise in temperature, so one thing you could try is to get the temperature of the board that's not working slightly higher (maybe place it near the garage light bulb and use an incandescent bulb).  If it starts working fine it would confirm this hypothesis.

jschafermo, 2023-12-26:
> Yeah, sounds like the data line is not pulled low enough to reliably triggering the mosfet on the board that's not working.
>
> A mosfet threshold voltage decreases a bit with the rise in temperature, [...]

I will try warming the board to see if it makes a difference.

Thanks!

jschafermo, 2023-12-27:
So far, warming of the board hasn't helped.  I will try a few more times to see if the Ratgdo starts responding correctly.

Thanks again for your help!

MrSt1k, 2024-01-04:
I too have what started as one device with the exact same behavior. One of three devices (also a version 2.5) installed on three identical garage doors would not report door opening or closing position. Tripping the obstruction sensor would trigger the reporting of the door to the correct open/closed position. The other two work perfectly.

After reflashing the device and checking the wiring multiple times, I concluded it must be a problem with the Ratgdo board and ordered another. Installing the second device (version 2.5i) yielded the same results. I have run out of ideas and will be watching this discussion hoping that a solution is foud.

jschafermo, 2024-01-05:
> I too have what started as one device with the exact same behavior. One of three devices (also a version 2.5) installed on three identical garage doors would not report door opening or closing [...]

Have you swapped a working Ratgdo board to the non-working opener?  When I swap Ratgdo boards between my two openers, the problem follows the problem Ratgdo board. 

I will try a new Ratgdo board soon.

MrSt1k, 2024-01-05:
I have tried a new, unused board and, while it was not a known good board, the problem seemed to stay with the opener. I will try to swap it with a good board/opener tomorrow and report back…

MrSt1k, 2024-01-06:
After swapping the boards on a working opener with a non-working opener, I can confirm the problem with mine is with the opener.

It is a mystery what the problem is though. It worked briefly after changing out CFL bulbs and I thought maybe they were causing interference, but the problem returned. I swapped out the wall controls without any luck. I connected the wall control with new wires without any luck.

It has mysteriously started working again, so I'm not messing with it anymore. The only other possibilities seem to be the safety sensors or their wires, or the board in the opener itself.

jschafermo, 2024-01-08:
An update to my problem.  I received my replacement Ratgdo board and installed it on the problem opener.  Happy to report my problem is resolved and the sensors on both doors update immediately and working correctly.

Thanks for the help!

ue-cthompson, 2024-01-16:
I am also having this issue.  I have the latest board, 2.53.  I guess the only way to find the issue is to buy another board?

hellboyfishman, 2024-01-19:
Somehow my other account was marked as spam, so I'm reposting this, hopefully you can all see it:

> I am also having this issue. I have the latest board, 2.53. I guess the only way to find the issue is to buy another board?

I can say, from my experience, don't buy another board. I just spent time installing a replacement (this one is a 2.53i vs the 2.5i I had) and it is the same result. I'm starting to suspect an issue with the GDO itself not reporting. Light and door status aren't updating, but motion and obstruction are reporting. I can toggle the door but the status just won't update, I'm not getting the 5 min poll events either.

I guess at least this one always says the door is closed vs the 2.5i that said it had been open for the past week.

Also I get this message every time I reboot the device:

    Failed to communicate with garage opener on startup; Check the ratgdo main Rolling code counter number entity history and set the entity to one number larger than the largest value in history. ESPHome devices

Quick edit, after cycling the door it now thinks it is open again. I give up, unless anyone else has ideas on where to go from here. At this point I'm wondering if the battery backup in the GDO is causing any issues. I'm tempted to just disconnect it for a while to see.

mariusmuja, 2024-01-19:
There have been several people reporting issues on GDOs with battery backup (DC motor). I suspect the first models of the logic boards in these GDOs have an issue where when transmitting it doesn't fully bring the data line to ground and ratgdo cannot receive the messages transmitted. There's also a case where the wall button was not receiving messages from the GDO (was not pairing after a GDO restart).

hellboyfishman, 2024-01-19:
I was just reading through that really long thread about it.   I haven't yet seen the issue where, long term, the wall button wouldn't re-pair with the GDO.   It does seem like after plugging it all back in, it does take about 5 minutes before I can press the wall button and have it pair up properly.

I ordered a new wall button just in case, I don't feel like tearing my GDO apart in the cold garage.... yet.  I'm assuming this isn't something that can be solved with software since people are replacing boards?

MrSt1k, 2024-01-19:
My problem did return (almost immediately) and it is a dc motor door with battery backup. My other 2 doors are the exact same openers, but they have worked flawlessly. Has anyone suggested a fix or workaround for those problem openers?

hellboyfishman, 2024-01-19:
Do all GDOs with backup have a DC motor?   I need to get up there and find my model number and start digging.

mariusmuja, 2024-01-19:
> Do all GDOs with backup have a DC motor? I need to get up there and find my model number and start digging.

Yes, I'm pretty sure no GDOs with a 12V battery backup has a power inverter built in to power an AC motor.

hellboyfishman, 2024-01-19:
> Yes, I'm pretty sure no GDOs with a 12V battery backup has a power inverter built in to power an AC motor.

You're right, I found the model number and the replacement board listed is the 050DCTBFMC DC logic board.   I really want to order one but I know I'll snap the tabs installing it out in the cold 10F garage.   I'll wait to see if that other thread pans out when they get new boards.  

As a side note, I'm so glad I accidentally figured out my other github account is "marked as spam", probably for using a simple login alias, because I opened a really long issue here that I updated over several days and got 0 replies, but now I know that no one probably even saw it.   Thank you for all the work you guys have put into the ESPhome port of this.  My other GDO (AC) has worked like a champ since I installed the RATGDO, but that also makes this issue more frustrating.

mariusmuja, 2024-01-19:
Let's see what @NukeThemTillTheyGlow reports back here: https://github.com/ratgdo/esphome-ratgdo/issues/174

hellboyfishman, 2024-01-19:
I did turn the light on the GDO on for about 10 min starting at 15:55 and it started reporting status.  I'm about to head out, so I'm going to see if it reports door position when I leave, 

> 
> 16:08:24 | [D] | [ratgdo:210] | Status: door=CLOSED light=ON lock=UNLOCKED
> 16:08:24 | [D] | [cover:170] | 'Door' - Publishing:
> 16:08:24 | [D] | [cover:173] | Position: 0%
> 16:08:24 | [D] | [cover:186] | Current Operation: IDLE
> 16:08:24 | [D] | [switch:016] | 'Status door' Turning OFF.
> 16:08:25 | [D] | [lock:054] | 'Lock remotes': Sending state UNLOCKED
> 16:08:25 | [D] | [binary_sensor:036] | 'Motor': Sending state OFF
> 16:08:25 | [D] | [number:012] | 'Rolling code counter': Sending state 2124.000000
> 16:08:28 | [D] | [esp8266.preferences:238] | Saving preferences to flash...
> 16:13:32 | [D] | [ratgdo:210] | Status: door=CLOSED light=ON lock=UNLOCKED
> 16:18:37 | [D] | [ratgdo:210] | Status: door=CLOSED light=ON lock=UNLOCKED

mariusmuja, 2024-01-19:
> the replacement board listed is the 050DCTBFMC DC logic board.

That's correct. Before that it was 050DCTBMC and before that 050DCTB or 050DCTWF. So many revisions might indicate initial boards had some issue...

mariusmuja, 2024-01-19:
> I did turn the light on the GDO on for about 10 min starting at 15:55 and it started reporting status.

We did notice the problem to be more likely to manifest itself when very cold. Leaving the GDO light on for a while made it to start working most of the time.

jschafermo, 2024-01-19:
My problem has returned with the replacement Ratgdo board.  Motion detection and light status not updating once again, triggered from the wall controller.   The replacement worked great for a few days, but has stopped working.  My other opener, exact model, still working like a champ.  My openers have battery backups and it was turned cold.  Currently garage temperature is 42°, will see what happens when temperature warms up.

jschafermo, 2024-01-22:
Quick update on my problem Ratgdo/GDO.  Today, motion and light status is quickly reporting when triggered from the wall controller.   I'm watching the logs from the web GUI.  I've made no changes in wiring or firmware, only change being a few degrees warmer in the garage. Don't know if temperature is the cause, but noting temperature after reading another post.

jschafermo, 2024-01-23:
My Ratgdo stopped detecting motion and light being turned on and off from the wall controller.   Oddly, lock remotes from the wall controller is being captured.   I swapped the Ratgdo boards between my flawless gdo and problem gdo again, the problem follows the Ratgdo board.   The Ratgdo board has been replaced once, I keep searching for the cause of the problem.

tcouey, 2025-10-27:
I have a Security + 2.0 GDO as well and recently (a couple of weeks ago) my ratgdo32 stopped reporting motion events (which I had been using in automation in Home Assistant). I tried everything I could think of as far as power cycling the GDO and ratgdo. Today I finally had time and updated HASS, ESPHome, and the firmware on the ratgdo (which was about 6 months old). No change. The motion sensing still doesn't work. I live in a warm environment where the temperature in the garage is always in the range of 75F to 98F every day of the year. It's mostly in the mid to high 90s throughout the day and early evening and doesn't even drop into the mid 80s until the late evening. The motion sensor itself is fine, by the way, it still turns on the lights on the GDO just fine, the ratgdo just doesn't see or report it.

I'm wondering if the relatively warm temps (commercial spec parts are should still be okay at 100F) are causing failures on the PCB or maybe just comm issues with the GDO. Also the mainboard of the GDO is at least a decade old (it's branded as Craftsman), probably closer to 15 years now, I don't remember exactly when I put it up. I'm not sure if I can still get a replacement control board for the GDO at this point, but since others have tried that without success and the door opener works otherwise, I'm hesitant to go that route. I should also note that this is a battery backed unit, but it is currently battery-less as I got tired of replacing it every few years and lead acid batteries are expensive here.

I guess I just wanted to add my name to list of people having problems with sensors not being reported. At some point I might try looking over the PCB and seeing if I see anything visually happening with the interface circuitry (I'm an electronics hobbyist that's designed and built my own PCBs so I'm pretty comfortable with this stuff) and maybe test some of the parts. If anyone has any ideas as to what's going on I'd be interested in hearing what you think.

kramttocs, 2025-11-23:
Same here. Motion stopped reporting and the light status doesn't report. 

Have two gdos. One still works fine.