ratgdo/esphome-ratgdo #161: Supporting home made ratgdo boards (as per rat-ratgdo repo)

Kaldek, 2023-12-29:
Hi folks, I am the maintainer of the [rat-ratgdo](https://github.com/Kaldek/rat-ratgdo) repository which was created for people who want to build their own ratgdo circuit boards.  In that repo, I [recommend the use of ESPHome](https://github.com/Kaldek/rat-ratgdo/blob/main/ESPHome%20vs%20native%20ratgdo.md).

We're now seeing people create their own boards using "raw" ESP8266 modules, usually with 4MB flash (i.e. ESP-12F/ESP-12S).  The ratgdo ESPHome repo only considers use of the Wemos D1 Mini Lite when using an ESP8266, which has 1MB flash.  However, 1MB flash is not enough to allow subsequent OTA updates of ESPHome without a two-stage reflash (flash without ratgdo package, then re-flash with package included in the YAML).

I have created a workaround for supporting boards with 4MB flash via adding the following line to the device YAML:
```yaml
esp8266:
  board: d1_mini
```

I am actually not sure what exact what module Paul Wieland is providing; it may indeed even be the D1 Mini with 4MB flash, which would solve the OTA issue in one hit.  If that is the case, I ask that the existing ESP8266 YAML configurations provided here are updated to the above setting so that the OTA issues go away.

Optionally, I also ask that this repo add a new YAML file for a generic ESP8266 with 2MB flash (as there are a few).  I have forked the repo and tested this, using a YAML file that refers to the esp_wroom_02 as the board type and correct pin mappings.  If this is amenable, please either create the file or request that I add the file via a Pull Request.

The relevant YAML changes are below:
```yaml
substitutions:
  uart_tx_pin: GPIO1
  uart_rx_pin: GPIO3
  input_obst_pin: GPIO13
  status_door_pin: GPIO16
  status_obstruction_pin: GPIO15
  dry_contact_open_pin: GPIO14
  dry_contact_close_pin: GPIO12
  dry_contact_light_pin: GPIO0

esp8266:
  board: esp_wroom_02
  restore_from_flash: true
```

Kaldek, 2023-12-29:
For the optional support for ESP8266 with 2MB flash, I do not expect that there should be any support or documentation within this repo for flashing in the web gui of these modules using custom PCBs.  Just the YAML file will suffice, as people flashing their own custom devices will be getting the correct guidance from my own repo.

rlowens, 2023-12-29:
> The ratgdo ESPHome repo only considers use of the Wemos D1 Mini Lite when using an ESP8266, which has 1MB flash.

The v2.5i boards use 12F modules with 4MB flash and the config was corrected in https://github.com/ratgdo/esphome-ratgdo/pull/156 5 days ago. The [v25iboard.yaml](https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v25iboard.yaml) already uses "board: d1_mini"

Adding a 2MB config as you suggest would also be good.

Kaldek, 2023-12-29:
> The v2.5i/v2.52i boards use 12F modules with 4MB flash and the config was corrected in #156 5 days ago. The [v25iboard.yaml](https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v25iboard.yaml) already uses "board: d1_mini"

Gah!  The one thing I didn't check in their existing YAML!