ratgdo/esphome-ratgdo #162: Errors when adopting to esphome dash board

demfam, 2023-12-30:
I hope somebody can help me. I installed RATGOD25 on a esp32 d1 mini and integrated with HA no problems. I don't have a motion 
sensor and I would like implement BT on the ESP32 for remote sensors.  But when I try to adopt it to ESPHome dashboard I get the following errors: 

> dov25-27e468.yaml
> INFO Reading configuration /config/esphome/ratgdov25-27e468.yaml...
> INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
> INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
> ERROR Unable to load component ratgdo.number:
> Traceback (most recent call last):
>   File "/esphome/esphome/loader.py", line 162, in _lookup_module
>     module = importlib.import_module(f"esphome.components.{domain}")
>   File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>   File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
>   File "<frozen importlib._bootstrap_external>", line 790, in exec_module
>   File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
>   File "/config/esphome/.esphome/external_components/74f605c3/components/ratgdo/number/__init__.py", line 23, in <module>
>     number.number_schema(RATGDONumber)
> AttributeError: module 'esphome.components.number' has no attribute 'number_schema'
> ERROR Unable to load component ratgdo.number:
> Traceback (most recent call last):
>   File "/esphome/esphome/loader.py", line 162, in _lookup_module
>     module = importlib.import_module(f"esphome.components.{domain}")
>   File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>   File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
>   File "<frozen importlib._bootstrap_external>", line 790, in exec_module
>   File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
>   File "/config/esphome/.esphome/external_components/74f605c3/components/ratgdo/number/__init__.py", line 23, in <module>
>     number.number_schema(RATGDONumber)
> AttributeError: module 'esphome.components.number' has no attribute 'number_schema'
> ERROR Unable to load component ratgdo.number:
> Traceback (most recent call last):
>   File "/esphome/esphome/loader.py", line 162, in _lookup_module
>     module = importlib.import_module(f"esphome.components.{domain}")
>   File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>   File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
>   File "<frozen importlib._bootstrap_external>", line 790, in exec_module
>   File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
>   File "/config/esphome/.esphome/external_components/74f605c3/components/ratgdo/number/__init__.py", line 23, in <module>
>     number.number_schema(RATGDONumber)
> AttributeError: module 'esphome.components.number' has no attribute 'number_schema'
> ERROR Unable to load component ratgdo.number:
> Traceback (most recent call last):
>   File "/esphome/esphome/loader.py", line 162, in _lookup_module
>     module = importlib.import_module(f"esphome.components.{domain}")
>   File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>   File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
>   File "<frozen importlib._bootstrap_external>", line 790, in exec_module
>   File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
>   File "/config/esphome/.esphome/external_components/74f605c3/components/ratgdo/number/__init__.py", line 23, in <module>
>     number.number_schema(RATGDONumber)
> AttributeError: module 'esphome.components.number' has no attribute 'number_schema'
> INFO Detected timezone 'America/New_York'
> Failed config
> 
> number.ratgdo: [source /config/esphome/.esphome/packages/74f605c3/base.yaml:144]
>   
>   Platform not found: 'number.ratgdo'.
>   platform: ratgdo
>   id: ratgdov25_rolling_code_counter
>   type: rolling_code_counter
>   entity_category: config
>   ratgdo_id: ratgdov25
>   name: Rolling code counter
>   mode: box
>   unit_of_measurement: codes
> number.ratgdo: [source /config/esphome/.esphome/packages/74f605c3/base.yaml:153]
>   
>   Platform not found: 'number.ratgdo'.
>   platform: ratgdo
>   id: ratgdov25_opening_duration
>   type: opening_duration
>   entity_category: config
>   ratgdo_id: ratgdov25
>   name: Opening duration
>   unit_of_measurement: s
> number.ratgdo: [source /config/esphome/.esphome/packages/74f605c3/base.yaml:161]
>   
>   Platform not found: 'number.ratgdo'.
>   platform: ratgdo
>   id: ratgdov25_closing_duration
>   type: closing_duration
>   entity_category: config
>   ratgdo_id: ratgdov25
>   name: Closing duration
>   unit_of_measurement: s
> number.ratgdo: [source /config/esphome/.esphome/packages/74f605c3/base.yaml:169]
>   
>   Platform not found: 'number.ratgdo'.
>   platform: ratgdo
>   id: ratgdov25_client_id
>   type: client_id
>   entity_category: config
>   ratgdo_id: ratgdov25
>   name: Client ID
>   mode: box
> esphome: None
>   name: ratgdov25
>   
>   [friendly_name] is an invalid option for [esphome]. Please check the indentation.
>   friendly_name: RATGDOv2.5 None
>   name_add_mac_suffix: False
>   project: 
>     name: ratgdo.esphome
>     version: 2.5
>   build_path: .esphome/build/ratgdov25
> wifi: [source /config/esphome/.esphome/packages/bf922882/v25board_esp32_d1_mini.yaml:48]
>   
>   expected a dictionary.
>   ap:  [source /config/esphome/.esphome/packages/bf922882/v25board_esp32_d1_mini.yaml:49]
>

rlowens, 2023-12-31:
How are you running ESPHome? Via the Home Assistant Addon or stand-alone? Are you able to compile other devices?

demfam, 2023-12-31:
I am running ESP home as an add-on in Home Assistant. I have compiled both ESP32 and ESP8266 devices but none of them get packages from GitHub

rlowens, 2023-12-31:
Any special firewall or routing issues from your Home Assistant box to the internet?

demfam, 2023-12-31:
I don't have any unique firewall or router issues. I am running HA in VirtualBox on a windows PC and my router is a TP-Link Delco

demfam, 2024-01-02:
looks like I fixed it. I deleted a depreciated version of ESPHome running on my system.

closed 2024-01-02 by demfam