ratgdo/esphome-ratgdo #163: Lock remotes switch in HA unavailable

mattfox27, 2024-01-01:
I have this lock remores switch that is unavailable[image]

mulcmu, 2024-01-01:
#132 changed this from a switch to lock component.  Looks like your setup didn't pull in changes from the base.yaml.
```
lock:
  - platform: ratgdo
    id: ${id_prefix}_lock_remotes
    ratgdo_id: ${id_prefix}
    name: "Lock remotes"
```

bdraco, 2024-01-02:
You should be able to delete the entity in ha after restarting once

shred86, 2024-01-02:
Hm, it's still just showing up as "Unavailable" with Delete being grayed out after an HA restart.

rlowens, 2024-01-02:
I would delete the device in Home Assistance and let it re-add it. Note: deleting the device not the .yaml file.

mattfox27, 2024-01-02:
> #132 changed this from a switch to lock component. Looks like your setup didn't pull in changes from the base.yaml.
> 
> ```
> lock:
>   - platform: ratgdo
>     id: ${id_prefix}_lock_remotes
>     ratgdo_id: ${id_prefix}
>     name: "Lock remotes"
> ```

Thats what I figured happened