ratgdo/esphome-ratgdo #166: Wired correctly, powers on and connects to wifi and HA but doesnt control door/lights/etc

changes13, 2024-01-04:
[I have this panel ](https://www.amazon.com/Chamberlain-41A6317-Original-Equipment-Manufacturer/dp/B071CRNQC4/ref=cm_cr_arp_d_product_top?ie=UTF8)

Attached pic of motor.
Attached log.

The first boot, the panel is not connected.
I rebooted with the panel connected later on in the log session.
[without panel then with panel.txt](https://github.com/ratgdo/esphome-ratgdo/files/13826502/without.panel.then.with.panel.txt)
[image]

What happens:
In both cases, ratdgo does not control the motor or the light.  

Additional info:
Random times where the door opened during boot or manual sync.
The panel is fully functional when wired even with ratdgo connected.
Quadruple checked wiring.

Any help appreciated!

wastingaeons, 2024-01-04:
looks like a secplus1 device, flash the mqtt firmware as this firmware doesn't support secplus1 yet.

changes13, 2024-01-04:
flashed mqtt.  works mostly (sometimes a command doesnt work)

closed 2024-01-04 by changes13