ratgdo/esphome-ratgdo #167: RATGDO - GDO, GDC Sensor Behaviour

rcblackwell, 2024-01-04:
I recently installed a RATGDO device on a Liftmaster garage door opener. Installation was straight forward and easily achieved. Once installed, I modified one of my NodeRed flows to utilize the “binary_sensor.garage_overhead_door” expecting the sensor to report on when the door was open or closed. Unfortunately the flow failed to work properly as the “binary_sensor.garage_overhead_door” sensor doesn’t report a steady state. It flips between “door was opened” and “door was closed” at approximately 2 second intervals.

The RATGDO is flashed with ESPHOME. Here’s a copy of my log during the time the garage door was opened then closed;

```
Garage - Overhead Door was closed
9:51:39 AM - 3 hours ago
Garage - Overhead Door was opened
9:51:32 AM - 3 hours ago
Garage - Overhead Door was closed
9:51:32 AM - 3 hours ago
Garage - Overhead Door was opened
9:51:32 AM - 3 hours ago
Garage - Overhead Door was closed
9:51:32 AM - 3 hours ago
Garage - Overhead Door was opened
9:51:31 AM - 3 hours ago
Garage - Overhead Door was closed
9:51:31 AM - 3 hours ago
Garage Door Motor turned off
9:51:28 AM - 3 hours ago
Garage Door Door was opened
9:51:28 AM - 3 hours ago
Garage - Overhead Door was opened
9:51:25 AM - 3 hours ago
Garage - Overhead Door was closed
9:51:25 AM - 3 hours ago
Garage - Overhead Door was opened
9:51:23 AM - 3 hours ago
Garage - Overhead Door was closed
9:51:23 AM - 3 hours ago
Garage - Overhead Door was opened
9:51:22 AM - 3 hours ago
Garage - Overhead Door was closed
9:51:22 AM - 3 hours ago
Garage - Overhead Door was opened
9:51:19 AM - 3 hours ago
Garage - Overhead Door was closed
9:51:19 AM - 3 hours ago
Garage - Overhead Door was opened
9:51:18 AM - 3 hours ago
Garage - Overhead Door was closed
9:51:18 AM - 3 hours ago
Garage - Overhead Door was opened
9:51:17 AM - 3 hours ago
Garage - Overhead Door was closed
9:51:17 AM - 3 hours ago
Garage Door Motor turned on
9:51:16 AM - 3 hours ago
Garage Door Door is opening
9:51:15 AM - 3 hours ago
Garage Door Light turned on
9:51:15 AM - 3 hours ago
Garage - Overhead Door was opened
9:51:15 AM - 3 hours ago
Garage Door Button turned off
9:51:14 AM - 3 hours ago
Garage Door Button turned on
9:51:14 AM - 3 hours ago
```
Any idea what could be causing this behaviour?

rcblackwell, 2024-01-08:
Here's a little more information related to my installation and experience;

Liftmaster garage door opener - Model 8355W
Liftmaster GD remote control - Model 882LMW

Both manufactured in 2016.

While troubleshooting the issue I disconnected the remote control to see if that might stop repeated messages when the door was open. It did not.

rcblackwell, 2024-01-15:
Stumbled across something interesting that I thought I should share. If the RATGDO logs are viewed using the ESPHome Log viewer rather than the Home Assistant log viewer, the multiple open, close entries that are present in the latter log don't appear. Here's a copy of the log as viewed directly from the ESPHome Log;

```
INFO ESPHome 2023.12.6
INFO Reading configuration /config/esphome/garage-door-opener.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Detected timezone 'America/Toronto'
INFO Starting log output from 192.168.86.82 using esphome API
INFO Successfully connected to garage-door-opener @ 192.168.86.82 in 0.005s
INFO Successful handshake with garage-door-opener @ 192.168.86.82 in 2.559s
[13:42:28][I][app:102]: ESPHome version 2023.12.6 compiled on Jan 15 2024, 12:50:45
[13:42:28][I][app:104]: Project ratgdo.esphome version 2.5i
[13:42:28][C][wifi:573]: WiFi:
[13:42:28][C][wifi:405]:   Local MAC: 08:3A:8D:F9:E0:C6
[13:42:28][C][wifi:410]:   SSID: [redacted]
[13:42:28][C][wifi:411]:   IP Address: 192.168.86.82
[13:42:28][C][wifi:412]:   BSSID: [redacted]
[13:42:28][C][wifi:414]:   Hostname: 'garage-door-opener'
[13:42:28][C][wifi:416]:   Signal strength: -48 dB ▂▄▆█
[13:42:28][C][wifi:420]:   Channel: 11
[13:42:28][C][wifi:421]:   Subnet: 255.255.255.0
[13:42:28][C][wifi:422]:   Gateway: 192.168.86.1
[13:42:28][C][wifi:423]:   DNS1: 192.168.86.1
[13:42:28][C][wifi:424]:   DNS2: 0.0.0.0
[13:42:28][C][logger:439]: Logger:
[13:42:28][C][logger:440]:   Level: DEBUG
[13:42:28][C][logger:441]:   Log Baud Rate: 115200
[13:42:28][C][logger:443]:   Hardware UART: UART0
[13:42:28][C][ratgdo.number:021]: RATGDO Number 'Rolling code counter'
[13:42:28][C][ratgdo.number:021]:   Unit of Measurement: 'codes'
[13:42:28][C][ratgdo.number:025]:   Type: Rolling Code Counter
[13:42:28][C][ratgdo.number:021]: RATGDO Number 'Opening duration'
[13:42:28][C][ratgdo.number:021]:   Unit of Measurement: 's'
[13:42:28][C][ratgdo.number:027]:   Type: Opening Duration
[13:42:28][C][ratgdo.number:021]: RATGDO Number 'Closing duration'
[13:42:28][C][ratgdo.number:021]:   Unit of Measurement: 's'
[13:42:28][C][ratgdo.number:029]:   Type: Closing Duration
[13:42:28][C][ratgdo.number:021]: RATGDO Number 'Client ID'
[13:42:28][C][ratgdo.number:023]:  Type: Client ID
[13:42:28][C][switch.gpio:068]: GPIO Switch 'Status door'
[13:42:28][C][switch.gpio:090]:   Restore Mode: always OFF
[13:42:28][C][switch.gpio:031]:   Pin: GPIO16
[13:42:28][C][switch.gpio:068]: GPIO Switch 'Status obstruction'
[13:42:28][C][switch.gpio:090]:   Restore Mode: always OFF
[13:42:28][C][switch.gpio:031]:   Pin: GPIO15
[13:42:28][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact open'
[13:42:28][C][gpio.binary_sensor:016]:   Pin: GPIO14
[13:42:28][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact close'
[13:42:28][C][gpio.binary_sensor:016]:   Pin: GPIO12
[13:42:28][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact light'
[13:42:28][C][gpio.binary_sensor:016]:   Pin: GPIO0
[13:42:28][C][light:103]: Light 'Light'
[13:42:28][C][ratgdo:081]: Setting up RATGDO...
[13:42:28][C][ratgdo:082]:   Output GDO Pin: GPIO5
[13:42:28][C][ratgdo:083]:   Input GDO Pin: GPIO4
[13:42:28][C][ratgdo:087]:   Input Obstruction Pin: GPIO13
[13:42:28][C][ratgdo:089]:   Rolling Code Counter: 94
[13:42:28][C][ratgdo:090]:   Client ID: 7488825
[13:42:28][C][ratgdo.sensor:012]: RATGDO Sensor 'Openings'
[13:42:28][C][ratgdo.sensor:012]:   State Class: ''
[13:42:28][C][ratgdo.sensor:012]:   Unit of Measurement: 'openings'
[13:42:28][C][ratgdo.sensor:012]:   Accuracy Decimals: 0
[13:42:28][C][ratgdo.sensor:012]:   Icon: 'mdi:open-in-app'
[13:42:28][C][ratgdo.sensor:013]:   Type: Openings
[13:42:28][C][ratgdo.lock:012]: RATGDO Lock 'Lock remotes'
[13:42:28][C][ratgdo.lock:013]:   Type: Lock
[13:42:28][C][ratgdo.binary_sensor:036]: RATGDO BinarySensor 'Motion'
[13:42:28][C][ratgdo.binary_sensor:036]:   Device Class: 'motion'
[13:42:28][C][ratgdo.binary_sensor:038]:   Type: Motion
[... 197 lines cut ...]
[13:44:08][D][cover:170]: 'Door' - Publishing:
[13:44:08][D][cover:173]:   Position: 17%
[13:44:08][D][cover:186]:   Current Operation: CLOSING
[13:44:08][D][cover:170]: 'Door' - Publishing:
[13:44:08][D][cover:173]:   Position: 13%
[13:44:08][D][cover:186]:   Current Operation: CLOSING
[13:44:09][D][cover:170]: 'Door' - Publishing:
[13:44:09][D][cover:173]:   Position: 9%
[13:44:09][D][cover:186]:   Current Operation: CLOSING
[13:44:09][D][cover:170]: 'Door' - Publishing:
[13:44:09][D][cover:173]:   Position: 5%
[13:44:09][D][cover:186]:   Current Operation: CLOSING
[13:44:09][D][ratgdo:636]: Cancelling position callbacks
[13:44:09][D][ratgdo:210]: Status: door=CLOSED light=ON lock=UNLOCKED
[13:44:09][D][cover:170]: 'Door' - Publishing:
[13:44:09][D][cover:173]:   Position: 0%
[13:44:09][D][cover:186]:   Current Operation: IDLE
[13:44:09][D][switch:016]: 'Status door' Turning OFF.
[13:44:09][D][switch:055]: 'Status door': Sending state OFF
[13:44:09][D][number:012]: 'Rolling code counter': Sending state 97.000000
[13:44:09][D][binary_sensor:036]: 'Motor': Sending state OFF
[13:44:10][D][ratgdo:237]: Openings: 4067
[13:44:10][D][sensor:093]: 'Openings': Sending state 4067.00000 openings with 0 decimals of accuracy
[13:44:11][D][esp8266.preferences:238]: Saving preferences to flash...
```

PaulWieland, 2024-01-15:
have you upgraded home assistant to the latest release?

rcblackwell, 2024-01-15:
I'm running;

```
Core - 2024.1.3
Supervisor - 2023.12.0
Operating System - 11.4
Frontend - 20240104.0
```

on an HA Blue (ODROID-N2/N2+)

rcblackwell, 2024-01-23:
I found the problem and, boy, do I ever feel **REALLY, REALLY STUPID!**

I have an ESP8266 programmed and attached to reed switches that detect the state of my exterior doors. One of those doors is, you guest it, the garage door. When setting up a NodeRed flow for exterior lighting triggered by the garage door opening, I mistakenly selected the reed switch named Garage Door. Things started acting up again today so I dug deeply into the logs. It took some time to realize the error I made. With the correct sensor now selected, my NodeRed lighting flow is working the way it should.

Many apologies for the misleading information. I hope I haven't caused to much trouble for you.

closed 2024-01-23 by rcblackwell