ratgdo/esphome-ratgdo #169: Ratgdo 2.5i Wifi Connection issue

007GTCS, 2024-01-05:
I have a 2.5i board and had issues connecting to 2.4ghz channel on my router. I had to set it to legacy mode to get it to connect. Because this board uses an esp8266 chip, I did some research and found I had to add a snippet of code to the yaml that fixes the auth and association issues. It has no problem connecting now.

Also it helps to have your 2.4ghz connection on channel 1, 6, or 11.

on_boot:
    priority: 300
    then:
      lambda: |-
        WiFi.setPhyMode(WIFI_PHY_MODE_11G);

[image]

gtinjr, 2024-01-07:
I ran into the same issue router Asus AX11000 with smart connect on.  I had created a guest network 2.4GHz for IOT devices and my konnected.io boards that is the same chipset as the ratgdo (ESP-12F) have been running in the guest network for over 2 years.  This code above solve the issue for me.  Thx

jfarlik, 2024-01-08:
I also have a wifi connection issue that could be solved by adding "fast_connect: true" to the wifi section of the 2.5i yaml, but for the life of me I cannot figure out how to clone/build my own custom fork. Can this be added to the yaml for the web installer or someone point me in the direction on how to do that. I forked, but the install page under static gets fails to download the manifest.

bdraco, 2024-01-08:
Install the dashboard addon, adopt the device, and adjust the yaml

https://esphome.io/guides/getting_started_hassio.html

jfarlik, 2024-01-08:
I can't, running docker core in a container.

rlowens, 2024-01-08:
> the 2.5i yaml, but for the life of me I cannot figure out how to clone/build my own custom fork.

If you can't adopt it, you can just start from this device .yaml and compile and flash that. Change to fit your needs:

```
substitutions:
  name: <name what you want or the old name if you want to maintain that, mine was ratgdov2-11999c>
  friendly_name: <as above, name as you want>
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: [put whatever you want here since you are going to have to USB flash the first time anyway, eg xxxxxxghy47j8zEC4IVqJousY+SZI+U1S0X1VWLp3Zz8=]
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

```

chocobo-recluse, 2024-01-12:
I would like to thank @007GTCS for raising this post.
I just got my v2.53i board and experiencing the same wifi connection issue.
I can't get it to connect to my wifi network after installing the firmware via the [Web Tool Flasher](https://ratgdo.github.io/esphome-ratgdo/)

After reading this post, in Home Assistant ESPHome add-on, I added a new device, replicated this [yaml](https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v25iboard.yaml) content, added the "on_boot" section, and api encryption key. Then, installed it on the board.
When finished, It showed up in ESPHome automatically, ready to adopt.

I am not exactly sure which part did the magic (I am a ESPHome Noob), but it worked.
During the adoption, I even installed it for a second time, with a yaml similar to rlowens suggestion https://github.com/ratgdo/esphome-ratgdo/issues/169#issuecomment-1881630562 , and it still able connecting to my wifi afterward.

Anyway, just want to share my experience and say thank you