ratgdo/esphome-ratgdo #172: OTA failing on size?

calisro, 2024-01-08:
Probably something I can solve myself but I'm very new to esphome and I am really not sure.  I could probably connect directly and flash a new one but that would be a real pain to do everytime there's an update.  

Is this a device problem?  

Please advise?

```
INFO Successfully compiled program.
INFO Resolving IP address of ratgdov25-8166d2.local
INFO  -> 10.100.1.221
INFO Uploading /data/build/ratgdov25-8166d2/.pioenvs/ratgdov25-8166d2/firmware.bin (614928 bytes)
INFO Compressed to 414510 bytes
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)
====== [ERROR] /config/esphome/ratgdov25-8166d2.yaml ======

======================== [SUMMARY] ========================
  - /config/esphome/ratgdov25-8166d2.yaml: FAILED

```

rlowens, 2024-01-08:
Duplicate of https://github.com/ratgdo/esphome-ratgdo/issues/155

calisro, 2024-01-08:
I just found this one.  Sorry to duplicate.

https://github.com/ratgdo/esphome-ratgdo/issues/155

Does that mean the next version will update automatically rather than the 2 step process in the comments of that issue?

calisro, 2024-01-08:
> Duplicate of #155

Yes I just found that too at the same time.

closed 2024-01-08 by calisro

rlowens, 2024-01-08:
> Does that mean the next version will update automatically rather than the 2 step process in the comments of that issue?

What template are you using?
If it is "package_import_url: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main"
then yes, this is corrected from "board: d1_mini_lite" (1MB flash) to "board: d1_mini" (4MB flash)

If you are using "package_import_url: github://ratgdo/esphome-ratgdo/v25board_esp8266_d1_mini_lite.yaml@main"
then, no, it still uses the "board: d1_mini_lite" (1MB flash) setting.

If you are set to use this template, you can use the code at https://github.com/ratgdo/esphome-ratgdo/issues/155#issuecomment-1873856317 to test if your ratgdo really has 4MB flash and you can change to the other template.

calisro, 2024-01-08:
It's currently using "github://ratgdo/esphome-ratgdo/v25board_esp8266_d1_mini_lite.yaml@main".  

If the board is a 4MB, how do I get it switched over to the github://ratgdo/esphome-ratgdo/v25iboard.yaml@main firmware?  Can I do that wirelessly?  or do I need to plug it in and reflash?  

Thanks for your help.

calisro, 2024-01-08:
Following this I was able to fix it.  I think that should address future OTAs now too.  

https://github.com/ratgdo/esphome-ratgdo/issues/136#issuecomment-1865903037

Thanks for your help!

rlowens, 2024-01-08:
> If the board is a 4MB, how do I get it switched over to the github://ratgdo/esphome-ratgdo/v25iboard.yaml@main firmware?

FYI: You can just edit your device .yaml and change 
`package_import_url: github://ratgdo/esphome-ratgdo/v25board_esp8266_d1_mini_lite.yaml@main
`
 to 
`package_import_url: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main"
`
then compile and OTA update.

Or just stick with the 
```
esp8266:
  board: d1_mini
```
added to the bottom as you have now.

calisro, 2024-01-08:
> FYI: You can just edit your device .yaml and change
> `package_import_url: github://ratgdo/esphome-ratgdo/v25board_esp8266_d1_mini_lite.yaml@main `
> to
> `package_import_url: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main `
> then compile and OTA update.

When I tried this it failed still.  But I did get it sorted eventually with the 'esp8266:
  board: d1_mini'

rlowens, 2024-01-08:
You would need to go thru the "minimal firmware install with the 4mb setting correction, then back to full firmware" steps either way, but now that it is working with a 4mb firmware installed you can specify it either way now (with the v25iboard or with the custom overrride at the bottom).