ratgdo/esphome-ratgdo #174: ratgdo out of sync with door position (open vs. closed)

NukeThemTillTheyGlow, 2024-01-15:
ratgdo 2.5i with 2.57 ESPHome.  One of three, other ESPHome is fine, 3rd is MQTT (also fine).

I noticed late last night that HAOS is showing the garage door as open, when its physically closed.  MYQ app shows it closed too.  Too late to mess with it, so I go to bed.  This am, check door.  Its closed, been closed all night.  HAOS/ESPHome shows door open.  web ui for that device shows it open.  Door is closed.

Not sure how they got out of sync.  Other than getting ladder out and pulling power to it to reset, how can I reset?  When I look at the HAOS/ESPHome UI, I have options for "toogle door", "restart", "sync" and more, but I can't find any documentation that describes what they do and/or what I should be doing in this situation.  Same for the device web ui.    

**What should I do/try?**

NukeThemTillTheyGlow, 2024-01-15:
> Just to confirm you are using the ESP Home firmware with a security + 2.0 opener?
> 
> Sec+ 2.0 sends out the current door status every 5 minutes or so which would normally prevent the scenario you are describing. Can you please share the log from ESP Home?

Yes, Security 2.0.  8550W GDO.  Thats what I thought was the case with the sync.  Log attached.
[logs_ratgdov25i-e5ec77_logs (1).txt](https://github.com/ratgdo/esphome-ratgdo/files/13941248/logs_ratgdov25i-e5ec77_logs.1.txt)

PaulWieland, 2024-01-15:
Can you click the query status button and paste the log?
[image]

```12:03:38	[D]	[button:010]	'Query status' Pressed.
12:03:38	[D]	[number:012]	'Rolling code counter': Sending state 230.000000
12:03:38	[D]	[ratgdo:210]	Status: door=OPEN light=OFF lock=UNLOCKED```

NukeThemTillTheyGlow, 2024-01-15:
> Can you click the query status button and paste the log? <img alt="image" width="463" src="https://private-user-images.githubusercontent.com/4663918/296822324-8c2b638e-be77-4824-ba5a-b762b2375750.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUzMzg1OTgsIm5iZiI6MTcwNTMzODI5OCwicGF0aCI6Ii80NjYzOTE4LzI5NjgyMjMyNC04YzJiNjM4ZS1iZTc3LTQ4MjQtYmE1YS1iNzYyYjIzNzU3NTAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDExNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMTVUMTcwNDU4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTRlNzExMzVkOGE1ZDQ4ZGNiMDI0NTZlNTQ1MGI2M2VlYWNlYzdkN2JhMTk3YWViN2JiN2U2OWQwYmUxM2NlZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.R1ivZ9WxWNdeLg6kdVwMwxhHPd-hofzunvNwn19XB4o">
> 
> ```
> 12:03:38	[D]	[button:010]	'Query status' Pressed.
> 12:03:38	[D]	[number:012]	'Rolling code counter': Sending state 230.000000
> 12:03:38	[D]	[ratgdo:210]	Status: door=OPEN light=OFF lock=UNLOCKED
> ```

[logs_ratgdov25i-e5ec77_logs (2).txt](https://github.com/ratgdo/esphome-ratgdo/files/13941422/logs_ratgdov25i-e5ec77_logs.2.txt)

PaulWieland, 2024-01-15:
The GDO isn't responding. Does it respond to door/light commands?

NukeThemTillTheyGlow, 2024-01-15:
> The GDO isn't responding. Does it respond to door/light commands?

Light yes, works on and off.  Only have down arrow and when I press that nothing happens.   Proper status is displayed in MyQ app.  So I opened the door via MyQ app, door opened.  Door controls didn't change in HAOS, but when I pressed the down arrow the door closed, but arrow remained down and up greyed out.  I recorded the log of all of this and attached.
[logs_ratgdov25i-e5ec77_logs (3).txt](https://github.com/ratgdo/esphome-ratgdo/files/13941488/logs_ratgdov25i-e5ec77_logs.3.txt)
 

Is there a way to reboot the ratgdo?  I don't want to just start clicking things I don't understand.

NukeThemTillTheyGlow, 2024-01-15:
Couple of things:

1. I tried the "restart" button in the device web UI and that didn't seem to do anything.  Door down, but button in HA shows only down arrow/up arrow is greyed out.
2.  I glance at HA notifications and now I'm seeing this message " GDO_Denali sync failed
Failed to communicate with garage opener on startup; Check the GDO_Denali Rolling code counter number entity history and set the entity to one number larger than the largest value in history. ESPHome devices"

Should I make that edit/change with the rolling counter number?

3. HA said there was a ESPHome update and I applied it. Now getting notice to update both of my ESPHome ratgdo devices from 2023.12.5 to .6.   Should I go ahead?

mariusmuja, 2024-01-15:
@NukeThemTillTheyGlow 

> Should I make that edit/change with the rolling counter number?

From the symptoms you describe above, ratgdo is able to transmit to the garage opener, so the rolling code is ok. The problem is that it's unable to receive the opener responses, so it thinks it's in an incorrect state (door open, when in fact it is closed, etc). Because of this the up/down arrows are not going to work correctly. The toggle door button should continue to work correctly because it doesn't depend on the state of the door, it behaves the same a pushing the button on the wall.

I have noticed something similar for my opener since the weather got very cold, the signal from the opener got "worse", it would not bring the power/data line to (or close to) zero when sending a response, but only to 2-4V and the low threshold looks "noisy". This causes an issue that it doesn't reliably trigger the RX mosfet and the response cannot be read correctly. In my case, if leaving the light on the opener on for some time, the signal from the opener got better and ratgdo started being able to read responses again.

I don't if it's the same for you, it might be just my opener that has an issue, but the symptoms are the same: ratgdo is able to send commands, is able to received motion events from the wall panel, but is not able to receive responses form the opener. Did the temperature also got colder where you are since the issue began?

In my case, I got it to work more reliably by putting a voltage divider between the power/data line and ground at the gate of the RX mosfet.

NukeThemTillTheyGlow, 2024-01-15:
> @NukeThemTillTheyGlow
>
>
> From the symptoms you describe above, ratgdo is able to transmit to the garage opener, so the rolling code is ok. The problem is that it's unable to receive the opener [...]

Interesting.  I meant to mention that it'd been very cold here the past for days.  Was 7F this am here and my garage isn't heated, so I assume it was in the low teens inside.    Sounds like it could be the GDO being cold too.  hhmmmm, the other ESPHome ratgdo is on a different model Liftmaster GDO and is working fine as is the MQTT on the very old (20 yr) builder grade Chamberlain GDO.

PaulWieland, 2024-01-15:
@mariusmuja which version of the control board did you notice this on?

@NukeThemTillTheyGlow which version of the control board are you having the issue with?

mariusmuja, 2024-01-15:
@PaulWieland My case is quite possible just an issue with my GDO, the signal from the GDO was looking quite "bad" on the oscilloscope (unfortunately I didn't save an image). I'm pulling 3.3v power from the GDO, today I'll open it up and remove that, it's possible that noise introduced by the esp board on the power rail causes issues or the esp is pulling more power than that rail can provide. I know of people reporting decreased remote range when taping into the 3.3v GDO power...

PaulWieland, 2024-01-15:
Yes the wrong internal 3.3v supply can cause issues if there isn't enough overhead available. I found a rock solid source on my model, there are many logic boards out there. I've tested in extreme temperatures and have never seen an issue.

NukeThemTillTheyGlow, 2024-01-15:
> @mariusmuja which version of the control board did you notice this on?
> 
> @NukeThemTillTheyGlow which version of the control board are you having the issue with?

2.5i

NukeThemTillTheyGlow, 2024-01-16:
Checked temp with ir thermometer. 34.5 F

NukeThemTillTheyGlow, 2024-01-16:
I applied the ESPHome updates to both ratgdos and results are the same, one works, the other one doesn't.

PaulWieland, 2024-01-16:
Since you have two of them, can you swap the control boards to see if the issue follows the ratgdo board or stays with the GDO?

NukeThemTillTheyGlow, 2024-01-16:
> Since you have two of them, can you swap the control boards to see if the issue follows the ratgdo board or stays with the GDO?

Not a bad idea.  It's supposed to warm up a little over the next two days, then drop down again.  Let me see if I can swap them today or tomorrow.  If I understand the situation correctly, the garage door opener isn't updating the ratgdo with the right status info for the door and its state/location?

PaulWieland, 2024-01-16:
I sincerely doubt this is a temperature related issue, but lets see...

NukeThemTillTheyGlow, 2024-01-16:
reset the break for the garage ceiling plugs in an effort to see if a power cycle of the GDO might help.  Nothing changed.  Then again, both ESPHome ratgdo's are connected to battery backed up openers, so that really didn't fully test out.  The MQTT/non battery backed up ratgdo/opener combo works great.  I'm going to go out and swap them shortly.

mariusmuja, 2024-01-16:
> reset the break for the garage ceiling plugs in an effort to see if a power cycle of the GDO might help. Nothing changed. Then again, both ESPHome ratgdo's are connected to battery backed up openers, so that really didn't fully test out.

It might be worth trying to disconnect the battery too to fully reset the GDO.

mariusmuja, 2024-01-16:
After removing the 3.3V tap, the responses from the GDO look good now, so that was potentially the issue it in my case... I say just potentially, because I remember that when I tested with the 2.5i board a couple days ago I powered it from a USB charger, not from the 3.3v tap and the GDO responses still looked bad...

There's a second confounding variable: before removing the 3.3V tap I was probing around on the GDO board (while on top of a ladder) and I manager to short the 12V bus to the 3.3V bus :grimacing: . The GDO stopped working and I was sure I completely fried the board. After a quick look with the a thermal camera, I noticed that the embedded wifi board was overheating (indicating a short inside it) and since I had nothing to loose I ripped off the wifi board and the GDO board started working again.

So now with two less power drain sources on the 3.3V rail (the embedded wifi board and the esp), the GDO responses look nice and sharp, so there you go @NukeThemTillTheyGlow , you just need to rip out the embedded wifi board... `</joking>`.

Here's a pic of the carnage:
[image]

mariusmuja, 2024-01-16:
> I sincerely doubt this is a temperature related issue, but lets see...

Probably, for me it was correlated with the temperature drop, but correlation is not causation... Maybe it was borderline working fine for months and the temperature change just pushed it over the edge.

NukeThemTillTheyGlow, 2024-01-16:
Houston I have a problem!   This was the first GDO I installed a ratgdo on of the 3 total I have and I struggled a bit with the wiring at first.  I took the red/white wire from the wall control (I think its the 880LMW) and inserted it into the red/ctrl port on the upper left of the ratgdo and then 3 wires from the ratgdo blk/obst, wht/GND, red/ctrl to the GDO in positions blk, left most white, red respectively.  Thats worked fine until this current issue.   But on ratgdo's 2 and 3, I just ran red, white and black wires between the ratgdo & GDO to keep it simple.

So I test power off/battery out, that doesn't change anything.  So first think I notice when I climb up to swap ratgdo's is that I have the 4th wire (red/ctrl) in the top corner of the ratgdo and I decide to clean that up first to keep everything the same.  Now I get error message on wall control display saying "press the push bar to activate control".  I do that.. long pressed, short presses, etc and nothing happens.  Now wall control doesn't sync up/work for GDO.  I can use MyQ app and it works fine.  ratgdo still doesn't work other than to close the garage door if its in the up position, up arrow doesn't illuminate at all if down.

I have to make sure I get wall control back working or wife will kill me.  Any suggestions??

mariusmuja, 2024-01-16:
The "press the push bar to activate control" happens because you removed power/battery from the GDO and the wall panel needs to "pair" with the GDO again. They exchange some messages when you press the button for the first time.

The fact that the button doesn't pair actually fits with the symptoms ratgdo was showing, that it cannot read responses from the GDO. Before you power cycled the GDO the button was working because it didn't need to receive anything from the GDO, it was just sending the button/light presses, motion events. Now it needs to read the pair request to respond to it and it cannot do it. All this indicates that something is wrong with the GDO...

* You can try to remove ratgdo to remove a variable (just the GDO and the wall panel) and try like that. It probably won't help, but worth trying.

* Try removing the battery and have the GDO just on line power, just in case the battery charging circuitry causes an issue.

* Try the GDO unplugged, just on battery power.

If none of that works, probably there's something wrong with the GDO board and you might need to replace the board (cheaper than replacing the GDO). I was joking before, but if you get there, you have nothing to loose by ripping out the embedded wifi card like I did, just in case it was the one causing the issue.

NukeThemTillTheyGlow, 2024-01-16:
> The "press the push bar to activate control" happens because you removed power/battery from the GDO and the wall panel needs to "pair" with the GDO again. They exchange some messages when you press [...]

Great minds think alike!   So, here's what I've done do far:

1. unplugged battery so that I can control GDO power.  Unplugged GDO and disconnected ratgdo from GDO/powered it off.
2. Took 880LMW off of wall, wire connections were tight, but disconnected them and put wire toner on to confirm wire pair.  On GDO used toner to confirm wire pair and made sure red/white was in red socket, white was in white.
3. Powered up GDO, get pairing message on control panel.....click and hold a few times as well as click and release.  Nothing, no pairing.

Your comments make sense related to there being an issue with core functionality of the GDO, but the MyQ app will open and close the GDO.

mariusmuja, 2024-01-16:
> but the MyQ app will open and close the GDO.

The issue is only with the GDO not sending messages on the wire correctly (would be interesting to put an oscilloscope on the wire to see what the signal looks like). The MyQ app talks to the GDO over wifi, not wireline messages involved (so yeah, that would stop working if you rip out the embedded wifi card... ).

mariusmuja, 2024-01-16:
By the same logic, if you have a keypad, car remote, etc, those communicate with the GDO over radio and should be still working.

NukeThemTillTheyGlow, 2024-01-16:
Found this article on the Liftmaster site: Seems to fit my situation well.  [https://support.chamberlaingroup.com/s/article/How-to-troubleshoot-a-door-control-panel-1484145519278?_=1571434473434&anonymousBody=List%3CHow_To__kav%3E%20articleIds%20%3D%20%5B%20SELECT%20Id%2C%20KnowledgeArticleId%20FROM%20How_To__kav%20WHERE%20PublishStatus%20%3D%20%27Online%27%20AND%20Language%20%3D%20%27en_US%27AND%20Original_Article_Number__c%20%3D%20%27%27%20LIMIT%20100%20%5D%3B%0A%0A%2F%2Ffor%20each%20article%20in%20the%20set%2C%20create%20a%20new%20draft%20version%20%28without%20making%20the%20published%20version%20draft%29%0Afor%28How_To__kav%20article%20%3A%20articleIds%29%7B%0A%20%20%20%20try%7B%0A%20%20%20%20%20%20%20%20KBManagement.PublishingService.editOnlineArticle%28article.KnowledgeArticleId%2C%20false%29%3B%20%20%20%20%0A%20%20%20%20%7D%0A%20%20%20%20catch%28exception%20e%29%7B%0A%20%20%20%20%20%20%20%20system.debug%28%27article%20already%20has%20a%20draft%20version%3A%20%27%20%20article%29%3B%0A%20%20%20%20%7D%0A%7D&r=140&ui-knowledge-components-aura-actions.KnowledgeArticleVersionCreateDraftFromOnlineAction.createDraftFromOnlineArticle=1](url)

Results: Used scrap 22 g wire directly from 880lmw to terminals on the GDO.  panel lights up, but no messages or display details.  I've tried clicking and holding the button, no success.  Running an errand right now and will be back in an hour, but its not looking good.  I guess this takes the wires out of the fault tree, but could be the control panel.. could be the GDO, right?

NukeThemTillTheyGlow, 2024-01-17:
Back home and now the display on the control panel displays the same message as earlier : "Press the push bar to activate the control".  I tried it a few times and nothing happens.   Looking at logic board pricing & availability, any definitive way to decide if its the control panel or the logic board?

mariusmuja, 2024-01-17:
>  Looking at logic board pricing & availability, any definitive way to decide if its the control panel or the logic board?

Given that both ratgdo and the wall panel cannot talk to the GDO, my guess is that it's the board inside the GDO. 

If you're looking for a *definitive* way, that would be an oscilloscope to look at how the signal on the data line looks like when the GDO is transmitting.

Here's something you can try: when my GDO was acting this way (ratgdo was not able to receive responses from it), if I would turn on the light on the GDO after a while (2-3 min), things started working. You could try to turn the light on the GDO (either through MyQ if it's an option, or open/close the door in MyQ would turn on the light). After the light's been on for a couple of minutes, try pressing the button on the wall panel to see if it pairs.

mariusmuja, 2024-01-17:
> I have to assume that is a clear indicator of the board not working properly?

Yeah, I would say so. When you unplug the GDO/plug it back in, do the leds around the arrow buttons light up? They should light up in sequence to indicate the board starting up.

NukeThemTillTheyGlow, 2024-01-17:
> > I have to assume that is a clear indicator of the board not working properly?
> 
> Yeah, I would say so. When you unplug the GDO/plug it back in, do the leds around the arrow buttons light up? They should light up in sequence to indicate the board starting up.

No, only center button between the up/down arrows lights up and it flickered a few times just now when I swapped wall controls.

I just went through what I think is the final test.  powered off GDO, swapped wall control from GDO next to it, exactly same model wall control, slightly differnt GDO.   "bad" wall control worked fine on other GDO, sync'd up on first button press.   "Good" wall control has not even had its display light up yet on the "bad" GDO.  Door sensors do light up, I was looking in wrong location earlier.  One green, one orange.  Door continues to open/close via MyQ app.   I'm going to give it a few more minutes, but I checked prices and shipping options and liftmaster has slightly more price on board ($118), but offers overnight shipping for only $15.

mariusmuja, 2024-01-17:
> I'm going to give it a few more minutes, but I checked prices and shipping options and liftmaster has slightly more price on board ($118), but offers overnight shipping for only $15.

If you're going to replace it anyway, you don't have anything to loose by trying to rip out the embedded wifi card like I did, who knows, maybe that would "fix" it for you too. If it doesn't you would loose the ability to open the door with the MyQ app though, until a new board shows up.

NukeThemTillTheyGlow, 2024-01-17:
> If you're going to replace it anyway, you don't have anything to loose by trying to rip out the embedded wifi card like I did, who knows, maybe that would "fix" it for you too. If it doesn't you [...]

Since the GDO is in my wifes garage bay, there would be more ripping than just the wifi if I don't get this fixed asap! lol

NukeThemTillTheyGlow, 2024-01-17:
I just did a final check one wall control swap, no display, nothing I can think of will lite it up.  I was about to get down ladder and order the new logic board, saw the ratgdo and said hey, why not?  No wall control wiring is installed, just the door sensors into the GDO.  So I connected the red, white ratgdo wires to the red, white wall control porta and then the black ground.  Powered up the ratgdo and came back to my PC.  

In HAOS, the dashboard display for the door looked like it did the other day, up arrow dimmed out, the stop and down arrows are lit.  I decided to see if I could login to the web ui and sure enough, no issues.  so I scroll down and for the heck of it, click on the up arrow in the web ui.  garage door was opened earlier by me via MyQ......**_the door closes!_**   So I experiment and the web ui works fine!  up, down and stop.   The HAOS display only will close the door, not open it (dimmed out).

So now I'm totally confused.  I don't remember trying to open/close/stop the door via the web ui before.

mariusmuja, 2024-01-17:
It's not that strange, the HAOS display dims out the buttons depending on the door state, but since ratgdo doesn't receive from the GDO, it thinks the door is in the wrong state. The web UI doesn't dim the buttons.

There is also a "Door toggle" button in both HAOS and web UI that should always toggle the door (exactly like the wall button does), not matter what the door state is.

NukeThemTillTheyGlow, 2024-01-17:
> It's not that strange, the HAOS display dims out the buttons depending on the door state, but since ratgdo doesn't receive from the GDO, it thinks the door is in the wrong state. The web UI doesn't [...]

Thank you, just making sure something else isn't going on.   Ahhh, I'm playing with the toggle button in both the web ui and HAOS, it works in both locations.   Interesting that the board will talk to the ratgdo, but not the wall control, but when I tested different wall controls, no luck just seems odd.  
Off to order the new logic board.

NukeThemTillTheyGlow, 2024-01-17:
> > Interesting that the board will talk to the ratgdo, but not the wall control
> 
> Yeah, that is bit strange. The GDO powers the wall panel though the data line, while ratgdo has it's own power.
> 
> Since the GDO cannot power the wall panel, it probably means that the voltage on the data line is too low, but enough for ratgdo to issue door commands and for the GDO to receives those commands.

I was thinking the same thing, its a power issue from the board.  I almost climbed back up to try the same wiring between the ratgdo and GDO as I was using to test the various wall controls, but your response helped confirm this for me and I just ordered the new board.

NukeThemTillTheyGlow, 2024-01-17:
Thinking....does the board have self contained power supply or is that a seperate device that I should be concerned about?

mariusmuja, 2024-01-17:
There's a transformer from line voltage to 12V AC inside the GDO separate from the board, but the rectifier and the 3.3V regulator are on the board. If the transformer were bad, nothing would work, so the issue is certainly with the board.

NukeThemTillTheyGlow, 2024-01-17:
> There's a transformer from line voltage to 12V AC inside the GDO separate from the board, but the rectifier and the 3.3V regulator are on the board. If the transformer were bad, nothing would work, so the issue is certainly with the board.

Thank you marius, you have been a huge help and sounding board.  Board ordered, I paid for overnight shipping to get this issue resolved and behind me.. and wife happy, asap!

mariusmuja, 2024-01-19:
@NukeThemTillTheyGlow 

> Board ordered, I paid for overnight shipping to get this issue resolved and behind me.. 

Other people reported having issues with openers with battery backup, can you confirm when you get the new board and install it that it solved the issue?

NukeThemTillTheyGlow, 2024-01-19:
I sure will.  Looks like Fedex is having issues so now board is delayed again, will not get until Monday.   I'm leaving for a work trip Tue morning, so I will work on it Monday night

NukeThemTillTheyGlow, 2024-01-23:
New board arrived today (Finally!) and I just installed it.  Good news, the wall control panel powered up and paired on the first try.  But I'm trying to get the initial setup of the GDO completed and I'm getting error 4-2 on the display indicating excessive force to life the door.  I am working to diagnose and fix it now.   Other good news was that the ratgdo is alive and indicates the door being in a differnt position (closed) and will work the lights on and off, but can't open the door as the GDO is not letting it, just like it won't on the keypad until error 4-2 is cleared.  So progress!

mariusmuja, 2024-01-23:
You need to set the open-closed limits on the door:

There's two up & down buttons and a black rectangular button in between them. Press the black button until the up button flashes. Now keep the up button pressed (the door will open) until the door gets to the desired open position. You can release/press to fine-tune the position. When at desired open position press the black button to set the open limit and now the down button will flash. Repeat the same for the closed position. Press the black button again to set the closed limit. Now the up button will flash again, a single press will open the door to test the up limit. Then the down button will flash and a single press will close the door to test the closed limit.

NukeThemTillTheyGlow, 2024-01-23:
> You need to set the open-closed limits on the door:
>
> There's two up & down buttons and a black rectangular button in between them. Press the black button until the up button flashes. Now keep the up [...]

Thank you!  I just found the install manual and just read about how to do it, your directions are spot on.  Going back to the garage now to fix.  Then when wife gets home, reprogram remote and then finally reset up MyQ...as I have not deployed the HAOS app and appropriate dashboard/controls for her and my kids yet.  This GDO acted as both a MyQ GDO as well as hub for the 3rd/old GDO via upgraded keypad/wall control.  I need to go back and find out/remember how to set that all up again.

NukeThemTillTheyGlow, 2024-01-23:
Setup complete!  Ratgdo fully functional again!

mariusmuja, 2024-01-23:
Good news and it's good to have confirmation that a GDO board replacement fixes the issue. To double check, the board replacement was model/version 050DCTBFMC?

NukeThemTillTheyGlow, 2024-01-23:
Now getting 4-3 excessive closing force errors.  Will reprogram.  Myq setup worked fine, but not able to get oldest gdo to be detected during myq process.  That mqtt ratgdo works fine

NukeThemTillTheyGlow, 2024-01-23:
> Now getting 4-3 excessive closing force errors. Will reprogram. Myq setup worked fine, but not able to get oldest gdo to be detected during myq process. That mqtt ratgdo works fine

I did another reset/setup on the travel and got it right this time.  handheld and myq openers setup and working.  Need to get the exterior keypad over the weekend done.  and also get the 3rd gdo to join the myq config.   But GDO Brain Transplant complete!

NukeThemTillTheyGlow, 2024-01-23:
> Good news and it's good to have confirmation that a GDO board replacement fixes the issue. To double check, the board replacement was model/version 050DCTBFMC?

Yes, its a 050DCTBFMC, that is copied/pasted right from the liftmaster order.

hellboyfishman, 2024-02-03:
Not sure if anyone is still following this, but I also just completed a GDO brain transplant with the 050DCTBFMC board.    It was pretty straight forward and my RATGDO is now functioning as expected.  It's been warmer than usual so the true test will be when the temps really dip again.

I didn't hook this board to MyQ, I had the other ones blocked from internet access anyway and I've had the wife using HomeAssistant.   Now to get all my automatons dialed in.   Thank you @NukeThemTillTheyGlow for being the first brain surgeon.

NukeThemTillTheyGlow, 2024-02-04:
> Not sure if anyone is still following this, but I also just completed a GDO brain transplant with the 050DCTBFMC board. It was pretty straight forward and my RATGDO is now functioning as expected. [...]

@hellboyfishman You're very welcome.  I'm in a similar situation as you...it got warmer and I'm curious to see what happens when/if we get back down to single digits.   Its also odd how the Lifetmaster 3 button remotes (8xxMax) are so flaky with working with that door now  Program one via wall control panel, it works fine to open/close, but then get in wifes car just outside the door and its hit or miss.  changed battery in it, etc.  I want to cut wife and kids over to super simple HA app for their iPhones, but need to solve for the requirement of getting them on the home network when outside of the house.
 
Other update is I found out that the 3rd opener I have, very old/20 years/original to house when we build it, was supported with ESPHome/Security 1.0 support.  So yesterday took down the ratgdo, reflashed it to ESPHome/Sec 1.0 and it works great.   Now to address the remote issue, the random light coming on the newest liftmast GDO..I think its the wall panelm its the original one from the opener that I replaced the board in.....but I moved it over to the other opener to see if it was the board or the wall control.. I think someting it wrong.

jschafermo, 2024-02-04:
> @hellboyfishman You're very welcome. I'm in a similar situation as you...it got warmer and I'm curious to see what happens when/if we get back down to single digits. Its also odd how the Lifetmaster [...]

Did you by chance add or change to LED bulbs when you started having remote issues?  When I switched over to LED bulbs, my first brand of LEDs would kill or interfere with all my remotes, including factory installed remotes in our vehicles.

hellboyfishman, 2024-02-05:
> I'm in a similar situation as you...it got warmer and I'm curious to see what happens when/if we get back down to single digits. Its also odd how the Lifetmaster 3 button remotes (8xxMax) are so [...]

We will be back down in the 20s and 30s overnight in the next week or so, I plan on testing the light immediately when I wake up in the morning to see how that is functioning.  I will say that the thing I had the most issues with on the old board, outside of light updates, was "opening" status.   Even as it warmed up into the 20s and 30s the RATGDO would most often only report opened, closing, and closed status.  Since the swap I always get an opening status which is great because that's how I prefer my alerting automatons to trigger.

I have reprogrammed 2x 3 button remotes for vehicles and 1 keypad and so far so good, I'm crossing my fingers on that one now.  

Also, I know this was another $100 to chamberlain but in the long run, I feel like it was worth it for the 10 min process to swap it and get off their cloud crap.

restauffer, 2024-02-05:
> Did you by chance add or change to LED bulbs when you started having remote issues? When I switched over to LED bulbs, my first brand of LEDs would kill or interfere with all my remotes, including [...]

Another possibility is interference from a 5V wall wart bring used to power the RATGDO. When I installed my RATGDO, remote openers became very unreliable. If I unplugged the wall wart, all worked OK.  I wasn’t surprised since wall warts are notorious RFI generators. I moved the RATGDO and wall wart to a sidewall of the garage and ran a longer cable from it to the GDO.  Remotes started to work properly again. The new arrangement had the added advantage of allowing access to the RATGDO without needing to use a ladder.

arrisfe, 2024-02-20:
I have a very similar issue to this, but when hitting the query openings or status buttons, they will not return a response when my garage is below 50 degrees Fahrenheit. During this time, if I open or close my garage door while using the wall button or a remote control, the Apple Home and Home Assistant app will be out of sync. Things stay in sync if I use the Apple Home or HA app exclusively.

While the apps are out of sync, I can still use them. If the garage door is open but the apps report it's closed, I can select the button to open it, and it will go through the motions as if it were opening the door. Then, if I hit the close button, the garage door will indeed close, and things will be in sync again.

These past few weeks have had many cold nights, dropping the temps in the garage below 50. When the out-of-sync issue happens, and I leave it that way on purpose, during the day when the garage warms to over 50 degrees, all the sudden, I'll get a notification that my garage door has either opened or closed (depending on the state it was in) and the apps will all the sudden become in sync again with the actual state of the door.

I never lose communication with the ratgdo as I've been monitoring it consistently since I've noticed the out-of-sync issue. When the temps are above 50 in the garage, I never have a sync issue, and I can hit the query openings/status buttons with 0% failure.

Is this also indicative of the GDO board itself as the issue and needs to be replaced?

hellboyfishman, 2024-02-20:
> Is this also indicative of the GDO board itself as the issue and needs to be replaced?

This sure sounds like the issue some of us have been facing.  

If you want to delay buying the $100 mainboard, there's one more step you can try, when it is out of sync, turn on the light on the GDO using the wall button, leave the light on for a while (10-30 min) and see if the status updates automatically.  I believe that leaving the light on warms the board enough to update it properly.

mariusmuja, 2024-02-20:
@arrisfe, can you also test @jschafermo's hypothesis above. Are you using LED light bulbs on the GDO, and if so what happens if you unscrew the light bulbs?

> Did you by chance add or change to LED bulbs when you started having remote issues? When I switched over to LED bulbs, my first brand of LEDs would kill or interfere with all my remotes, including factory installed remotes in our vehicles.

arrisfe, 2024-02-20:
Hey @mariusmuja, yes, I do have an LED bulb on the board side of the GDO, but the other side with the battery still has a normal bulb in it. I've had this LED bulb in for a long while now and have not noticed any issues with my remotes. I installed my Ratgdo device at the end of January (LED has been in for a much longer time). I did see the post about the LED bulb and did remove it from the system temporarily, but that did not resolve the issues I was having when it was below 50 degrees. Thanks for bringing this up, actually, as now I will switch the bulbs as the LED doesn't generate much heat at all, so I'll need the regular bulb closer to the GDO board for the test tonight.

arrisfe, 2024-02-21:
@hellboyfishman, sure enough, this worked. This morning, the garage was in the low 40s, and pressing the query openings/status buttons were unresponsive. I turned on the light via the wall button and left it on for 10 minutes. The query openings/status buttons were returning responses again.

At this point, I need to decide whether to swap out the GDO board. I'm not left with a good feeling that I can completely trust the door's status at this point w/ratgdo, though, so I may have to stick with the MyQ app (unfortunately), as it's always 100% accurate.

[7 more replies]