ratgdo/esphome-ratgdo #175: Can't make RATGDO work

MishaelRosenthal, 2024-01-16:
Hi,

I bought the [2.5 installation kit](https://checkout.square.site/buy/X3QUEQGEQG4I6B24NC7EBHL3) and have a Chamberlain garage opener with a red button. 
I tried a few wiring schemes and couldn't make the RATGDO work.
It connects to the wifi and home assistant but does not have any effect on that garage opener. 

Am I missing anything?
Thanks in advance,

Mish

[image]
[image]

[image]
[image]

jerrm, 2024-01-16:
The release esphome firmware only supports security+ 2.0.

There is a test security+1.0 branch, or try the mqtt build that has 1.0 support.

MishaelRosenthal, 2024-01-16:
Hi, thanks,

Sorry for thew ignorance. What is security+ 2.0 or 1.0?
How do I try 1.0? I am currently not using mqtt in any of my installations, do I need to?

Thanks in advance,
Mish

MishaelRosenthal, 2024-01-16:
I succeeded in adding a mosquitto broker and MQTT to my home assistant installation.

Then I installed the MQTT v2.57 firmware with security 1.0 on the RATGDO:
https://paulwieland.github.io/ratgdo/flash.html

After that I only had to configure the MQTT server params in the web interface on http://192.168.4.180/.

Now it seems to work! 
It seems like the MQTT version is not as feature-rich as the ESPHome, will you release a corresponding ESPHome version?

Thanks in advance,
Mish

rlowens, 2024-01-16:
> will you release a corresponding ESPHome version?

In the works. See https://github.com/ratgdo/esphome-ratgdo/issues/128

smarthomeaf, 2024-01-20:
@MishaelRosenthal depending on what wall panel you have the sec1 works well.   looks like @mariusmuja has committed the changes and should be ready to go.

MishaelRosenthal, 2024-01-20:
Thanks @smarthomeaf and @mariusmuja .

Do I need to add the yaml code to the configuration.yaml file?

Thanks in advance,
Mish