ratgdo/esphome-ratgdo #176: Does not work with Chamberlain RollerLift CR655MYQ

anuragpathak21, 2024-01-16:
I am using Australian Chamberlain RollerLift CR655MYQ model.

Tried ESPHome ratgdo with ratgdo 2.5 version following Version 2.5i Security + 1/2 Wiring Diagram
but it doesn't work.

Below is the iwring diagram of the RollerLift

[image]

stevearmitage, 2024-01-17:
my RATGDO arrived yesterday (16-Jan)
I have the same issues connecting to my roller door (same model as above)

I have installed the latest ESPHome firmware
```
esphome
by ratgdo
Firmware: 2023.12.7 (Jan 17 2024, 11:18:45)
Hardware: 2.5i
```

I get the following notification in Home Assistant when the RATGDO reboots
```
GDO Workshop sync failed
Failed to communicate with garage opener on startup; Check the GDO Workshop Rolling code counter number entity history and set the entity to one number larger than the largest value in history.
```

The startup logs
```
11:20:43 | [D] | [number:012] | 'Rolling code counter': Sending state 40.000000
-- | -- | -- | --
11:20:44 | [D] | [api.connection:1121] | Home Assistant 2024.1.3 (192.168.67.4): Connected successfully
11:20:44 | [D] | [time:044] | Synchronized time: 2024-01-17 11:20:44
11:20:45 | [D] | [esp8266.preferences:238] | Saving preferences to flash...
11:20:49 | [D] | [number:012] | 'Rolling code counter': Sending state 41.000000
11:20:50 | [D] | [esp8266.preferences:238] | Saving preferences to flash...
11:20:57 | [D] | [number:012] | 'Rolling code counter': Sending state 42.000000
11:21:00 | [D] | [esp8266.preferences:238] | Saving preferences to flash...
11:21:10 | [D] | [ratgdo:545] | Triggering sync failed actions.
11:21:10 | [D] | [number:012] | 'Rolling code counter': Sending state 43.000000
11:21:15 | [D] | [esp8266.preferences:238] | Saving preferences to flash...
```

Here are how the device properties show in Home Assistant
[image]
[image]
[image]

Would appreciate any ideas.

PaulWieland, 2024-01-17:
This model door opener doesn't have a serial communication port. You have to use dry contact control with it.

JustDFB, 2024-01-24:
As in use the dry contract protocol that doesn't exist for EPSHome yet (coming soon)?
If so, any ETA please as it appears there are a number of us in the same situation....
It does support 2.0 - unsure if that is relevant.
Thanks

steveshacks, 2024-02-08:
Bump. I also have this door and have been unable to get it working using Security protocol (red & white wires).

wotacs, 2024-02-12:
> Bump. I also have this door and have been unable to get it working using Security protocol (red & white wires).

Paul commented above that the CR655MYQ Opener does not support Security protocol. 

We both need to use a dry contact to operate our doors. Bit of a shame, actually. I've just received my 2 units to control my doors. I guess my next step is going to be hacking the radio signal side of things, either the Remote to Motor (433.92Mhz) or preferably the Hub to Motor protocol, which may carry more information as it looks to be a bi-directional link.

PaulWieland, 2024-05-16:
Dry contact support has been implemented in the ESP Home firmware.

closed 2024-05-16 by PaulWieland

anuragpathak21, 2024-11-23:
> Dry contact support has been implemented in the ESP Home firmware.

What does that mean for my setup. Will it work using ratgdo 2.5 for my Australian Chamberlain RollerLift CR655MYQ model following the wiring diagram Version 2.5i Security + 1/2 Wiring Diagram 
or do I have to make any changes or add reed switch ? 

[image]

chriscollingwood613, 2025-04-19:
@mborawi , @anuragpathak21 Chamberlain didn't make the CR655MYQ units use the fancy secure serial protocol for local control, which means the only option for you (and me) is going to be to use the RATGDO to deliver a contact closure to the motor control port or learn how to hack the radio protocol that goes between the opener and the control hub. I started assembling the tools to try and hack the hub-motor protocols but ran out of mental energy before I could complete that task. Have a read of the documentation around the RATGDO, you need to put it into contact mode and also add door-closed indicator reed switches.