ratgdo/esphome-ratgdo #178: Ratgdo 2.5i with Chamberlain B1381 status values working, but no command functions. Serial logs included

FuzzyFilm, 2024-01-19:
Ratgdo 2.5i  board with current firmware build using with Chamberlain B1381 opener,  

The status values are working, but no command functions.   Tried with MQTT version as well and getting same lack of response of commands. 

below is a serial dump of logs.  in this i used the ESPHOME device website to go thru the commands and then did following ..  turn on light, open, close, lock, unlock and turn off light from the wall button unit (that has display with temperature on it)  

let me know if additional debugging is needed in verbose level.  Or a logs from a different firmware version would help 

I have tried full power cycle of opener, including backup battery disconnection.  One other step was I going to attempt to wipe all remote codes and start over but it has been to cold recently to go thru that process.  

`sdd��<�$�|�l�c|����r�c�c��og�doo���cp��lrlsdx�n��l��cn�|���c��no�l��l`�g'�do{���';r�`p�g�r������bo�|�c��'n��l �gg�lgs���osl`s��osl`��x�l`��o�$[I][logger:355]: Log initialized
[C][ota:473]: There have been 1 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[D][number:012]: 'Rolling code counter': Sending state 4126.000000
[D][ratgdo:320]: Set opening duration: 15.4s
[D][number:012]: 'Opening duration': Sending state 15.400000
[D][ratgdo:326]: Set closing duration: 14.3s
[D][number:012]: 'Closing duration': Sending state 14.300000
[D][number:012]: 'Client ID': Sending state 3827001.000000
[C][switch.gpio:011]: Setting up GPIO Switch 'Status door'...
[D][switch:016]: 'Status door' Turning OFF.
[D][switch:055]: 'Status door': Sending state OFF
[D][switch:016]: 'Status door' Turning OFF.
[C][switch.gpio:011]: Setting up GPIO Switch 'Status obstruction'...
[D][switch:016]: 'Status obstruction' Turning OFF.
[D][switch:055]: 'Status obstruction': Sending state OFF
[D][switch:016]: 'Status obstruction' Turning OFF.
[C][light:035]: Setting up light 'Light'...
[D][light:036]: 'Light' Setting:
[D][light:041]:   Color mode: 
[D][binary_sensor:034]: 'Motion': Sending initial state OFF
[D][binary_sensor:034]: 'Obstruction': Sending initial state OFF
[D][binary_sensor:034]: 'Button': Sending initial state OFF
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: AC:0B:FB:E7:6E:E0
[D][wifi:061]: Loaded saved wifi settings: nothsd
[D][wifi:455]: Starting scan...
[D][binary_sensor:034]: 'Dry contact close': Sending initial state OFF
[D][binary_sensor:034]: 'Dry contact open': Sending initial state OFF
[D][binary_sensor:034]: 'Dry contact light': Sending initial state OFF
[D][number:012]: 'Rolling code counter': Sending state 4127.000000
[D][number:012]: 'Rolling code counter': Sending state 4138.000000
[D][number:012]: 'Rolling code counter': Sending state 4139.000000
[D][number:012]: 'Rolling code counter': Sending state 4140.000000
[D][esp8266.preferences:238]: Saving preferences to flash...
[W][component:214]: Component preferences took a long time for an operation (0.06 s).
[W][component:215]: Components should block for at most 20-30ms.
[D][number:012]: 'Rolling code counter': Sending state 4141.000000
[D][wifi:470]: Found networks:
[I][wifi:513]: - 'nothsd' [redacted]▂▄▆█
[D][wifi:515]:     Channel: 5
[D][wifi:516]:     RSSI: -54 dB
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[D][wifi:518]: - [redacted] [redacted]▂▄▆█
[... 442 lines cut ...]
[D][lock:054]: 'Lock remotes': Sending state LOCKED
[D][number:012]: 'Rolling code counter': Sending state 4181.000000
[D][esp8266.preferences:238]: Saving preferences to flash...
[D][ratgdo:218]: Status: door=CLOSED light=ON lock=UNLOCKED learn=INACTIVE
[D][lock:054]: 'Lock remotes': Sending state UNLOCKED
[D][number:012]: 'Rolling code counter': Sending state 4182.000000
[D][esp8266.preferences:238]: Saving preferences to flash...
[D][ratgdo:232]: Light: action=TOGGLE state=OFF
[D][ratgdo:218]: Status: door=CLOSED light=OFF lock=LOCKED learn=INACTIVE
[D][lock:054]: 'Lock remotes': Sending state LOCKED
[D][number:012]: 'Rolling code counter': Sending state 4183.000000
[D][esp8266.preferences:238]: Saving preferences to flash...
[D][ratgdo:218]: Status: door=CLOSED light=OFF lock=UNLOCKED learn=INACTIVE
[D][lock:054]: 'Lock remotes': Sending state UNLOCKED
[D][number:012]: 'Rolling code counter': Sending state 4184.000000
[D][esp8266.preferences:238]: Saving preferences to flash...
[D][ratgdo:218]: Status: door=CLOSED light=OFF lock=UNLOCKED learn=INACTIVE
[D][ratgdo:218]: Status: door=CLOSED light=OFF lock=UNLOCKED learn=INACTIVE
[D][ratgdo:218]: Status: door=CLOSED light=OFF lock=UNLOCKED learn=INACTIVE
[D][ratgdo:218]: Status: door=CLOSED light=OFF lock=UNLOCKED learn=INACTIVE
[D][ratgdo:218]: Status: door=CLOSED light=OFF lock=UNLOCKED learn=INACTIVE
[D][ratgdo:218]: Status: door=CLOSED light=OFF lock=UNLOCKED learn=INACTIVE
[D][ratgdo:218]: Status: door=CLOSED light=OFF lock=UNLOCKED learn=INACTIVE
[D][ratgdo:218]: Status: door=CLOSED light=OFF lock=UNLOCKED learn=INACTIVE
[D][ratgdo:218]: Status: door=CLOSED light=OFF lock=UNLOCKED learn=INACTIVE`

jtrumpio, 2024-02-23:
I am having the exact same issue.

cgbale, 2024-03-08:
I'm also having the exact some issue with one of my two garage doors. The two garage doors are identical (Liftmaster 8500), and both have identically-configured ratgdo v2.5i hardware.

PaulWieland, 2025-07-07:
From the ratgdo Web UI, try changing the client ID and then click the Query Status button. Then try turning the light on/off. Capture the logs after the client ID is changed.