ratgdo/esphome-ratgdo #179: No status updates after a few hours (Chamberlain ML1000ev)

michaeljoos72, 2024-01-19:
Hi all

Recently I have connected a ratgdo board with ESPHome-FW to my Chamberlain ML1000ev Garagedoor opener (“yellow learn button” Security + 2.0). The board I have purchased directly from ratgdo (V2.5) and is powered over USB. WIFI is 3m away from board (-40 dBm).

Connection/Wiring is OK because in general everything works (open/close commands, light, obstruction)

My problem: After a few hours the 5 minutes status reporting stops. This screenshot was taken at 16:30:

[image]

No more updates and opening the garage door with a remote control also reported no change in status.

"Query status" helped and the status was updated immediately:

[image]

Do you know what could be the root cause for this? Does the Garagedoor opener somehow have a ndeep sleep? And "query status" does wake-up?

Thanks & BR
Michael

tmjo, 2024-05-06:
Hi,

I think I see the same behavior as you report here. My setup is a Liftmaster LM100EVS with an original Ratgdo v2.5i flashed with latest esphome-version. Everything working fine (except close-command due to not having obstruction sensors ref [here](https://github.com/ratgdo/esphome-ratgdo/issues/146#issuecomment-1886904069), but toggle works perfectly).

I get correct status for a while, and then it stops updating. Seems like toggle-command from webUI or HA still works, and then status update is back again. A query status command also get things back on track.

Any ideas why this might happen?

damaar666, 2024-07-01:
> Hi,
>
> I think I see the same behavior as you report here. My setup is a Liftmaster LM100EVS with an original Ratgdo v2.5i flashed with latest esphome-version. Everything working fine (except [...]

This is similar to my experience,  with the obstruction sensors.

tmjo, 2024-07-05:
@PaulWieland; would you mind taking a look and see if you have any idea that might be causing this or how it could be fixed? Thanks for a great product!

kriss2938, 2024-08-10:
Here the same issue, for a long time, already. Even removed the RATGDO a while ago because it's unreliable.
Recently started using the device and did some troubleshooting, but in vain.

The RATGDO is connected to a Chamberlain ML700ev and after some hours it stops reporting door state.
- Pressing "Query status" and/or "Sync" doesn't help in my case.
- Restarting the device is the only way to get it working.

Troubleshooting:
- At first I didn't have any obstruction sensors connected. Currently, the sensors are connected, but still the same issue.
- Moved the RATGDO closer to a Wifi AP to rule out bad signal. Wifi signal is around -40db, so very good, but issue persists.
- Re-flashed RATGDO with latests firmware and erase option enabled, but issue persists.

Can we enable some kind of extended logging and download it from the device?

PaulWieland, 2024-08-10:
> Can we enable some kind of extended logging and download it from the device?

Open the web UI, the serial log is displayed there. A serial log is required in order to do any sort of meaningful troubleshooting.

Does the door respond to commands? What happens in the log when you press the sync button? Sync issues a query to the GDO, and the GDO should respond with the status. The status is also normally updated every 5 minutes by the GDO, but it might be different for these ML series openers.

PaulWieland, 2024-08-10:
> Do you know what could be the root cause for this? Does the Garagedoor opener somehow have a deep sleep? And "query status" does wake-up?

Could be a fundamental difference in this series of opener. Does this model have an e-serial terminal? 
If it responds to the sync/query command, then a simple fix might be to add a timeout after 10 minutes of no status, issue a status query.

Regardless, the polling shouldn't matter. The door should broadcast its status when you open/close it. Is that also not happening?

michaeljoos72, 2024-08-11:
_Does this model have an e-serial terminal?_
No, on my ML1000ev I only can see the pins that I have used for ratgdo.

_Does the door respond to commands?_
Yes, the commands always worked, and then the status immediately worked again for a few hours.

For me sending a "query status" every hour from openHAB does work as a workaround. 

_The status is also normally updated every 5 minutes by the GDO_
Yes, this is visible in the LOG. But somehow, it stops after a few hours standby without any activities.

_If it responds to the sync/query command, then a simple fix might be to add a timeout after 10 minutes of no status, issue a status query._
I think this would resolve my issue. Would be great if you can implement! :)

tmjo, 2024-08-12:
Hi and thanks for your reply! To my understanding, my Liftmaster LM100EVS has a e-serial (using terminal 0 instead of the red one).

[image]

Apart from that, the situation is the same as for @michaeljoos72. It seems to always respond to commands, and seems to update status correctly when controlled from ratgdo. However, it does not always catch status changes if I have used the "old" controls provided with the garage door opener. A sync/query command seems to solve it and I see this also in the opening counts, which may suddenly jump 3-5 numbers since the door has been opened in the traditional way without updating status to Home Assistant.

kriss2938, 2024-08-17:
Here's a screenshot of the communication stopping between RATGDO and GDO: 

[image]

&nbsp;
&nbsp;

> Does the door respond to commands? What happens in the log when you press the sync button? Sync issues a query to the GDO, and the GDO should respond with the status. The status is also normally updated every 5 minutes by the GDO, but it might be different for these ML series openers.

At first I thought it doesn't respond to any commands, but it seems it just takes some time (like 10 seconds) to respond:

[image]

&nbsp;
&nbsp;

Seems like my symptoms are the same as for @michaeljoos72 

I have created an automation in HA to trigger a "Query status" every hour. Will monitor this for a week to see if state changes of the GDO are being reported.

digitalmacdonald, 2024-08-31:
> Here's a screenshot of the communication stopping between RATGDO and GDO:
> 
> ![Screenshot from 2024-08-14 09-16-27](https://private-user-images.githubusercontent.com/178040080/358832812-5c5e9b7a-1f72-4885-bc2a-adeea309fae2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUwODY3MzksIm5iZiI6MTcyNTA4NjQzOSwicGF0aCI6Ii8xNzgwNDAwODAvMzU4ODMyODEyLTVjNWU5YjdhLTFmNzItNDg4NS1iYzJhLWFkZWVhMzA5ZmFlMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODMxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgzMVQwNjQwMzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iOWI2N2Y5MmI5ZWFjMGIxMDRmMmU2NGYxM2E1MDVjMWYyZjMzYjAxYzBjYzhjZTI2ZTkwMTYwN2M2OGJkMTM1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.SZdYuOJLofLR-C__FB5UZnwK2icT9TldZ-Byrbb7GQw)
> 
>    
> 
> > Does the door respond to commands? What happens in the log when you press the sync button? Sync issues a query to the GDO, and the GDO should respond with the status. The status is also normally updated every 5 minutes by the GDO, but it might be different for these ML series openers.
> 
> At first I thought it doesn't respond to any commands, but it seems it just takes some time (like 10 seconds) to respond:
> 
> <img alt="Scherm­afbeelding 2024-08-15 om 20 31 44" width="1426" src="https://private-user-images.githubusercontent.com/178040080/358832894-8f55c43f-d228-41cc-929e-20eefa907533.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUwODY3MzksIm5iZiI6MTcyNTA4NjQzOSwicGF0aCI6Ii8xNzgwNDAwODAvMzU4ODMyODk0LThmNTVjNDNmLWQyMjgtNDFjYy05MjllLTIwZWVmYTkwNzUzMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODMxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgzMVQwNjQwMzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02MmI4NzM5MzNlNjZmNDkyZWVkOGM5YTFiZWVlMzAwZTAzZDUwZDU0NWZlODBkODYwM2FiOWQ0YjI4ZDVhMTMyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.TRayQPRbLLSkF5QLFparxudiMDN9byVQrOJXaS0hW1c">
>    
> 
> Seems like my symptoms are the same as for @michaeljoos72
> 
> I have created an automation in HA to trigger a "Query status" every hour. Will monitor this for a week to see if state changes of the GDO are being reported.

For what it's worth, I have exactly the same issue and could produce identical screenshots of the log. Merlin (Chamberlain) MS105MYQ. Yellow Button.
[image]

Happy to provide any supporting logs or info if it helps the fault finding.

sjwright, 2026-04-03:
FYI, the same symptom occurs with an Australian-market Merlin MS105MYQ and the Konnected blaQ device. I have two of those motors and two Konnected blaQ devices. I think something more fundamental is going on, whether that is (a) an incomplete implementation of the wire protocol, such as an undocumented command that tells client devices to resynchronise, or (b) an electrically marginal serial connection that is not resilient enough 0.001% of the time. But I am a complete novice when it comes to wire protocols and data circuits, so I have absolutely no idea.

What I _have_ done successfully is modify Konnected’s version of gdo.c to run get_status whenever it receives a bad packet. See here:

https://community.konnected.io/t/aus-merlin-ms105myq-blaq-obstruction-sensor-not-reading-in-home-assistant/40459/5

https://github.com/sjwright/gdolib/tree/simplefix

I also wrote some more elaborate code (fancyfix) that wedges in a lot of extra logging so I could observe the connection going bad. In my YAML, I rerouted all logs to a syslog-ng server, which gave me a full history of all TX and RX activity. It tries to realign incoming data, but I am not sure how successful that was. **Running get_status whenever you receive a bad packet has proven to be a reliable solution.** This was based on the insight that turning the light on and off seemed to rejuvenate a bad connection. It turns out that what makes light on/off special in the code is that it also runs get_status, whereas most other commands, such as open/close, do not.

PaulWieland, 2026-04-15:
@sjwright your post was edited to stay on topic. 

A complete rewrite of the serial interface was released which reduces the occurrence of corrupted packets. I had a theory that the corrupted packets were causing the GDO to fall out of digital communication mode, similar to how security + 1 openers fall back to analog mode, but that doesn't seem to be the case based on @prikril's feedback.

I'm not actually sure that the packet being sent is corrupt, it may simply be mis-understood by the library.