ratgdo/esphome-ratgdo #180: Apparent issue with esphome web installer.

PedanticAvenger, 2024-01-20:
Just received my 2.5i board and went to flash it with the Esphome image.
Select board and type (Sec+1.0) and click connect, select COM port and then error "Failed to download manifest" pops up.
Reproduceable on Edge/Chrome/Firefox and my previous unit (before I realized I didn't have a Sec+2.0 opener) flashed fine.

Something change on the page recently?

PedanticAvenger, 2024-01-20:
To update, selecting 2.5i w/ Security+2.0 as an option proceeds properly to flashing without an error.

bercribe, 2024-01-20:
I just tried the same thing, having the exact same issue

bdraco, 2024-01-20:
Try clearing your browser cache. The firmware files for the 2.5i and 2.5 board should be up there now

PedanticAvenger, 2024-01-20:
Confirmed.
Thanks!

closed 2024-01-20 by bdraco