ratgdo/esphome-ratgdo #186: Unable to update config on esphome flashed 2.5i board w/ Sec+1.0

PedanticAvenger, 2024-01-20:
Flashm, connection to wireless and adoption into ESPHome dashboard works just fine.  
Device shows as online and all sensors show up in HA.

When trying to rename the device the upload attempt fails with the following log output:
```
INFO Reading configuration /config/esphome/ratgdov25i-86b4b5.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
ERROR Unable to load component ratgdo:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/74f605c3/components/ratgdo/__init__.py", line 49, in <module>
    cv.none, pins.gpio_input_pin_schema
AttributeError: module 'esphome.config_validation' has no attribute 'none'
ERROR Unable to load component ratgdo.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/74f605c3/components/ratgdo/__init__.py", line 49, in <module>
    cv.none, pins.gpio_input_pin_schema
AttributeError: module 'esphome.config_validation' has no attribute 'none'
ERROR Unable to load component ratgdo.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/74f605c3/components/ratgdo/__init__.py", line 49, in <module>
    cv.none, pins.gpio_input_pin_schema
AttributeError: module 'esphome.config_validation' has no attribute 'none'
ERROR Unable to load component ratgdo.lock:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
[... 351 lines cut ...]
  type: client_id
  entity_category: config
  ratgdo_id: ratgdov25i
  name: Client ID
  mode: box
cover.ratgdo: [source /config/esphome/.esphome/packages/74f605c3/base.yaml:210]
  
  Platform not found: 'cover.ratgdo'.
  platform: ratgdo
  id: ratgdov25i_garage_door
  device_class: garage
  name: Door
  ratgdo_id: ratgdov25i
  on_closed: 
    - switch.turn_off: ratgdov25i_status_door
  on_open: 
    - switch.turn_on: ratgdov25i_status_door
light.ratgdo: [source /config/esphome/.esphome/packages/74f605c3/base.yaml:221]
  
  Platform not found: 'light.ratgdo'.
  platform: ratgdo
  id: ratgdov25i_light
  name: Light
  ratgdo_id: ratgdov25i
```

bdraco, 2024-01-20:
You're running an old version of the dashboard, as new dashboard images are based on python, 3.11

PedanticAvenger, 2024-01-20:
Argh. Thanks. Been running Dev of dashboard and thought it would stay ahead but something slipped. Reverted to current regular release and life was good.

closed 2024-01-20 by PedanticAvenger