ratgdo/esphome-ratgdo #187: OTA firmware installation MQTT to ESPHome?

FrankJaco, 2024-01-20:
I am currently running MQTT firmware 2.57. I have a Sec1.0+ purple learn button Chamberlain HD9100D with a 041A6318 wall panel which I have isolated via a switch due to the issues it causes when it is connected with the ratgdo.  I would like to test the ESPHome firmware now that so much effort was put into it to make it work with Sec1+ to see if works better/worse or the same as the MQTT. I have used the MQTT OTA firmware updater successfully a few times by downloading the bin and using the web interface. This works reliable.  

Here comes the punchline...

As I didn't have any brains or foresight, I have made it exceptionally difficult to get to the ratdgo directly and hook up a USB cable with data-support. Is there any way to use the MQTT OTA firmware updater to load an ESPHome binary so that I can get it up and running without the need to connect it to a data cable? e.g. push an ESPHome bin with enough info I can grab wirelessly to finish any config?

possible? or $h!+ out of luck?

fj

ipha, 2024-01-20:
You can load the esphome firmware through the mqtt web updater.

After you load it and it boots, it will broadcast an open ssid which you'll connect to and configure the wifi credentials.

FrankJaco, 2024-01-21:
Thanks for the quick response.

I guess a bin is a bin is a bin?  Glad it should work.  It is my hope that ESPHome works at least as well as MQTT, so I least I can capture some good logs via ESPHome's OTA logging (which the MQTT version cannot not yet do).

If by chance I need to switch BACK to MQTT from ESPHome.....   Is there a way to do that OTA?

fj

D34DC3N73R, 2024-01-23:
You can also build the bin with your wifi creds so it'll just be on your network when it boots back up. If you use DHCP reservations in your router, it'll be at the same IP. And yes, there's an OTA file upload in the ESPHome web interface as well.

thomasshawn, 2024-01-27:
> You can also build the bin with your wifi creds so it'll just be on your network when it boots back up. If you use DHCP reservations in your router, it'll be at the same IP. And yes, there's an OTA file upload in the ESPHome web interface as well.

How do we build a bin? I understand building a yaml to upload once on esp already but never built a bin without a compiler and all the files.

I ask because I'm on mqtt and only reason I might chnage to esp is so I can tweak my debounce numbers. If I can chnage these in the config and then make my own bin to update I absolutely would. 

Otherwise I'll switch to esp version on this particular door.

D34DC3N73R, 2024-01-28:
I have esphome running in a docker container so I used that. I'm sure there are other ways to accomplish this, but this is how I did it. 

1. In ESPHome click "new device" add a name and click next
2. For Installation, "skip this step"
3. Choose any board (we're changing the yaml)
4. Copy the encryption key to be safe (can't remember if I needed this or not for HA integration), choose skip
5. Click edit for the offline device
6. Copy the right yaml for your board from this repo (for instance [static/v25iboard_secplusv1.yaml](https://raw.githubusercontent.com/ratgdo/esphome-ratgdo/main/static/v25iboard_secplusv1.yaml) if you have the 2.5i board)
7. If you have secrets.yaml from other projects you can use the following, otherwise change the ssid and password to actual values
```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```
8. Click save, then "install" and choose manual download.
9. You'll see it build and ratgdov25i.bin will download when it's complete. 

I'm pretty sure I later adopted the device into ESPHome after I confirmed it was up and running. At that point you'll get an encryption key for the HA integration.

thomasshawn, 2024-01-28:
Thanks.

That's my normal routine as well but was thinking you were discussing compiling your own bin file, in which case I might be able to steal a few steps to modify the mqtt version.

I have it working on multiple doors with mqtt however 2 of my doors are dry contact push button only. I've added reed switches but the bounce is Killing me Paul has already adjusted the bounce up to 300ms but I think K I personally need more and I can't really expect him to keep making changes.

I had requested a line on the mqtt webui that would allow individual de-bounce settings and he seemed to like the idea but I know he's busy.

Moving to esp version should hopefully solve my issue.

D34DC3N73R, 2024-01-28:
I think they are still working out some bugs for esphome dry contact #130

FeatherKing, 2024-02-05:
just to add, I was running 2.57 MQTT and I could not flash the Esphome builds. It would fail for not enough space on the ratgdo.

I went into Esphome and created a very basic build with no switches or anything, just the web interface and wifi credentials. i called that ratgdo-base and flashed that. Then i was able to upgrade to a full esphome build using the files from this repo

PHLAK, 2024-04-12:
> You can load the esphome firmware through the mqtt web updater.

How does one access the "mqtt web updater"?

thomasshawn, 2024-04-12:
> > You can load the esphome firmware through the mqtt web updater.
> 
> How does one access the "mqtt web updater"?

Open via the ip address

PHLAK, 2024-04-12:
> Open via the ip address

@thomasshawn I don't see any way to do an update when I do.

<p align="center">
  [image]
</p>

PaulWieland, 2024-04-12:
You are on an old version of the MQTT firmware that pre-dates OTA updates.

PHLAK, 2024-04-12:
> You are on an old version of the MQTT firmware that pre-dates OTA updates.

Okay, thank you. I guess I'll have to take it down and flash it manually then.

timespaced, 2024-04-23:
> You can load the esphome firmware through the mqtt web updater.
> 
> After you load it and it boots, it will broadcast an open ssid which you'll connect to and configure the wifi credentials.

Where is the ESPHome firmware?  I found the web installer site, but the bin's ar enot available to download there.  I can't find them on this github page, but they must be here.  I'm looking to OTA update from MQTT to ESPHome like OP.

timespaced, 2024-04-23:
> You can load the esphome firmware through the mqtt web updater.
> 
> After you load it and it boots, it will broadcast an open ssid which you'll connect to and configure the wifi credentials.

I did this - flashed a generic ESPHome.  The ratgdo upgraded from MQTT to ESPHome, and broadcasts its own wifi.  I connected to that, and hit the IP of the ratgdo - I see garage controls and a scrolling log, but no place to change the wifi for the ratgdo to connect to...