ratgdo/esphome-ratgdo #188: "Light" switch seems to throw the physical "Lock" button out of sync

jshatch, 2024-01-20:
Everything seems to work fine functionally with the garage door opener itself, but when I use the ratgdo to toggle the light it throws the "lock" function out of sync with the wall control.  I'll do my best to explain this behavior.

Steps to reproduce:
1) Use the wall control to toggle the lock (make the LED blink)
2) Use the ratgdo to perform some functions (except the light) and verify function
3) Use the ratgdo to lock / unlock the remotes, verify the indicator blinks on the wall control
4) Use the ratgdo to toggle the light
5) Use the ratgdo to lock / unlock the remotes, see that the LED no longer blinks, the lock state on the ratgdo does change and remotes are verified to be locked by testing an actual remote

- The gdo and wall button are now out of sync

6) Try to toggle the lock via the wall control, it will fail until you long press it a total of 3 times after which time the lock function will toggle successfully and the wall control will again be in sync
7) Toggle the light using the ratgdo and you will again be in the out of sync state in step 5

Note: Toggling the light using the wall control does not trigger this problem.

It seems that maybe something gets weird with the rolling code that the wall control is sending, but why that would only affect the lock function and not the open / light function is where I am at a loss.  Are there two separate rolling codes on these wall controllers?

I'm unsure if this is something that can even be addressed with a firmware change, but I do want to know if others see this behavior.  The wall button blinking should be an indicator of the lock state and right now that is not reliable.

I have 2 identical Liftmaster GDO's (non-wifi) and 2 identical ratgdo 2.5i boards wired and configured the same.  Using commit 727759e and ESPHome 2023.12.8

mwaterbu, 2024-05-16:
I am seeing this exact behavior after updating (from 2023.8.2), with the exception that the lock does **not** work for me under any condition.
After updating, the rolling code on my ratgdo reset, perhaps this is related to this sync issue?

mwaterbu, 2024-05-16:
After testing this a bit more, I noticed that the wall control will resync itself after some amount of time, but so far it has been variable (less than 15 minutes to over an hour). Really curious what is going on here.