ratgdo/esphome-ratgdo #192: Configuration Option to Disable Opening Door

brgaulin, 2024-01-22:
As a layer of security, I would like to block any "open" "position" "stop" or "toggle" commands from ever being issued from api sources(Home Assistant). 

My main use case for ratgdo is to report the current status, and to issue a "close" command if I've accidentally left a door open. I do not anticipate ever needing to remotely open the door, and would like to disable this feature. 

I understand coding in a lock like this is not a foolproof way to secure my door, but I fear the day my home assistant gets hacked and someone starts opening and closing my door, wasting my power.

At the moment I am exploring template covers to proxy the default cover(now made internal). ex:
```yaml
cover:
  - id: !extend ${id_prefix}_garage_door
    internal: true
    name: "Internal Door"
  - platform: template
    id: ${id_prefix}_exposed_door
    name: Door
    device_class: garage
    lambda: |-
      return id(${id_prefix}_garage_door).position;
    open_action:
      - logger.log: 
          format: "Remote command recieved to open the door. Intentionally refusing the command."
          level: "WARN"
    close_action:
      - cover.close: ${id_prefix}_garage_door
```

This code somewhat works, but I am missing the OPENING/CLOSING states, and I suspect the template is doing more work in its loop than it needs to. 

I am open to writing a PR to enable this functionality as a configuration option. Before I go down that road I was hoping for a discussion on if this is a good idea or if maybe I have overlooked something.

Potentially related to #138 

Potentially relevant sections of code:
https://github.com/ratgdo/esphome-ratgdo/blob/f3f8f96/components/ratgdo/cover/ratgdo_cover.cpp#L73
or in these functions: https://github.com/ratgdo/esphome-ratgdo/blob/f3f8f96/components/ratgdo/ratgdo.cpp#L423

Thank you for any advice you may have.

brgaulin, 2024-01-22:
A working example of how to achieve my desired state today with opening/closing. These lines could be added to any configuration to modify the base `cover` object.

```yaml
# Create a proxy cover. Only allow closing/stopping the door. Do not allow any actions that could open the door.
cover:
  - id: !extend ${id_prefix}_garage_door
    internal: true
    name: "Internal Door"
    on_opening:
      - wait_until:
          binary_sensor.is_on: ${id_prefix}_motor
      - cover.template.publish:
          id: ${id_prefix}_exposed_door
          current_operation: !lambda return COVER_OPERATION_OPENING;
      - wait_until:
          binary_sensor.is_off: ${id_prefix}_motor
      - cover.template.publish:
          id: ${id_prefix}_exposed_door
          current_operation: !lambda return COVER_OPERATION_IDLE;
    on_closing:
      - wait_until:
          binary_sensor.is_on: ${id_prefix}_motor
      - cover.template.publish:
          id: ${id_prefix}_exposed_door
          current_operation: !lambda return COVER_OPERATION_CLOSING;
      - wait_until:
          binary_sensor.is_off: ${id_prefix}_motor
      - cover.template.publish:
          id: ${id_prefix}_exposed_door
          current_operation: !lambda return COVER_OPERATION_IDLE;
  - platform: template
    id: ${id_prefix}_exposed_door
    name: Door
    device_class: garage
    has_position: true
    lambda: |-
      return id(${id_prefix}_garage_door).position;
    close_action:
      - if:
          condition:
            lambda: 'return id(${id_prefix}_garage_door).position != COVER_CLOSED;'
          then:
            - cover.close: ${id_prefix}_garage_door
    open_action:
      - logger.log: 
          format: "Remote command recieved to open the door. Intentionally refusing the command."
          level: "WARN"
    position_action:
      - logger.log: 
          format: "Remote command recieved to change door position. Intentionally refusing the command."
          level: "WARN"
```

brgaulin, 2024-01-23:
That is an epic idea!

I started to take a look at the structure of this repo, and my head is spinning a bit. Before I start thinking about a PR / testing this out(if that is the way we should do this), I had a few questions.

I see that there is a single place that exposes `door_state` and `position` changes today for the ratgdo cover: https://github.com/ratgdo/esphome-ratgdo/blob/main/components/ratgdo/cover/ratgdo_cover.cpp#L24

I'm wondering if it would be best to copy the ratgdo cover, and have a combined trigger that fires when door_state or position changes. Maybe call it `on_door_state_change` to keep it similar to the existing code?

Here would be an example of how that could look:
```yaml
cover:
  - id: !extend ${id_prefix}_garage_door
    internal: true
    name: "Internal Door"
    on_door_state_change:
      - cover.template.publish:
          id: ${id_prefix}_exposed_door
          current_operation: !lambda return id(${id_prefix}_garage_door).current_operation;
          position: !lambda return id(${id_prefix}_garage_door).position;
  - platform: template
    id: ${id_prefix}_exposed_door
    name: Door
    device_class: garage
    has_position: true
    close_action:
      - cover.close: ${id_prefix}_garage_door
    ...
```

brgaulin, 2024-01-23:
Or maybe I'm over complicating, and this could be done the same way the other triggers via `Cover::add_on_state_callback`. In this new trigger, there could be no filtering and all events could be returned.

mariusmuja, 2024-01-23:
> Or maybe I'm over complicating, and this could be done the same way the other triggers via `Cover::add_on_state_callback`. In this new trigger, there could be no filtering and all events could be returned.

Yes, you can add a trigger to `automation.h` that uses `RATGDOComponent::subscribe_door_state` and fires and event for each new state & position.

brgaulin, 2024-01-23:
I think I got it working. I gave it a test run with my garage door, and the state of the template cover mirrored the ratgdo cover exactly!

PR is linked to the issue

PR #195 (2024-07-03): feat: add cover.on_state_change trigger
fixes #192

This PR exposes a new automation trigger `on_state_change`. This new trigger occurs anytime `Cover::publish_state` is called. 

Example use case: 
A cover that reports status, and allows closing only. The template cover below can not be opened by external API calls. This satisfies the use case of closing the garage door remotely.

Example without base.yaml:
```yaml
cover:
  - platform: ratgdo
    id: ${id_prefix}_garage_door
    device_class: garage
    name: "Internal Door"
    internal: true
    ratgdo_id: ${id_prefix}
    on_state_change:
 [... cut]
closed 2024-07-03 by bdraco