ratgdo/esphome-ratgdo #193: Flashing on Linux

Linuturk, 2024-01-22:
Any tips for flashing the ESPHome firmware on Linux? The web based flashing tool isn't finding a tty with the ratgdo attached. I've checked my dmesg output, and it seems the serial briefly connects and then disconnects on ttyUSB0. 

```
[1212742.091208] usb 5-3: new full-speed USB device number 42 using xhci_hcd
[1212742.256553] usb 5-3: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice=81.34
[1212742.256560] usb 5-3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[1212742.256563] usb 5-3: Product: USB Serial
[1212742.261622] ch341 5-3:1.0: ch341-uart converter detected
[1212742.275627] usb 5-3: ch341-uart converter now attached to ttyUSB0
[1212742.842034] input: BRLTTY 6.4 Linux Screen Driver Keyboard as /devices/virtual/input/input68
[1212742.855559] usb 5-3: usbfs: interface 0 claimed by ch341 while 'brltty' sets config #1
[1212742.858677] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[1212742.858708] ch341 5-3:1.0: device disconnected
```

From what I understand the driver for this chipset has been in the kernel since 2.x so I'm assuming it should be working.

I am directly connected to my computer, no USB hub. And I've tried multiple USB cables.

PaulWieland, 2024-01-22:
Have you ruled out a permission issue? chrome normally doesn't have access to the serial ports by default.

Linuturk, 2024-01-22:
The other tty devices are showing in the list, just not ttyUSB0. I think it has something to do with the device almost instantly disconnecting.

Just to make sure though, how would I ensure permissions are right for Chrome to access the serial ports?

mariusmuja, 2024-01-22:
@Linuturk 

I think this is the problem:
```
> [1212742.855559] usb 5-3: usbfs: interface 0 claimed by ch341 while 'brltty' sets config #1
```

The first answer here should solve your issue: https://unix.stackexchange.com/questions/670636/unable-to-use-usb-dongle-based-on-usb-serial-converter-chip

Linuturk, 2024-01-22:
Awesome, thanks for finding that. I removed the service with `sudo apt remove brltty` and now I can see the tty in the chrome prompt.

closed 2024-01-22 by Linuturk

Linuturk, 2024-01-22:
For anyone coming in after the fact, make sure your user is part of the `dialout` group:

`sudo usermod -a -G dialout myuser`