ratgdo/esphome-ratgdo #202: Connecting more than one pair of obstruction sensors (Merlin Liftmaster)

Daz42, 2024-01-24:
Connecting more than one pair of obstruction sensors (Merlin Liftmaster). I have two pairs of obstruction sensors (genuine Chamberlain). If one pair is connected to the ratgdo then the sensors work with the ratgdo. If I connect both pairs the sensor does not report an obstruction any longer. It could be that the voltage tolerance needs to be adjusted to account for more than one set of obstruction sensors? I need two pairs due to the size of one my my SUVs. Any thoughts?

PaulWieland, 2024-01-24:
Do you have a manual that shows multiple sets of sensors being connected?

Daz42, 2024-01-24:
Hey Paul. I don't have any documentation to support this connectivity. However Merlin New Zealand were the installers of the system and said that they support 2 pairs of sensors in the installation and it was a simple matter of combining the + and - wires to the sensor contacts on the GDO. They work natively, as in, they stop the garage door from closing if obstructed. But the ratgdo stops sensing an obstruction while the gdo still does.

PaulWieland, 2024-01-24:
If you own a multimeter can you check the voltage of the line with two sets of sensors connected?

Daz42, 2024-01-24:
I do have one, here are the results:

One pair of sensors connected to the GDO:

5.28v constant (when unobstructed)
5.59v constant (when obstructed)

Two pair of sensors connected to the GDO:

5.08v - reading flicks between 5.08v and 5.09v (when unobstructed)
5.15v (mostly) - reading flicks between 5.10v and 5.20v (when obstructed)

Merlin NZ did say that fitting two pairs of sensors is quite common especially with commercial installations where they are dealing with small trucks etc.

PaulWieland, 2024-01-24:
That voltage should be okay so I'm wondering if the pulse frequency changes. I will have to rob a set of sensors off of another door so I can scope what its doing.

Daz42, 2024-01-24:
No worries. I am also quite technical so if there is anything I can do to help let me know.

Daz42, 2024-02-04:
> That voltage should be okay so I'm wondering if the pulse frequency changes. I will have to rob a set of sensors off of another door so I can scope what its doing.

Hey Paul, did you get anywhere with the sensor issue?

Daz42, 2024-02-14:
Hey @PaulWieland , sorry to bump but did you get anywhere with the sensor issue?

TesLocker, 2024-02-26:
I contacted Chamberlain’s US online support regarding this. They said that their door openers do not support multiple sets of obstruction sensors. Ratdgo support would be great, but it isn’t clear to me that it is natively supported.

Daz42, 2024-02-27:
I have two double garage doors fitted with Merlin Commanders - which are a Chamberlain product made for the AU/NZ region. The installers were Merlin themselves who fitted two sets of sensors per door. They do support them here in New Zealand.

Daz42, 2024-03-13:
Hello, I am going to look at another product to detect an obstruction at the garage door. I appreciate that in regions outside of NZ/AU Chamberlain don't support two pairs of obstruction sensors but there must be a way of supporting it?

For now, I need to get the obstruction status in through HA so I can be sure that the door is safe to close (no children in the way, bikes and a tailgate open) and also so I can mount a light at the back of the garage to show the door is cleared before pulling up the car handbrake! Thanks @PaulWieland, everything else works great but if you get around to the obstruction sensor support at some stage that would be great.