ratgdo/esphome-ratgdo #203: OTA binaries?

Guy559, 2024-01-24:
Suspect this is a noob question but: is there a folder or easily accessible location with OTA binaries?

Built (3) RatGDO devices using kaldek-ratgdo design and ESP8266s. Flashed with web tool about 2-3 months ago and working fine but suspect that modifications continue to be made. Would appreciate ability to update firmware to latest w/o removing boards to reflash via wired connection and Web Tool.

Thanks in advance.

rlowens, 2024-01-25:
But if you want the generic firmware they are at https://ratgdo.github.io/esphome-ratgdo/ratgdov25i-esp8266/ratgdov25i-esp8266.bin etc. Have to look in https://ratgdo.github.io/esphome-ratgdo/v25iboard-manifest.json etc to see the exact path (where "v25iboard" is the hardware name selected in HTML in view-source:https://ratgdo.github.io/esphome-ratgdo/

closed 2024-01-26 by PaulWieland

etho201, 2024-02-07:
@rlowens Can you please provide more info on how to compile the firmware in the ESPHome Addon in HA? I'm looking at the addon and see nothing about compiling or updating the firmware. My firmware is currently at `2023.12.3 (Dec 25 2023, 04:18:34)`.

etho201, 2024-04-15:
Thanks @rlowens! I am now managing my device using the ESPHome Dashboard and was able to successfully compile the firmware.