ratgdo/esphome-ratgdo #205: Feature Request: Dry Contact Virtual Limit Switch

Seechay, 2024-01-25:
Hello! I would like to request the ability to supply Home Assistant and/or MQTT binary sensors to use for limit switches instead of being wired to the pins. This could be used to replace the open/close triggers to any arbitrary sensor(s). In my use case I have two zigbee contact switches being used to monitor the garage door state, as wiring would be overly complex. There are also wireless tilt sensors that could be used, which would be able to provide binary triggers.

mariusmuja, 2024-01-25:
Give this branch a try: https://github.com/ratgdo/esphome-ratgdo/tree/dry_contact_with_sensors

Add the home assistant sensors to the `ratgdo` component like this:

```
ratgdo:
  id: ${id_prefix}
...
  protocol: drycontact
  door_open: ${id_prefix}_door_open
  door_closed: ${id_prefix}_door_closed
...
  ```

where the sensors are some homeassistant binary sensors (for example I've tested with some zigbee hall sensors):
```
  - platform: homeassistant
    name: "Door Open Sensor"
    id: ${id_prefix}_door_open
    entity_id: binary_sensor.door_sensor_open   # entity_id from home assistant
    filters: 
      - invert
 
  - platform: homeassistant
    name: "Door Closed Sensor"
    id: ${id_prefix}_door_closed
    entity_id: binary_sensor.door_sensor_closed     # entity_id from home assistant
    filters: 
      - invert
```

Seechay, 2024-01-25:
That does exactly what I want! There are a few sync issues when involving the wall panel, but I've seen other issues posted about that, so I don't believe it's an issue with this implementation.

mariusmuja, 2024-01-25:
Great, thanks for being the first tester of this new feature.

Under what conditions do you see the sync issues? Things like stopping the door from the wall panel while it's opening/closing?

Seechay, 2024-01-25:
Yes, that's exactly it. There was also a time when I stopped it from ratgdo and the state got confused and I had to press the opposite direction to get the door to move. It also had the status as inverted until I did a full transition again. I had logs running while experimenting, although I forgot to note at what time I performed which action and whether it was from HA or the wall panel.

```
INFO Starting log output from 192.168.30.38 using esphome API
INFO Successfully connected to garage_door_controller @ 192.168.30.38 in 31.893s
INFO Successful handshake with garage_door_controller @ 192.168.30.38 in 2.970s
[21:54:13][I][app:102]: ESPHome version 2023.12.9 compiled on Jan 24 2024, 21:53:20
[21:54:13][I][app:104]: Project ratgdo.esphome version 2.5i
[21:54:13][C][wifi:573]: WiFi:
[21:54:13][C][wifi:405]:   Local MAC: C8:2B:96:30:F5:D5
[21:54:13][C][wifi:410]:   SSID: [redacted]
[21:54:13][C][wifi:411]:   IP Address: 192.168.30.38
[21:54:13][C][wifi:412]:   BSSID: [redacted]
[21:54:13][C][wifi:414]:   Hostname: 'garage_door_controller'
[21:54:13][C][wifi:416]:   Signal strength: -52 dB ▂▄▆█
[21:54:13][C][wifi:420]:   Channel: 11
[21:54:13][C][wifi:421]:   Subnet: 255.255.255.0
[21:54:13][C][wifi:422]:   Gateway: 192.168.30.1
[21:54:13][C][wifi:423]:   DNS1: 192.168.2.69
[21:54:13][C][wifi:424]:   DNS2: 192.168.2.19
[21:54:13][C][logger:439]: Logger:
[21:54:13][C][logger:440]:   Level: DEBUG
[21:54:13][C][logger:441]:   Log Baud Rate: 115200
[21:54:13][C][logger:443]:   Hardware UART: UART0
[21:54:13][C][ratgdo.number:024]: RATGDO Number 'Rolling code counter'
[21:54:13][C][ratgdo.number:024]:   Unit of Measurement: 'codes'
[21:54:13][C][ratgdo.number:028]:   Type: Rolling Code Counter
[21:54:13][C][ratgdo.number:024]: RATGDO Number 'Opening duration'
[21:54:13][C][ratgdo.number:024]:   Unit of Measurement: 's'
[21:54:13][C][ratgdo.number:030]:   Type: Opening Duration
[21:54:13][C][ratgdo.number:024]: RATGDO Number 'Closing duration'
[21:54:13][C][ratgdo.number:024]:   Unit of Measurement: 's'
[21:54:13][C][ratgdo.number:032]:   Type: Closing Duration
[21:54:13][C][ratgdo.number:024]: RATGDO Number 'Client ID'
[21:54:13][C][ratgdo.number:026]:  Type: Client ID
[21:54:13][C][switch.gpio:068]: GPIO Switch 'Status door'
[21:54:13][C][switch.gpio:090]:   Restore Mode: always OFF
[21:54:13][C][switch.gpio:031]:   Pin: GPIO16
[21:54:13][C][switch.gpio:068]: GPIO Switch 'Status obstruction'
[21:54:13][C][switch.gpio:090]:   Restore Mode: always OFF
[21:54:13][C][switch.gpio:031]:   Pin: GPIO15
[21:54:13][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact light'
[21:54:13][C][gpio.binary_sensor:016]:   Pin: GPIO0
[21:54:13][C][light:103]: Light 'Light'
[21:54:13][C][homeassistant.time:010]: Home Assistant Time:
[21:54:13][C][homeassistant.time:011]:   Timezone: 'EST5EDT,M3.2.0,M11.1.0'
[21:54:13][C][ratgdo:120]: Setting up RATGDO...
[21:54:13][C][ratgdo:121]:   Output GDO Pin: GPIO5
[21:54:13][C][ratgdo:122]:   Input GDO Pin: GPIO4
[21:54:13][C][ratgdo:126]:   Input Obstruction Pin: GPIO13
[21:54:13][C][ratgdo_dry_contact:029]:   Protocol: dry contact
[21:54:13][C][ratgdo.sensor:041]: RATGDO Sensor 'Openings'
[21:54:13][C][ratgdo.sensor:041]:   State Class: ''
[21:54:13][C][ratgdo.sensor:041]:   Unit of Measurement: 'openings'
[21:54:13][C][ratgdo.sensor:041]:   Accuracy Decimals: 0
[21:54:13][C][ratgdo.sensor:041]:   Icon: 'mdi:open-in-app'
[21:54:13][C][ratgdo.sensor:043]:   Type: Openings
[21:54:13][C][ratgdo.sensor:041]: RATGDO Sensor 'Paired Devices'
[21:54:13][C][ratgdo.sensor:041]:   State Class: ''
[21:54:13][C][ratgdo.sensor:041]:   Unit of Measurement: ''
[21:54:13][C][ratgdo.sensor:041]:   Accuracy Decimals: 0
[21:54:13][C][ratgdo.sensor:041]:   Icon: 'mdi:remote'
[21:54:13][C][ratgdo.sensor:045]:   Type: Paired Devices
[21:54:13][C][ratgdo.lock:012]: RATGDO Lock 'Lock remotes'
[21:54:13][C][ratgdo.lock:013]:   Type: Lock
[21:54:13][C][ratgdo.switch:068]: RATGDO Switch 'Learn'
[21:54:13][C][ratgdo.switch:070]:   Icon: 'mdi:plus-box'
[21:54:13][C][ratgdo.switch:090]:   Restore Mode: always OFF
[21:54:13][C][ratgdo.switch:014]:   Type: Learn
[21:54:13][C][ratgdo.binary_sensor:036]: RATGDO BinarySensor 'Motion'
[... 662 lines cut ...]
[22:05:12][D][cover:084]:   Position: 0%
[22:05:12][D][ratgdo_dry_contact:050]: Door action: CLOSE, door state: STOPPED
[22:05:16][D][homeassistant.binary_sensor:026]: 'binary_sensor.garage_door_closed_contact': Got state OFF
[22:05:16][D][binary_sensor:036]: 'Door Closed Sensor': Sending state ON
[22:05:16][D][ratgdo:072]: Door closed sensor: 1
[22:05:16][D][ratgdo:133]: Door state=CLOSED
[22:05:16][D][cover:170]: 'Door' - Publishing:
[22:05:16][D][cover:173]:   Position: 0%
[22:05:16][D][cover:186]:   Current Operation: IDLE
[22:05:16][D][switch:016]: 'Status door' Turning OFF.
[22:05:16][D][switch:055]: 'Status door': Sending state OFF
[22:05:21][D][esp8266.preferences:238]: Saving preferences to flash...
[22:06:03][D][time:044]: Synchronized time: 2024-01-24 22:06:02
[22:06:03][D][time:044]: Synchronized time: 2024-01-24 22:06:02
[22:06:30][D][button:010]: 'Query openings' Pressed.
[22:06:36][D][button:010]: 'Query status' Pressed.
```

Logbook for "Closed Sensor"
[image]

Logbook for "Open Sensor"
[image]

I can't experiment anymore tonight, but can tomorrow if you'd like.