ratgdo/esphome-ratgdo #210: Migrating from MQTT/Security 1.0 to ESPHome/Security1.0?

NukeThemTillTheyGlow, 2024-01-28:
I have 3 ratgdo's, 2 are running ESPhome and working well and I like the feature set, but both GDO's are Security 2.0.  The 3rd GDO is an old Liftmaster Professional "Formula 1" Model 2280 1/2 HP Security 1.0 opener and its running ratgdo with MQTT and working well.  I'd like to get to one platform as well as use the same UI and features as I look to start adding some automations.   

Do I understand correctly that  with the latest versions of ESPHome for ratgdo that I can now migrate directly to ESPHome via OTA update or do I need to take it down and reflash?  Can someone step me through the process please?

NukeThemTillTheyGlow, 2024-02-02:
And I have a 888LM, not 889LM.  Will that work?   It works with the MQTT software

dmzimmerman, 2024-02-02:
> And I have a 888LM, not 889LM.  Will that work?   It works with the MQTT software

Both of my 888LMs are working fine with the ESPHome firmware; I updated them over-the-air from the MQTT firmware last week.

cjhesse, 2024-02-02:
This is a duplicate of #187.

For the record, I just used ESPHome to compile the .bin for my ratgdo v2.5i and attempted to install it via OTA, but... I forgot to change the Wi-Fi credentials to point to my secrets. So it appears to have worked fine, but I still had to pull it down to reflash it.

Before I flashed and added it to Home Assistant, I disabled the MQTT version of the device so I wouldn't have conflicting entity IDs. Unfortunately ESPHome decided to add an identifier suffix to the device name anyway, which meant all the entity IDs also got that suffix, and I had to remove them by hand to get everything that used the old entity IDs working again.

dmzimmerman, 2024-02-02:
You didn't, really. You could have flashed it again by connecting to its fallback WiFi network.

cjhesse, 2024-02-02:
> > So it appears to have worked fine, but I still had to pull it down to reflash it.
> 
> You didn't, really. You could have flashed it again by connecting to its fallback WiFi network.

I connected to it but the page was blank. Was I supposed to navigate to something other than 192.168.4.1?

In any event, it works now.

dmzimmerman, 2024-02-02:
That's odd. When I connected to its dedicated network, it gave me exactly the same page it gives me now that it's on my actual WiFi network, scrolling logs and all. Maybe I got lucky.

NukeThemTillTheyGlow, 2024-02-03:
Update:  Took board down, flashed it with the updated ESPHome flasher, adopted to HA, reinstalled and works great!!  Thanks, glad this is working and now all on ESPHome.

closed 2024-02-04 by NukeThemTillTheyGlow

timespaced, 2024-04-23:
> That's odd. When I connected to its dedicated network, it gave me exactly the same page it gives me now that it's on my actual WiFi network, scrolling logs and all. Maybe I got lucky.

Is there no way to change the wifi ssid on the esphome firmware when you flash the generic bin and directly connect to the ratgdo via it's fallback wifi?

dmzimmerman, 2024-04-23:
> Is there no way to change the wifi ssid on the esphome firmware when you flash the generic bin and directly connect to the ratgdo via it's fallback wifi?

No, I don't believe there is. Or if there is, I haven't figured out how.