ratgdo/esphome-ratgdo #211: Sec 1.0 - communication interrupted after command.

rairai82, 2024-01-28:
I previously opened an issue under MQTT when I had that installed w/r/t random issues with my Sec 1.0 opener. See [https://github.com/ratgdo/mqtt-ratgdo/issues/65] for background on the issues I had with the MQTT Sec 1.0 install.

At Paul's recommendation I installed the esphome sec 1.0 code to use the logging capability of esphome.

I am finding that I am having more repeatable issues with the esphome code where it appears that communication is being interrupted for a period of time after a command is sent and/or incomplete packets. For the former, essentially status updates cease most of the time after a command is sent. 

On some occasions I get a bunch of incomplete packet errors.

I have a 1265 Liftmaster and a 889LM wall control.

See below examples:

I've also attached a log.

[logs_ratgdov25i-a8534a_logs-2.txt](https://github.com/ratgdo/esphome-ratgdo/files/14077171/logs_ratgdov25i-a8534a_logs-2.txt)

Incomplete packets:

`[12:17:39][D][ratgdo:095]: Door state=OPEN
[12:17:39][D][ratgdo:204]: Light state=ON
[12:17:39][D][ratgdo:210]: Lock state=UNLOCKED
[12:17:40][D][cover:076]: 'Door' - Setting
[12:17:40][D][cover:084]:   Position: 0%
[12:17:41][D][ratgdo:095]: Door state=CLOSING
[12:17:41][D][cover:170]: 'Door' - Publishing:
[12:17:41][D][cover:173]:   Position: 100%
[12:17:41][D][cover:186]:   Current Operation: CLOSING
[12:17:41][D][switch:012]: 'Status door' Turning ON.
[12:17:41][D][ratgdo:204]: Light state=ON
[12:17:41][D][ratgdo:210]: Lock state=UNLOCKED
[12:17:42][D][ratgdo:095]: Door state=CLOSING
[12:17:42][D][ratgdo:204]: Light state=ON
[12:17:42][D][ratgdo:210]: Lock state=UNLOCKED
[12:17:42][D][ratgdo:095]: Door state=CLOSING
[12:17:43][D][ratgdo:204]: Light state=ON
[12:17:43][D][ratgdo:210]: Lock state=UNLOCKED
[12:17:45][D][ratgdo:095]: Door state=UNKNOWN
[12:17:45][D][cover:170]: 'Door' - Publishing:
[12:17:45][D][cover:173]:   Position: 100%
[12:17:45][D][cover:186]:   Current Operation: IDLE
[12:17:45][D][switch:012]: 'Status door' Turning ON.
[12:17:47][D][ratgdo:204]: Light state=ON
[12:17:47][D][ratgdo:210]: Lock state=UNLOCKED
[12:17:48][D][ratgdo:234]: Button state=UNKNOWN
[12:17:48][D][binary_sensor:036]: 'Button': Sending state ON
[12:17:49][W][ratgdo_secplus1:262]: [82218] Discard incomplete packet: [38 ...]
[12:17:49][D][ratgdo:204]: Light state=ON
[12:17:49][D][ratgdo:210]: Lock state=UNLOCKED
[12:17:49][W][ratgdo_secplus1:262]: [82737] Discard incomplete packet: [39 ...]
[12:17:50][W][ratgdo_secplus1:262]: [82997] Discard incomplete packet: [38 ...]
[12:17:50][W][ratgdo_secplus1:262]: [83257] Discard incomplete packet: [3A ...]
[12:17:50][W][ratgdo_secplus1:262]: [83518] Discard incomplete packet: [39 ...]
[12:17:50][W][ratgdo_secplus1:262]: [83775] Discard incomplete packet: [38 ...]
[12:17:51][W][ratgdo_secplus1:262]: [84035] Discard incomplete packet: [3A ...]
[12:17:51][W][ratgdo_secplus1:262]: [84294] Discard incomplete packet: [39 ...]
[12:17:51][W][ratgdo_secplus1:262]: [84553] Discard incomplete packet: [38 ...]
[12:17:51][W][ratgdo_secplus1:262]: [84801] Discard incomplete packet: [3A ...]
[12:17:52][W][ratgdo_secplus1:262]: [85336] Discard incomplete packet: [38 ...]
[12:17:52][W][ratgdo_secplus1:262]: [85595] Discard incomplete packet: [3A ...]
[12:17:53][D][ratgdo:204]: Light state=ON
[12:17:53][D][ratgdo:210]: Lock state=UNLOCKED
[12:17:53][D][ratgdo:095]: Door state=CLOSED
[12:17:53][D][ratgdo:530]: Cancelling position callbacks
[12:17:53][D][cover:170]: 'Door' - Publishing:
[12:17:53][D][cover:173]:   Position: 0%
[12:17:53][D][cover:186]:   Current Operation: IDLE
[12:17:53][D][switch:016]: 'Status door' Turning OFF.
[12:17:53][D][switch:055]: 'Status door': Sending state OFF
[12:17:54][D][ratgdo:204]: Light state=ON
[12:17:54][D][ratgdo:210]: Lock state=UNLOCKED
[12:17:54][D][ratgdo:095]: Door state=CLOSED`

> 1 minute gap in status messages:

`[12:27:26][D][ratgdo:095]: Door state=CLOSED
[12:27:26][D][ratgdo:204]: Light state=OFF
[12:27:26][D][ratgdo:210]: Lock state=UNLOCKED
[12:27:27][D][ratgdo:204]: Light state=OFF
[12:27:27][D][ratgdo:210]: Lock state=UNLOCKED
[12:27:27][D][button:010]: 'Toggle door' Pressed.
[12:28:45][D][ratgdo:210]: Lock state=UNLOCKED
[12:28:45][D][ratgdo:204]: Light state=ON
[12:28:45][D][ratgdo:210]: Lock state=UNLOCKED
[12:28:45][D][ratgdo:204]: Light state=ON
[12:28:45][D][ratgdo:210]: Lock state=UNLOCKED
[12:28:46][D][ratgdo:095]: Door state=OPEN
[12:28:46][D][cover:170]: 'Door' - Publishing:
[12:28:46][D][cover:173]:   Position: 100%
[12:28:46][D][cover:186]:   Current Operation: IDLE
[12:28:46][D][switch:012]: 'Status door' Turning ON.
[12:28:46][D][switch:055]: 'Status door': Sending state ON
[12:28:46][D][ratgdo:204]: Light state=ON
[12:28:46][D][ratgdo:210]: Lock state=UNLOCKED
[12:28:47][D][ratgdo:204]: Light state=ON
[12:28:47][D][ratgdo:210]: Lock state=UNLOCKED
[12:28:47][D][ratgdo:095]: Door state=OPEN`

rairai82, 2024-02-03:
Any feedback or recommendations on this issue?  I'm happy to do more troubleshooting or logging if that would help narrow it down.

thekev, 2024-02-04:
Having the same issue, using esphome v1 with a 888LM (post-recall, with sticker, mfg 12/2018). When it gets really mad, it will restart its sync over and over and nothing will work. I could not reproduce that today. For a simple test, I just tried the light:

```
[2024-02-04 13:21:06] doors/garage/2/debug [D][ratgdo:204]: Light state=ON
[2024-02-04 13:21:06] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:21:06] doors/garage/2/debug [D][ratgdo:204]: Light state=ON
[2024-02-04 13:21:06] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:21:06] doors/garage/2/debug [D][ratgdo:095]: Door state=CLOSED
[2024-02-04 13:21:07] doors/garage/2/debug [D][ratgdo:204]: Light state=ON
[2024-02-04 13:21:07] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:21:07] doors/garage/2/debug [D][ratgdo:204]: Light state=ON
[2024-02-04 13:21:07] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:21:07] doors/garage/2/debug [D][light:036]: 'Light' Setting:
[2024-02-04 13:21:07] doors/garage/2/debug [D][light:047]:   State: OFF
[2024-02-04 13:21:07] doors/garage/2/light/light/state {"color_mode":"onoff","state":"OFF","color":{}}
[2024-02-04 13:21:07] doors/garage/2/light/light/state {"color_mode":"onoff","state":"OFF","color":{}}
[2024-02-04 13:21:08] doors/garage/2/debug [W][ratgdo_secplus1:262]: [5966745] Discard incomplete packet: [38 ...]
[2024-02-04 13:21:08] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:21:08] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:22:25] doors/garage/2/debug [D][ratgdo:095]: Door state=CLOSED
[2024-02-04 13:22:26] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:22:26] doors/garage/2/debug [D][ratgdo:204]: Light state=OFF
[2024-02-04 13:22:26] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:22:26] doors/garage/2/debug [D][ratgdo:204]: Light state=OFF
[2024-02-04 13:22:26] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:22:27] doors/garage/2/debug [D][ratgdo:204]: Light state=OFF
[2024-02-04 13:22:27] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:22:27] doors/garage/2/debug [D][ratgdo:095]: Door state=CLOSED
```
Took a bit over a minute to resync. However, when I operate the door, it takes about 4 minutes to resync. In this test, I tried closing it before the sync had finished. The GDO did not respond and the door remained open.
```
 [2024-02-04 13:27:07] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:27:08] doors/garage/2/debug [D][ratgdo:204]: Light state=OFF
[2024-02-04 13:27:08] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:27:08] doors/garage/2/debug [D][ratgdo:095]: Door state=CLOSED
[2024-02-04 13:27:08] doors/garage/2/debug [D][cover:076]: 'Door' - Setting
[2024-02-04 13:27:08] doors/garage/2/debug [D][cover:084]:   Position: 100%
[2024-02-04 13:27:08] doors/garage/2/debug [D][ratgdo:204]: Light state=OFF
[2024-02-04 13:27:08] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:27:27] doors/garage/2/debug [D][ratgdo:095]: Door state=OPEN
[2024-02-04 13:27:27] doors/garage/2/debug [D][cover:170]: 'Door' - Publishing:
[2024-02-04 13:27:27] doors/garage/2/debug [D][cover:173]:   Position: 100%
[2024-02-04 13:27:27] doors/garage/2/debug [D][cover:186]:   Current Operation: IDLE
[2024-02-04 13:27:27] doors/garage/2/debug [D][switch:012]: 'Status door' Turning ON.
[2024-02-04 13:27:27] doors/garage/2/debug [D][switch:055]: 'Status door': Sending state ON
[2024-02-04 13:27:27] doors/garage/2/cover/door/position/state 100
[2024-02-04 13:27:27] doors/garage/2/cover/door/state open
[2024-02-04 13:27:36] doors/garage/2/debug [D][cover:076]: 'Door' - Setting
[2024-02-04 13:27:36] doors/garage/2/debug [D][cover:084]:   Position: 0%
[2024-02-04 13:27:56] doors/garage/2/debug [D][ratgdo:095]: Door state=CLOSED
[2024-02-04 13:27:56] doors/garage/2/debug [D][cover:170]: 'Door' - Publishing:
[2024-02-04 13:27:56] doors/garage/2/debug [D][cover:173]:   Position: 0%
[2024-02-04 13:27:56] doors/garage/2/debug [D][cover:186]:   Current Operation: IDLE
[2024-02-04 13:27:56] doors/garage/2/debug [D][switch:016]: 'Status door' Turning OFF.
[2024-02-04 13:27:56] doors/garage/2/debug [D][switch:055]: 'Status door': Sending state OFF
[2024-02-04 13:27:56] doors/garage/2/cover/door/position/state 0
[2024-02-04 13:27:56] doors/garage/2/cover/door/state closed
[2024-02-04 13:28:13] doors/garage/2/debug [D][esp8266.preferences:238]: Saving preferences to flash...
[2024-02-04 13:30:53] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:30:53] doors/garage/2/debug [D][ratgdo:204]: Light state=ON
[2024-02-04 13:30:53] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:30:53] doors/garage/2/light/light/state {"color_mode":"onoff","state":"ON","color":{}}
[2024-02-04 13:30:54] doors/garage/2/debug [D][ratgdo:204]: Light state=ON
[2024-02-04 13:30:54] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:30:54] doors/garage/2/debug [D][ratgdo:204]: Light state=ON
[2024-02-04 13:30:54] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:30:54] doors/garage/2/debug [D][ratgdo:095]: Door state=OPEN
[2024-02-04 13:30:54] doors/garage/2/debug [D][cover:170]: 'Door' - Publishing:
[... cut ...]
[2024-02-04 13:37:22] doors/garage/2/debug [D][ratgdo:204]: Light state=ON
[2024-02-04 13:37:22] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:37:22] doors/garage/2/debug [D][ratgdo:204]: Light state=ON
[2024-02-04 13:37:22] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:37:22] doors/garage/2/debug [D][ratgdo:204]: Light state=ON
[2024-02-04 13:37:22] doors/garage/2/debug [D][ratgdo:210]: Lock state=UNLOCKED
[2024-02-04 13:37:23] doors/garage/2/debug [D][ratgdo:095]: Door state=CLOSED
```

RamblinMan86, 2024-02-08:
Similar issue with Liftmaster 8500 and 888lm control unit. I heard it could be the capacitors, so I ordered new ones and soldered those on. Now the unit won't even open at all... Still tinkering on it, but was curious if anyone else made progress?

Mint137, 2024-02-11:
I have a similar issue, it seems like my 889LM which is brand new keeps rebooting. If I disconnect power to the Ratgdo it doesn't happen.

I'm connected to a Security 1.0 garage door (1280R).

I don't have any logs since I had to remove it for the time being since the 889LM kept rebooting every few seconds and I couldn't operate the door normally. I'll try to obtain logs next time I get the chance to have it installed.