ratgdo/esphome-ratgdo #213: CLOSE command not being sent?

danps1, 2024-01-30:
Hi,

Received my Ratgdo yesterday and have installed it with latest ESPHome image, and am experiencing some strange behavior where it appears that the command to close the is not being sent correctly, but other commands are.

Ratgdo is 2.5 and GDO is Chamberlain CS105MYQ.

Here is the behaviour that I experience...

Log Headers...
```
INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/garage_door_opener.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Detected timezone 'Australia/Sydney'
INFO Starting log output from [redacted] using esphome API
INFO Successfully connected to garage_door_opener @ [redacted] in 0.004s
INFO Successful handshake with garage_door_opener @ [redacted] in 2.669s
[12:28:49][I][app:102]: ESPHome version 2023.12.9 compiled on Jan 29 2024, 14:03:14
[12:28:49][I][app:104]: Project ratgdo.esphome version 2.5i
[12:28:49][C][wifi:573]: WiFi:
[12:28:49][C][wifi:405]:   Local MAC: [redacted]
[12:28:49][C][wifi:410]:   SSID: 'HA-Net'[redacted]
[12:28:49][C][wifi:411]:   IP Address: [redacted]
[12:28:49][C][wifi:412]:   BSSID: E[redacted]
[12:28:49][C][wifi:414]:   Hostname: 'garage_door_opener'
[12:28:49][C][wifi:416]:   Signal strength: -45 dB ▂▄▆█
[12:28:49][C][wifi:420]:   Channel: 11
[12:28:49][C][wifi:421]:   Subnet: 255.255.255.0
[12:28:49][C][wifi:422]:   Gateway: [redacted]
[12:28:49][C][wifi:423]:   DNS1: 0.0.0.0
[12:28:49][C][wifi:424]:   DNS2: 0.0.0.0
[12:28:49][C][logger:439]: Logger:
[12:28:49][C][logger:440]:   Level: DEBUG
[12:28:49][C][logger:441]:   Log Baud Rate: 115200
[12:28:49][C][logger:443]:   Hardware UART: UART0
[12:28:49][C][ratgdo.number:024]: RATGDO Number 'Rolling code counter'
[12:28:49][C][ratgdo.number:024]:   Unit of Measurement: 'codes'
[12:28:49][C][ratgdo.number:028]:   Type: Rolling Code Counter
[12:28:49][C][ratgdo.number:024]: RATGDO Number 'Opening duration'
[12:28:49][C][ratgdo.number:024]:   Unit of Measurement: 's'
[12:28:49][C][ratgdo.number:030]:   Type: Opening Duration
[12:28:49][C][ratgdo.number:024]: RATGDO Number 'Closing duration'
[12:28:49][C][ratgdo.number:024]:   Unit of Measurement: 's'
[12:28:49][C][ratgdo.number:032]:   Type: Closing Duration
[12:28:49][C][ratgdo.number:024]: RATGDO Number 'Client ID'
[12:28:49][C][ratgdo.number:026]:  Type: Client ID
[12:28:49][C][switch.gpio:068]: GPIO Switch 'Status door'
[12:28:49][C][switch.gpio:090]:   Restore Mode: always OFF
[12:28:49][C][switch.gpio:031]:   Pin: GPIO16
[12:28:49][C][switch.gpio:068]: GPIO Switch 'Status obstruction'
[12:28:49][C][switch.gpio:090]:   Restore Mode: always OFF
[12:28:49][C][switch.gpio:031]:   Pin: GPIO15
[12:28:50][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact open'
[12:28:50][C][gpio.binary_sensor:016]:   Pin: GPIO14
[12:28:50][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact close'
[12:28:50][C][gpio.binary_sensor:016]:   Pin: GPIO12
[12:28:50][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact light'
[12:28:50][C][gpio.binary_sensor:016]:   Pin: GPIO0
[12:28:50][C][light:103]: Light 'Light'
[12:28:50][C][ratgdo:082]: Setting up RATGDO...
[12:28:50][C][ratgdo:083]:   Output GDO Pin: GPIO5
[12:28:50][C][ratgdo:084]:   Input GDO Pin: GPIO4
[12:28:50][C][ratgdo:088]:   Input Obstruction Pin: GPIO13
[12:28:50][C][ratgdo_secplus2:059]:   Rolling Code Counter: 2207
[12:28:50][C][ratgdo_secplus2:060]:   Client ID: 427321
[12:28:50][C][ratgdo_secplus2:061]:   Protocol: SEC+ v2
[12:28:50][C][ratgdo.sensor:041]: RATGDO Sensor 'Openings'
[12:28:50][C][ratgdo.sensor:041]:   State Class: ''
[12:28:50][C][ratgdo.sensor:041]:   Unit of Measurement: 'openings'
[... 184 lines cut ...]
[12:31:51][D][cover:170]: 'Door' - Publishing:
[12:31:51][D][cover:173]:   Position: 0%
[12:31:51][D][cover:186]:   Current Operation: CLOSING
[12:31:51][D][switch:016]: 'Status door' Turning OFF.
[12:31:51][D][light:036]: 'Light' Setting:
[12:31:51][D][number:012]: 'Rolling code counter': Sending state 2215.000000
[12:31:56][D][esp8266.preferences:238]: Saving preferences to flash...
[12:32:10][D][ratgdo:095]: Door state=CLOSED
[12:32:10][D][ratgdo:530]: Cancelling position callbacks
[12:32:10][D][ratgdo:204]: Light state=ON
[12:32:10][D][ratgdo:210]: Lock state=UNLOCKED
[12:32:10][D][ratgdo:189]: Learn state=INACTIVE
[12:32:10][D][number:012]: 'Rolling code counter': Sending state 2216.000000
[12:32:10][D][cover:170]: 'Door' - Publishing:
[12:32:10][D][cover:173]:   Position: 0%
[12:32:10][D][cover:186]:   Current Operation: IDLE
[12:32:10][D][switch:016]: 'Status door' Turning OFF.
[12:32:10][D][ratgdo:270]: Openings: 651
[12:32:10][D][sensor:093]: 'Openings': Sending state 651.00000 openings with 0 decimals of accuracy
[12:32:15][I][ota:117]: Boot seems successful, resetting boot loop counter.
[12:32:15][D][esp8266.preferences:238]: Saving preferences to flash...
```
Is there any additional serial logging that I should provide to help diagnose the issue?
Thanks for any help.
Dan

rlowens, 2024-01-30:
Do you have obstruction sensors connected and working? This is usual behaviour if the obstruction sensors aren't connected.

Ref https://github.com/ratgdo/esphome-ratgdo/issues/146

danps1, 2024-01-30:
Thanks @rlowens - it looks like https://github.com/ratgdo/esphome-ratgdo/issues/146 is exactly the issue I'm experiencing. Should have checked already open issues more thoroughly before creating a new one! Will close this one, and add to that thread instead