ratgdo/esphome-ratgdo #214: Sec+ 1 & 2, Dry contact door toggle

PaulWieland, 2024-01-30:
When both trigger open & trigger close are pulled low together, ratgdo should toggle the door.

This feature doesn't currently work in the ESP Home firmware.

ColColonCleaner, 2024-02-11:
This would be great to have!

Jefficus, 2024-02-28:
Do I understand correctly that this would let me wire up a basic doorbell button to use as a secondary trigger without having to pay through the nose for a fancy digital push-button? If so, then I'll be eagerly watching this space for breaking announcements.

PaulWieland, 2024-02-28:
Yes thats what the dry contact inputs are for. There are discrete open and close dry contact inputs, so you need two momentary buttons at the moment until this change is implemented.

saschawj, 2024-03-06:
Are there any updates to this from the ESPHome developers? Is it something that will be worked on or is it going to be ignored?

mattjm, 2024-05-06:
So what's the "correct" way to go about implementing this?  As far as I can tell ESPHome doesn't have a "trigger on two binary sensors" function built-in.  I made a first pass at [using lambdas and global variables](https://github.com/mattjm/esphome-ratgdo/blob/9f08a209434138cc0086f0f55aa2134041feb6bf/base.yaml#L144) to run the toggle function when both buttons had been pressed, while preserving the single button function.  It's not currently working but if this is the completely wrong approach I'd rather direct my efforts at a smarter way.

saschawj, 2024-05-27:
Now that dry contacts have been implemented, is there a configuration setting for triggering the toggle?

mattjm, 2024-05-28:
There's no configuration or implementation for toggle via dry contact yet, as far as I know.  This issue is to update the firmware such that if you wire one side of your switch to both contacts, pushing the button will toggle the door (so really the "configuration" is in hardware 😂 ).  

I abandoned the lambda approach I posted earlier, and I finally [got something working](https://github.com/mattjm/esphome-ratgdo/blob/1f3ea8ae129426f99fcdbba651ff4f0d6f4fd414/base.yaml#L129), but I need to test it a while longer. There is a question of appropriate values for delayed_on and delayed_off--I've changed them to be lower here since my dry contact "button" is actually electronically triggered (it's an old system that uses a separate transmitter/receiver to trigger the door using a dry contact--specifically the "flash2pass" system marketed to motorcycles).  

I'm using a 50ms delay to make sure both buttons are in the "on" state (if applicable) before deciding to toggle or open/close.  I chose that arbitrarily and it seems to work but I'd appreciate any input on the value from someone more familiar with esphome.  In my testing, without a delay the dry contact for close would always come on first and trigger the single contact "close" behavior before the open contact could trigger the toggle.  I wonder if we really just need a 16ms delay to do the check on the next loop (not sure if I understand the "main loop" concept properly though).

heffneil, 2024-06-05:
Sorry I didn't know this was out there.  What is the status still waiting for a software patch to have the toggle option vis ESPhome?

lukescott, 2024-06-08:
Sorry if I'm a bit out of date. I was on MQTT (probably an old one) and I just transitioned to ESP Home for my Security 1.0 doors.

There was some discussion about being able to use the dummy panels (emulation mode) as a dry contact switch. If the toggle functionality is added, will that work in this case? I do want to keep the door position and obstruction functionality. I recall there being issues with the dummy panels Lock/Light button causing issues when pressed. If I go the doorbell route, can I just do an open/close and keep the other functionality?

heffneil, 2024-06-08:
I mean this seems like a simple thing to implement.  The ratgdo knows where the door is.  The only thing it needs to know is previous motion.

Contact Closed:
    If Door is Close, Open Door
    If  Door is Open, Close Door
    if Door is neither closed or open, stop motion and record STOPPED and Current Motion
    if Door is stopped reverse current motion

I think this is the current logic on my garage doors that have basic closed contact control?

mattjm, 2024-06-12:
The toggle function is already implemented in firmware, this issue is just for triggering the toggle function from the dry contacts.  

In the currently released firmware you can trigger an individual contact for open or close, but there's no way to simply toggle from dry contacts.  I have a pull request in to support it--just waiting for feedback and merge.  

If folks want to try my version with toggle support, I can write up a procedure on how to use the esphome docker container to build and flash it.  (maybe there is an easier way but that is what I figured out flailing around with no experience).  It may be a couple weeks before I get to it.

heffneil, 2024-06-12:
Yes I don't need a write up so much as some instructions.  I run esphome in docker so I can flash my own firmware.  I am retrofitting my previous relay system to work with the new operators and the ratgdo is the middleman so to speak.  I would like both options (ratdgo control and relay control) to toggle the doors.

heffneil, 2024-07-10:
Gosh a good bit of time has passed. Is there any update?  I unfortunately only have a single pair of wires to toggle the door.  If there is a way to implement this it would be great.  Any update is appreciated.

lukescott, 2024-07-11:
Just out of curiosity (and re: https://github.com/ratgdo/esphome-ratgdo/pull/284), why when open/closed are triggered together? Could there be a configuration option to make just the open dry contact act as a toggle? Wouldn't that be easier?

And with open setup as a toggle, I would also love to repurpose the close contact as an axillary door open sensor w/ a reed switch. I still want to use the door state / obstruction state Ratgdo gets, so I would rather this axillary sensor be separate. What I really want is a way to know if the door is forced open for a security system. I'm assuming if someone manually opens the door the door state from the motor doesn't reflect the door being open.

saschawj, 2024-07-17:
Is there any special YAML config that is needed for this to work now? 
Just updated, tested and still doesn't work. 
Current YAML: 

`substitutions:
  id_prefix: "garage_ratgdov25i"
  friendly_name: "Garage-RatGDO"
  uart_tx_pin: D1
  uart_rx_pin: D2
  input_obst_pin: D7
  status_door_pin: D0
  status_obstruction_pin: D8
  dry_contact_open_pin: D5
  dry_contact_close_pin: D6
  dry_contact_light_pin: D3

web_server:

esphome:
  name: ${id_prefix}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: ratgdo.esphome
    version: "2.5i"

esp8266:
  board: d1_mini
  restore_from_flash: true

dashboard_import:
  package_import_url: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main

packages:
  remote_package:
    url: https://github.com/ratgdo/esphome-ratgdo
    files: [base.yaml]
    refresh: 1s

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

api:
  id: api_server
  encryption:
    key: <key>
ota:

improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:`

mattjm, 2024-09-13:
To an earlier question:  No special YAML config is required, it should "just work".  

The feature was implemented but not from my PR.  The current version requires the trigger on both inputs be pulled low (or high?  I  forget) for 500ms, which is usually longer than a digital signal will do it.  That's probably why it's not working for you.  You can confirm if it's working by triggering the wires manually for at least a half second.  

In my PR (#214) I had `delayed_on` set to 50ms and that is working well where I have a digital trigger.  IMO 500ms is way too long--even when I had a physical button hooked up to it I had to intentionally hold the button to trigger open/close.

ethierjf, 2025-05-02:
Very useful feature! Thanks for adding it! Allowed to reuse the old switch instead of creating more garbage!