ratgdo/esphome-ratgdo #215: Random Door Openings/Closing

preppietechie, 2024-02-01:
I'm using the current version available from the web installer (as of 2/1/24) on 2.5i hardware with an S+ 1.0 with an 888LM. 

The door is opening and closing frequently. The 888LM is flashing a yellow light every minute or so, and occasionaly beeping (like it's just received power after being unplugged for a bit). 

I had a similar problem with the MQTT version (See [Issue #34](https://github.com/ratgdo/mqtt-ratgdo/issues/34) for the MQTT Version), but that was closed suggesting that my MQTT broker was "spamming mqtt messages to ratgdo for some reason." Now that I'm on the ESPHome version, I've got the same issue. Within one hour of flashing and installing the firmware, I'm having random door open and close event. They seem far worse using the ESPHome version. The logs scrolled past too quickly for me to get it in time, but I did see the unusual string below.  Note that RATGDO believes that the button was pressed. Unless my house is haunted, I can confirm that the button was not pressed, as I'm the only one home.

<html><body>

10:18:39 | [D] | [ratgdo:210] | Lock state=UNLOCKED
-- | -- | -- | --
10:18:39 | [D] | [lock:054] | 'Lock remotes': Sending state UNLOCKED
10:18:39 | [D] | [ratgdo:204] | Light state=OFF
10:18:39 | [D] | [ratgdo:210] | Lock state=UNLOCKED
10:18:40 | [W] | [ratgdo_secplus1:262] | [741657] Discard incomplete packet: [3A ...]
10:18:40 | [D] | [ratgdo:234] | Button state=PRESSED
10:18:41 | [W] | [ratgdo_secplus1:262] | [742441] Discard incomplete packet: [3A ...]
10:18:41 | [W] | [ratgdo_secplus1:262] | [742716] Discard incomplete packet: [38 ...]
10:18:42 | [D] | [ratgdo:210] | Lock state=UNLOCKED
10:18:42 | [D] | [ratgdo:204] | Light state=ON
10:18:43 | [D] | [ratgdo:210] | Lock state=LOCKED
10:18:43 | [D] | [lock:054] | 'Lock remotes': Sending state LOCKED
10:18:43 | [W] | [ratgdo_secplus1:262] | [744559] Discard incomplete packet: [3A ...]
10:18:43 | [D] | [ratgdo:204] | Light state=OFF
10:18:44 | [D] | [ratgdo:210] | Lock state=UNLOCKED
10:18:44 | [D] | [lock:054] | 'Lock remotes': Sending state UNLOCKED
10:18:44 | [W] | [ratgdo_secplus1:262] | [745351] Discard incomplete packet: [3A ...]
10:18:44 | [D] | [ratgdo:210] | Lock state=UNLOCKED
10:18:45 | [W] | [ratgdo_secplus1:262] | [746639] Discard incomplete packet: [3A ...]
10:18:45 | [D] | [ratgdo:204] | Light state=OFF
10:18:45 | [D] | [ratgdo:210] | Lock state=UNLOCKED
10:18:47 | [W] | [ratgdo_secplus1:262] | [748216] Discard incomplete packet: [3A ...]
10:18:48 | [D] | [ratgdo:204] | Light state=ON
10:18:48 | [D] | [ratgdo:204] | Light state=ON
10:18:48 | [D] | [ratgdo:210] | Lock state=UNLOCKED
10:18:49 | [D] | [ratgdo:210] | Lock state=LOCKED
10:18:49 | [D] | [lock:054] | 'Lock remotes': Sending state LOCKED
10:18:51 | [D] | [ratgdo:204] | Light state=ON
10:18:51 | [W] | [ratgdo_secplus1:262] | [752447] Discard incomplete packet: [3A ...]
10:18:51 | [W] | [ratgdo_secplus1:262] | [752951] Discard incomplete packet: [3A ...]
10:18:52 | [D] | [ratgdo:210] | Lock state=UNLOCKED
10:18:52 | [D] | [lock:054] | 'Lock remotes': Sending state UNLOCKED
10:18:54 | [D] | [ratgdo:095] | Door state=CLOSING
10:18:54 | [D] | [cover:170] | 'Door' - Publishing:
10:18:55 | [D] | [cover:173] | Position: 0%
10:18:55 | [D] | [cover:186] |  

</body>
</html>

cjhesse, 2024-02-05:
I have an 889LM and sometimes experience similar issues on the ESPHome firmware. I never experienced the issue when using the MQTT version. MY door is Security+ 1.0 with purple learn button.

A couple times when I've had power to the garage turned off, then turned it back on, turning both the ratgdo and GDO at the same time, the door began cycling several times uncommanded, with the button making a beeping sound prior to each occasion. This didn't stop until I unplugged the ratgdo and plugged it back in. Unfortunately, I wasn't able to capture logs from the event because I was working outside and simply needed to get it to stop. When I have time, I can see if I can reproduce it.

Just this morning, the garage door began opening when my wife opened the door to the garage. I have an automation that turns the light on when the door opens, so I'm wondering if an issue with that command caused the garage door to open too. I don't have ratgdo logs of the events, but here's the Home Assistant histories and automation traces:
[image]
[image]
[image]
[image]
[image]
[image]

The gist of the automation is:
- Triggered by the kitchen door sensor changing from closed to open.
  1. Turn the garage door opener's light on.
  2. Wait 5 minutes.
  3. Turn the garage door opener's light off.
- The automation mode is restart so the light will turn off 5 minutes after the _last_ time the door was opened.

The light has also been less reliable with the ESPHome firmware. There's been a few times the light didn't turn on or didn't turn off, but I don't recall having any of those issues with the MQTT version. The history above shows a quick off-on-off sequence for the light when the last step of the automation occurs, but the camera in my garage shows the light only turning off, so it worked fine this time even though the history looks odd.

edrikk, 2024-02-05:
@mariusmuja 
This "initial boot" experience for V1 could be a much bigger issue if people are away from the home e.g. for vacation etc.
From reading the various posts (in this ticket and others), it's clear that ratgdo has to start after the gdo/wall panel is up and running.

MQTT firmware specifically handles this scenario:

https://github.com/ratgdo/mqtt-ratgdo/commit/6f62cc2b5f0e70afcbf75ff949737f0fe65d3e12

`door = 6; // stop from entering emulation after power outage when 889lm is booting up`

My understanding is that this was accidentally dropped in the MQTT firmware, which caused issues, and was then added back by the above commit.

ESPHome should do the same/a similar thing I would think...

cjhesse, 2024-02-08:
I haven't testing power cycles again yet, but I got another occurrence of an uncommanded opening that coincided with an attempted light command.

When my automation (as described above) attempted to turn off the opener's light at the end of the sequence, the light did not turn off and the door opened.

Unfortunately I don't have logs from the ratgdo because I was not there. Is it possible theESPHome  Sec+1 version has issues sending commands?

mariusmuja, 2024-02-09:
> Are such incomplete packets expected?

This happens when the GDO doesn't reply to a query within 100ms from a when a query is sent. A query is sent to the GDO every 250ms (by the wall panel or by ratgdo if it's in emulation mode) and normally a reply is sent immediately after. In this case the queries `0x39` and `0x3A` receive no reply.

DesertHillsautomation, 2024-02-11:
I am having the same issue. I have actually replaced my 88LM in the past because it died. I have RATGDO on two doors, same units with 88LM and one is acting up and the other is normal so far. I have an 8500 Jackshaft oddball unit that is maybe 1.0 but every thing we read we're not sure if it's a 2.0 protocol. Kinda bummed.

gullygossner, 2024-02-13:
I just had a similar occurrence this morning after a good stretch of flawless operation. With the ratgdo power on, I powered the opener on. The opener then opened without any interaction. 

The wall panel was doing a similar thing where it would be unresponsive, light up and then beep. It was almost like a boot loop. I restarted the ratgdo and it seems to have synced up and fixed itself. 

Easy fix when one is available to handle it, terrifying if you're not home.

preppietechie, 2024-02-13:
I followed the advice @PaulWieland gave and replaced the 888LM with an 889LM (using the replacement program Paul mentioned. Random door events have since stopped. It’s been 72 hours and I would have seen one by now if the problem were something else. Closing this because the problem was not with the RATGDO But with the 888LM.

tl;dr - if you’re using an 888LM, you’re gunna have a bad time. Swap it with an 889LM.

closed 2024-02-13 by preppietechie

gullygossner, 2024-02-13:
> I followed the advice @PaulWieland gave and replaced the 888LM with an 889LM (using the replacement program Paul mentioned. Random door events have since stopped. It’s been 72 hours and I would have [...]

I have a 889LM.

preppietechie, 2024-02-13:
> I have a 889LM.

It almost sounds like the capacitors in your 889LM are failing the same way the 888LMs do. It’s a super easy swap if you have a soldering iron, and new caps are super cheap, just match the same values listed on the side of the existing caps. You also might want to check your wiring to the wall button and make sure it’s getting the full voltage (and isn’t loose or anything). Either way, the initial issue (888LMs are bad) is resolved.

mariusmuja, 2024-02-15:
> I just had a similar problem. MQTT has been working flawlessly, but when I put the ESPHome version on the board, it almost immediately started acting up.

@tmeekes 

Sounds like you can easily replicate the issue, can you capture a log (with verbose logging enabled) of when this happens?

tmeekes, 2024-02-15:
I already swapped it back, but setup is easy enough I can look at doing a bit of testing and see what I can get out of it. Might have time to try it out tonight.