ratgdo/esphome-ratgdo #216: [Question] Homebridge

edrikk, 2024-02-02:
Hi,

Just wondering if anyone has successfully run the esphome firmware through Homebridge instead of HA?

If yes, sharing the howto would be amazing.  I assume changed to the yaml file, running an esphome docket, and using the homebridge-esphome-ts, but again a howto would be awesome.

Thanks!

hjdhjd, 2024-04-18:
https://github.com/hjdhjd/homebridge-ratgdo - latest versions support ESPHome firmwares.

Enjoy. 😄

edrikk, 2024-04-19:
Thank you!  I installed the plugin and it's been running perfectly for the last 2 days.

In case anyone would like it, below is the ESPHome yaml I use, which does a few things that I have found useful.  It also removes dependency to HomeAssistant for NTP by using sntp.

I'll leave this open for a couple of weeks and then close it.

Thank you again!

```
#This is a modified v25board_esp8266_d1_mini_secplusv1.yaml
#Updated time to use sntp instead of Home Assistant
#Updated wifi section to reduce power and add credentials
#Added reboot_timeout: 0s so that it doesn't reboot every 15 minutes
#Added on_boot delay of 120000
#Added level: INFO
#Added early_pin_init: false
#Added preferences:  flash_write_interval: 1440min

substitutions:
  id_prefix: ratgdov25i
  name: garagedoorleft
  friendly_name: Left Garage Door
  uart_tx_pin: D1
  uart_rx_pin: D2
  input_obst_pin: D7
  status_door_pin: D0
  status_obstruction_pin: D8
  dry_contact_open_pin: D5
  dry_contact_close_pin: D6
  dry_contact_light_pin: D3

web_server:

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  project:
    name: ratgdo.esphome
    version: "2.5i"
# Delay 120 seconds on boot to allow GDO and 889LM wall panel to boot first.  This is handled in code but maybe there's a bug?
  on_boot:
    priority: 800
    then:
      - lambda: |-        
          delay(120000);

esp8266:
  board: d1_mini
  restore_from_flash: true
  early_pin_init: false

preferences:
   flash_write_interval: 1440min

dashboard_import:
  package_import_url: github://ratgdo/esphome-ratgdo/v25board_esp8266_d1_mini_secplusv1.yaml@main

packages:
  remote_package:
    url: https://github.com/ratgdo/esphome-ratgdo
    files: [base_secplusv1.yaml]
    refresh: 1s

api:
  reboot_timeout: 0s
  encryption:
    key: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

# Sync time with NTP Servers.
time:
  - platform: sntp
    id: sntp_time
    timezone: XXXXXXXX/XXXXXXX
    servers:
     - time.nist.gov
     - pool.ntp.org
     - time.cloudflare.com

ota:

improv_serial:

wifi:
  output_power: 15.0dB
  ap:
    ssid: "Left GDO Hotspot"
    password: !secret wifi_password
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:
  level: INFO
```

hjdhjd, 2024-05-01:
Super helpful @edrikk Happy to add this as well to the documentation on the Homebridge end - might want to add instructions as well on how to upload the yaml for newbies. Feel free to ping me on the Homebridge Discord. Would love to discuss the YAML further.

edrikk, 2024-05-01:
Well the yaml is actually on the esphome side.  Meaning once the device is adopted, ESPHome allows for the yaml to be updated in the web gui.  The YAML is then used to compile the firmware.

In all honesty that might be beyond the scope of Homebridge (or plug-in) documentation…

donavanbecker, 2024-05-01:
Maybe an optional firmware can be generated specifically for Homebridge.

hjdhjd, 2024-05-01:
That's where I'm trending. 😄 I think I'll be doing so in the near future and making it available to folks.

edrikk, 2024-05-01:
Interesting.  Only issue may be that as far as I know there’s no way to update credentials (eg wifi SSID and pass) after the firmware is built…. Meaning the yaml includes this info, which is then backed into the firmware…. 

There may be a way around this that I don’t know, but that may be an issue for this specific approach…

hjdhjd, 2024-05-05:
I've created a [custom YAML](https://github.com/hjdhjd/homebridge-ratgdo/blob/main/homebridge-ratgdo.yaml) for those interested in using it with [homebridge-ratgdo](https://github.com/hjdhjd/homebridge-ratgdo). Enjoy!