ratgdo/esphome-ratgdo #217: Reverse functionality [Enhancement]

mvannas, 2024-02-03:
I have two door garage with sideways opening doors. 

Due to space constraints one door opener is installed in reverse. This means it shows OPEN when close and CLOSED when open. There is no way to make the Liftmaster operate in reverse, but I was wondering if a reverse functionality could be implemented in ratgdo, at least so that open/close status and buttons would show correctly to ESPHome/Home assistant etc.?

rlowens, 2024-02-03:
You can do this by editing your device .yaml file. Here's how I cam up with my suggestion:

1. Get the ratgdo working (backwards in your case) and adopted in ESPHome Dashboard so you have a device .yaml to edit.
2. Digging around here on github we find the source file we need to modify https://github.com/ratgdo/esphome-ratgdo/blob/main/base.yaml Here's what the current cover looks like:
```
cover:
  - platform: ratgdo
    id: ${id_prefix}_garage_door
    device_class: garage
    name: "Door"
    ratgdo_id: ${id_prefix}
    on_closed:
      - switch.turn_off: ${id_prefix}_status_door
    on_open:
      - switch.turn_on: ${id_prefix}_status_door

```
3. With this knowledge, you can then add your own modifications to the end of your device .yaml to change that cover how you want.. I suggest we change that original cover to be internal to ESPHome (so it won't be reported to Home Assistant) and create another template cover that reverses everything for us.

```
#custom modifications using https://esphome.io/guides/configuration-types.html#extend
cover:
  - id: !extend ${id_prefix}_garage_door
    internal: true
    name: "Internal Door"
#ref https://esphome.io/components/cover/template.html
  - platform: template
    id: ${id_prefix}_exposed_door
    name: Door
    device_class: garage
    has_position: false
    open_action:
      - cover.close: ${id_prefix}_garage_door
    close_action:
      - cover.open: ${id_prefix}_garage_door
    stop_action:
      - cover.stop: ${id_prefix}_garage_door
    toggle_action:
      - cover.toggle: ${id_prefix}_garage_door
    optimistic: true
```

I think that would do it? Note that once PR https://github.com/ratgdo/esphome-ratgdo/pull/195 is merged this can be re-written and enable position-control instead of just open/closed only.

FrankJaco, 2024-02-04:
@mvannas  Heads up.   I just saw a youtube previewing the features of Home Assistant that will be released this Wednesday (v2024.2).  They added an invert feature for lock entities.   Perhaps this will meet requires, no coding needed? 

fj

rlowens, 2024-02-04:
> They added an invert feature for lock entities.

Cover is not a Lock

FrankJaco, 2024-02-04:
Typo...   An invert feature for covers and locks. Both apparently.

YouTube was by "Smart Home Junkie".

mvannas, 2024-02-06:
Interesting, I will check it out and see how it works. It that doesn't work, I try rlowens modification.

mvannas, 2024-02-11:
Unfortunately this Home Assistant helper only allows you to change the functionality of an entity type switch. That can then be shown as cover, switch or lock. But it doesn't allow to invert an entity of cover type.