ratgdo/esphome-ratgdo #218: Button entity is stuck "on" (Security+ 1.0 / 889LM)

cjhesse, 2024-02-03:
When first migrated from the MQTT version to ESPHome, the button entity was "off". The first time I pressed it, it changed to "on" where it's been ever since then, persisting through reboots.

I'm using firmware 2023.12.9 with a Secuity+ 1.0 GDO and 889LM wall panel.

I don't have the logs from that very first button press, ran a test today. I pressed the wall button at xx:xx:30, held it for three seconds, and released it. I pressed the button again to stop the door at xx:xx:40 and one final time at xx:xx:45 to close it.

Here are the log entries from today involving the button (all other entries omitted):
```
[17:11:31][D][ratgdo:234]: Button state=PRESSED
...
[17:11:40][D][ratgdo:234]: Button state=PRESSED
...
[17:11:45][D][ratgdo:234]: Button state=PRESSED
```

Does the 889LM only send a signal when the button is first pressed, but nothing when released? If this is the case, it seems the ESPHome firmware should create the button as a button entity for such setups rather than a binary sensor.

neheb, 2024-05-17:
This is hilarious. The FAQ says to get an 889lm if you want a light button, but the esphome firmware doesn’t work with it. Is that what I’m gathering?

PaulWieland, 2024-05-17:
> but the esphome firmware doesn’t work with it

How did you come to that conclusion?

neheb, 2024-05-17:
Apparently when replacing the old wall panel ratgdo needs to be rebooted. Sorry for the noise.

cjhesse, 2024-06-09:
This issue isn't an operational issue; it's a reporting issue. ESPHome creates an entity that, at least in name, is supposed to be the status of the wall button (pressed or not). The entity is created as a binary sensor (i.e. binary_sensor.ratgdo_button) that starts as "off", but is latched to "on" after the button is pressed. It remains "on" indefinitely, making the sensor effectively useless.

I think this _should_ be implemented as a timestamp. When the button is pressed, update the timestamp. This would make it suitable for use as a trigger (or condition) in automations.

neheb, 2024-06-09:
This happened again when the power went out. I think the garage door was cycling for 3 hours before I noticed. I might need to get rid of ratgdo if I can’t find a solution.

cjhesse, 2024-06-09:
> This happened again when the power went out. I think the garage door was cycling for 3 hours before I noticed. I might need to get rid of ratgdo if I can’t find a solution.

You're describing a completely different issue. Please see #272, #254, #235, etc.

neheb, 2024-06-09:
Ah my bad.