ratgdo/esphome-ratgdo #221: [Feature Resuest] Default "Closed" upon restart

shanelord01, 2024-02-07:
I'm having issues recently where Home Assistant is reporting an error with sync with my ratgdo - resulting in Home Assistant reporting the door as being OPEN when it is CLOSED.

If I go to the ratgdo web interface - it is also showing it OPEN. When I hit the close button it reports as closed. Also if I instead restart the ratgdo from its UI, upon restart it correctly reports as CLOSED.

A quick fix would be to have ratgdo default to CLOSED - but I'm guessing some watchdog to re-check the door position (if not already there) would be ideal.

Not sure how to deal with this issue at the moment as I get door open reports randomly.

PaulWieland, 2024-02-07:
Which control protocol are you using?

The actual gdo status is updated at a regular interval by the gdo itself. For security + 2, the status is broadcast by the gdo every 5 minutes or so. The status is also be synced when ratgdo boots up.  

When yours is out of sync, what happens if you press the sync button? Or if you just wait?

shanelord01, 2024-02-07:
Hi Paul. Thanks for responding.

I am getting this error randomly in the Home Assistant alerts:
"ratgdo left sync failed
Failed to communicate with garage opener on startup."

When that happens the door immediately reports as open (Home Assistant Homekit bridge alerts everyone in my home).

I have tried changing WiFi networks to see if that resolves things, and it seems to for a while but I got the above error after I updated the WiFi networks in the yaml and sent an Install request from ESPHome.

```
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.78 using esphome API
INFO Successfully connected to ratgdov25i-8e63e0 @ 192.168.1.78 in 31.520s
INFO Successful handshake with ratgdov25i-8e63e0 @ 192.168.1.78 in 3.090s
[17:58:01][I][app:102]: ESPHome version 2023.12.9 compiled on Feb  7 2024, 17:56:47
[17:58:01][I][app:104]: Project ratgdo.esphome version 2.5i
[17:58:01][C][wifi:573]: WiFi:
[17:58:01][C][wifi:405]:   Local MAC: BC:FF:4D:8E:63:E0
[17:58:01][C][wifi:410]:   SSID: [redacted]
[17:58:01][C][wifi:411]:   IP Address: 192.168.1.78
[17:58:01][C][wifi:412]:   BSSID: [redacted]
[17:58:01][C][wifi:414]:   Hostname: 'ratgdov25i-8e63e0'
[17:58:01][C][wifi:416]:   Signal strength: -51 dB ▂▄▆█
[17:58:01][C][wifi:420]:   Channel: 6
[17:58:01][C][wifi:421]:   Subnet: 255.255.255.0
[17:58:01][C][wifi:422]:   Gateway: 192.168.1.1
[17:58:01][C][wifi:423]:   DNS1: 192.168.1.1
[17:58:01][C][wifi:424]:   DNS2: 0.0.0.0
[17:58:01][C][logger:439]: Logger:
[17:58:01][C][logger:440]:   Level: DEBUG
[17:58:01][C][logger:441]:   Log Baud Rate: 115200
[17:58:01][C][logger:443]:   Hardware UART: UART0
[17:58:01][C][ratgdo.number:024]: RATGDO Number 'Rolling code counter'
[17:58:01][C][ratgdo.number:024]:   Unit of Measurement: 'codes'
[17:58:01][C][ratgdo.number:028]:   Type: Rolling Code Counter
[17:58:01][C][ratgdo.number:024]: RATGDO Number 'Opening duration'
[17:58:01][C][ratgdo.number:024]:   Unit of Measurement: 's'
[17:58:01][C][ratgdo.number:030]:   Type: Opening Duration
[17:58:01][C][ratgdo.number:024]: RATGDO Number 'Closing duration'
[17:58:01][C][ratgdo.number:024]:   Unit of Measurement: 's'
[17:58:01][C][ratgdo.number:032]:   Type: Closing Duration
[17:58:01][C][ratgdo.number:024]: RATGDO Number 'Client ID'
[17:58:01][C][ratgdo.number:026]:  Type: Client ID
[17:58:01][C][switch.gpio:068]: GPIO Switch 'Status door'
[17:58:01][C][switch.gpio:090]:   Restore Mode: always OFF
[17:58:01][C][switch.gpio:031]:   Pin: GPIO16
[17:58:01][C][switch.gpio:068]: GPIO Switch 'Status obstruction'
[17:58:01][C][switch.gpio:090]:   Restore Mode: always OFF
[17:58:01][C][switch.gpio:031]:   Pin: GPIO15
[17:58:01][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact open'
[17:58:01][C][gpio.binary_sensor:016]:   Pin: GPIO14
[17:58:01][D][api:102]: Accepted 192.168.1.202
[17:58:01][W][component:214]: Component api took a long time for an operation (0.29 s).
[17:58:01][W][component:215]: Components should block for at most 20-30ms.
[17:58:01][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact close'
[17:58:01][C][gpio.binary_sensor:016]:   Pin: GPIO12
[17:58:01][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact light'
[17:58:01][C][gpio.binary_sensor:016]:   Pin: GPIO0
[17:58:01][C][light:103]: Light 'Light'
[17:58:02][C][ratgdo:082]: Setting up RATGDO...
[17:58:02][C][ratgdo:083]:   Output GDO Pin: GPIO5
[17:58:02][C][ratgdo:084]:   Input GDO Pin: GPIO4
[17:58:02][C][ratgdo:088]:   Input Obstruction Pin: GPIO13
[17:58:02][C][ratgdo_secplus2:059]:   Rolling Code Counter: 1684
[17:58:02][C][ratgdo_secplus2:060]:   Client ID: 4470073
[17:58:02][C][ratgdo_secplus2:061]:   Protocol: SEC+ v2
[17:58:02][C][ratgdo.sensor:041]: RATGDO Sensor 'Openings'
[17:58:02][C][ratgdo.sensor:041]:   State Class: ''
[... 75 lines cut ...]
[17:58:13][D][ratgdo_secplus2:213]: Query paired devices of type: ACCESSORY
[17:58:13][W][component:214]: Component ratgdo took a long time for an operation (0.18 s).
[17:58:13][W][component:215]: Components should block for at most 20-30ms.
[17:58:13][D][number:012]: 'Rolling code counter': Sending state 1712.000000
[17:58:17][D][ratgdo_secplus2:213]: Query paired devices of type: ALL
[17:58:17][D][ratgdo_secplus2:213]: Query paired devices of type: REMOTE
[17:58:17][D][ratgdo_secplus2:213]: Query paired devices of type: KEYPAD
[17:58:17][D][ratgdo_secplus2:213]: Query paired devices of type: WALL_CONTROL
[17:58:17][D][ratgdo_secplus2:213]: Query paired devices of type: ACCESSORY
[17:58:17][W][component:214]: Component ratgdo took a long time for an operation (0.18 s).
[17:58:17][W][component:215]: Components should block for at most 20-30ms.
[17:58:17][D][number:012]: 'Rolling code counter': Sending state 1719.000000
[17:58:21][D][ratgdo_secplus2:213]: Query paired devices of type: ALL
[17:58:21][D][ratgdo_secplus2:213]: Query paired devices of type: REMOTE
[17:58:21][D][ratgdo_secplus2:213]: Query paired devices of type: KEYPAD
[17:58:21][D][ratgdo_secplus2:213]: Query paired devices of type: WALL_CONTROL
[17:58:21][D][ratgdo_secplus2:213]: Query paired devices of type: ACCESSORY
[17:58:21][W][ratgdo_secplus2:107]: Triggering sync failed actions.
[17:58:21][W][component:214]: Component ratgdo took a long time for an operation (0.22 s).
[17:58:21][W][component:215]: Components should block for at most 20-30ms.
[17:58:21][D][number:012]: 'Rolling code counter': Sending state 1726.000000
[17:58:28][D][esp8266.preferences:238]: Saving preferences to flash...
[17:58:28][W][component:214]: Component preferences took a long time for an operation (0.07 s).
[17:58:28][W][component:215]: Components should block for at most 20-30ms.
```

shanelord01, 2024-02-07:
When I hit sync I get allot of this:

```
18:07:34 | [D] | [number:012] | 'Rolling code counter': Sending state 1877.000000
-- | -- | -- | --
18:07:37 | [D] | [ratgdo_secplus2:213] | Query paired devices of type: ALL
18:07:37 | [D] | [ratgdo_secplus2:213] | Query paired devices of type: REMOTE
18:07:37 | [D] | [ratgdo_secplus2:213] | Query paired devices of type: KEYPAD
18:07:38 | [D] | [ratgdo_secplus2:213] | Query paired devices of type: WALL_CONTROL
18:07:38 | [D] | [ratgdo_secplus2:213] | Query paired devices of type: ACCESSORY
18:07:38 | [W] | [component:214] | Component ratgdo took a long time for an operation (0.17 s).
18:07:38 | [W] | [component:215] | Components should block for at most 20-30ms.
18:07:38 | [D] | [number:012] | 'Rolling code counter': Sending state 1884.000000
18:07:41 | [D] | [ratgdo_secplus2:213] | Query paired devices of type: ALL
18:07:41 | [D] | [ratgdo_secplus2:213] | Query paired devices of type: REMOTE
18:07:41 | [D] | [ratgdo_secplus2:213] | Query paired devices of type: KEYPAD
18:07:41 | [D] | [ratgdo_secplus2:213] | Query paired devices of type: WALL_CONTROL
18:07:42 | [D] | [ratgdo_secplus2:213] | Query paired devices of type: ACCESSORY
18:07:42 | [W] | [component:214] | Component ratgdo took a long time for an operation (0.17 s).
18:07:42 | [W] | [component:215] | Components should block for at most 20-30ms.
```

shanelord01, 2024-02-07:
OK se end like the problem is worse and the ratgdo itself is getting out of sync with the door. 

Woke up to an “OPEN” door that was actually closed and has been according to home assistant opening and closing all night - but it never physically opened. 

[image]

edrikk, 2024-02-07:
Not saying these will resolve the issue, but maybe try adding the following to your yaml...

esp8266:
  early_pin_init: false

api:
  reboot_timeout: 0s

Just in case the device is losing connectivity intermittently and rebooting.

PaulWieland, 2024-02-07:
It seems like you have a wifi issue.

shanelord01, 2024-02-08:
> It seems like you have a wifi issue.

Possibly - but why is the ratgdo getting out of sync with the door? And not syncing again on its own?

The webui of the ratgdo is showing open when the physical door is closed - this has nothing to do with WiFi.

shanelord01, 2024-02-09:
Just downgraded my Unifi WiFi Firmware to 6.5.71 as people on Unifi Community Forum reporting issues with ESP devices on 6.6.x code. Will see how it goes. WebUI speed has definitley improved.

Still not sure why the ratgdo's seem to have secplus sync issues. I'll try changing Client ID and see if that makes a difference.

shanelord01, 2024-02-11:
OK - I’m over Unifi networking gear. Just borrowed some TP Link Omada gear from a friends work. 

EAP783 WiFi 7 access points.

All of my problems have gone away (including issues I was having with Matter over Thread devices). 

Just had to disable these settings and all is golden. 

Multicast/Broadcast Management
- Multicast-to-Unicast Conversion
- IPv6-Multicast-to-Unicast Conversion

Not to mention my site coverage has improved as well as performance vs Unifi WiFi 7 AP’s.

I’ll close this off for now.

closed 2024-02-11 by shanelord01