ratgdo/esphome-ratgdo #222: Only seeing discard incomplete packet messages, security+ v1

jfthomps, 2024-02-09:
First, thanks for creating this project!

I made my own board to use with ratgdo. I’m pretty sure I have the hardware correct. I'm using a NodeMCU board. I've tested that toggling the Tx pin drives the control line low, and that when the control line is driven low by another source, the Rx pin goes high.

I’m getting messages like the following when I have it connected and loaded with the esphome version of ratgdo:

|18:22:40|[W]|[ratgdo_secplus1:262]|[145114] Discard incomplete packet: [38 ...]|
| --- | --- | --- | --- |
|18:22:41|[W]|[ratgdo_secplus1:262]|[145373] Discard incomplete packet: [3A ...]|
|18:22:41|[W]|[ratgdo_secplus1:262]|[145641] Discard incomplete packet: [39 ...]|
|18:22:41|[W]|[ratgdo_secplus1:262]|[145899] Discard incomplete packet: [38 ...]|
|18:22:41|[W]|[ratgdo_secplus1:262]|[146167] Discard incomplete packet: [3A ...]|
|18:22:42|[W]|[ratgdo_secplus1:262]|[146426] Discard incomplete packet: [39 ...]|
|18:22:42|[W]|[ratgdo_secplus1:262]|[146678] Discard incomplete packet: [38 ...]|

My garage door opener is an older model Craftsman with a red learn button and 3 screw terminals where you wrap the wires around the screws (vs the terminal block type of connections). From what I’ve read it sounds like it should be using security plus v1.

I'm using a slightly modified version of this file: v25iboard_secplusv1.yaml

I can’t use the mqtt version of ratgdo because my board pins don’t match what it is built for.

Do you have any suggestions as to what the problem may be?

mariusmuja, 2024-02-09:
Most likely your GDO does not support sec+v1 on the wire, not all GDOs with a red button do. For example, I have a GDO with a red button and 3 screw terminals that does not.

jfthomps, 2024-02-09:
Does that mean ratgdo doesn't work with it at all, or do I need to configure it differently? If it doesn't support sec+v1, is it controlled similarly, but less securely? I'd be happy to help with adding support if that's possible.

rlowens, 2024-02-10:
See https://github.com/ratgdo/esphome-ratgdo/issues/205#issuecomment-1909213238

jfthomps, 2024-02-13:
I opened my wall switch yesterday, and I'd definitely say it's on the "dumb" side. It has 3 switches, 2 caps, 1 resistor, and one LED. The button that operates the door just connects the red and white wires together. The buttons for the light and lock each connect the red and white wires together through a cap. The resistor is just used with the LED to show it is powered.

Thanks for the help.

jfthomps, 2024-02-13:
oops, I meant to close this with the last comment, trying again

closed 2024-02-13 by jfthomps

fraserb2, 2024-12-19:
> Red button GDOs that don't support sec + 1 can be used with the dry contact control method. There isn't yet a prebuilt esp home firmware for this (there will be, its just not ready yet), but you can adapt the yaml file to do so on your own if you are so inclined. The MQTT firmware has dry contact as a control protocol option.

What is the current best method for these doors with the red button that doesnt support sec +1.