ratgdo/esphome-ratgdo #224: Garage Door - RATGDO - Security 1.0 - No Stop Cover Or Percentage Open + Logs Flooding

Steven4x4, 2024-02-09:
I originally posted this in the wrong github area. Hopefully this is the right one.

### The problem

I'm new to ESP Home. RATGDO is my first ESP Home device.
I'm not entirely sure what ESP Platform the v2.53i  is.

When I go to the ESP Home Devices Webpage Directly...
With the RATDGO v2.53i board with the Security 1.0 firmware it is flooding the logs every other second with:
14:46:32 | [D] | [ratgdo:210] | Lock state=UNLOCKED
14:46:33 | [D] | [ratgdo:204] | Light state=OFF
14:46:33 | [D] | [ratgdo:210] | Lock state=UNLOCKED
14:46:35 | [D] | [ratgdo:204] | Light state=OFF
14:46:35 | [D] | [ratgdo:210] | Lock state=UNLOCKED
14:46:36 | [D] | [ratgdo:204] | Light state=OFF
Video - https://www.youtube.com/watch?v=F8vxAVPcjjo
In the video Security 1.0 on the left, security 2.0 on the right.

For reference on the Security 2.0 firmware, it only does this every 5 minutes. (Which is about what I would expect.

Secondly, inside of home assistant, on the integrations > esp home > garage door device.
When you try to use the feature where you can open to a specific percentage. It does no work on the 1.0 firmware, but does work on the 2.0 firmware. I reach out to RATDGO directly, and he directed me to post here.
Additional, Open Cover works, but Stop Cover does not.

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.6

### What platform are you using?

ESP32

### Board

RATGDO v2.53i

### Example YAML snippet

```yaml
ESP HOME YAML:
substitutions:
  name: ratgdov25i-ca08b2
  friendly_name: Garage Door 1
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: <REDACTED>

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

rlowens, 2024-02-10:
What model of garage door opener do you have? As far as I know, they are only ever Security+ 2.0 or 1.0 on the wire, never both. So I wouldn't expect the Security+ 1.0 and 2.0 firmwares to both work at all.

Also, the percent open/close is done by timing how long a full open/full close takes and then stopping after a shorter amount of time, so only works after you have successfully run an open/close to calibrate it. Also wouldn't work if the stop command isn't working.

Also confirm how you have the ratgdo wired. And does it show the correct status of the obstruction sensors when you block them/unblock them?

PaulWieland, 2024-02-10:
> With the RATDGO v2.53i board with the Security 1.0 firmware it is flooding the logs every other second with:
> 14:46:32 | [D] | [ratgdo:210] | Lock state=UNLOCKED
> 14:46:33 | [D] | [ratgdo:204] | Light [...]

This is how Security + 1.0 works. Every 250 ms or so a query is sent and the status is reported back.

Steven4x4, 2024-02-10:
> What model of garage door opener do you have? What color Learn button?
> 
> As far as I know, they are only ever Security+ 2.0 or 1.0 on the wire, never both. So I wouldn't expect the Security+ 1.0 and 2.0 firmwares to both work at all.
> 
> Also, the percent open/close is done by timing how long a full open/full close takes and then stopping after a shorter amount of time, so only works after you have successfully run an open/close to calibrate it. Also wouldn't work if the stop command isn't working.
> 
> Also confirm how you have the ratgdo wired. And does it show the correct status of the obstruction sensors when you block them/unblock them?

Thank you for the response.I originally loaded the Security 1.0 firmware. I did notice when I added it to HA that it quickly did an OTA update. The device opens and closes, but the "Stop Cover" function does not work.It did calibrate itself; however, I did notice the calibration times were pretty far off what they should be. should be around 10-11 and it calibrated around 18–19. Even when using those times and setting my own, the percentage function was not working. I'm assuming that is due to the "Stop Cover" not working.I have it wired using the three colored wires, color-to-color. LiftMaster 3800 PLD - Purple Learn Button.

I just tested the obstruction sensors. The correct status is updating when I test them on the device page.

I have a second device on my other security 2.0 opener using the firmware for the 2.0 devices. Wired Color-To-Color and everything works as expected.

> This is how Security + 1.0 works. Every 250 ms or so a query is sent and the status is reported back.
Oh Ok. Thank you. I don't see the logs unless I connect directly to the device.
I just wasn't sure if it was cause for concern of it overheating based on the constant data flow and querying.

kmycek, 2024-03-26:
So my "stop" works just fine, but I am still not able to set the door at any percentage with Security 1.0+

its-pronounced-fee, 2024-03-30:
I'm having a similar issue. My doors never read as closed, they're always open even when they're closed. In my HA the UP and DOWN arrows are grayed out leaving only the 'STOP' button. While closed, if I hit the STOP in HA it'll open my garage. Going directly into the client (via ip) and hitting stop yields "the door is not moving".  Obviously my percentage doesn't work either but the priority for me is that ratgdo can read opened and closed. 

Light controls work in both HA and direct.

- ratgdo v.25i
- Running ESPhome Security 1.0 FW: 2024.3.0 Mar 25 2024, 04:18:21
- Both openers are Chamberlain 041DB002-2 circa 2011 - Purple Button

I wired my double as per the suggestion in the diagram but for my single, i just put the three leads into 1, 2, and 4. They both seem to work exactly the same.

slappula, 2024-04-17:
@its-pronounced-fee Not sure if this will help, but I had similar issues with one of my two garages after an ESPHome update through Home Assistant. I ended up pulling the Rat module down and flashing the firmware again (starting from scratch). Since then, it has worked perfectly. I did have an issue with the other garage (not showing the right status in HA) that was resolved by hitting the Learn button on the garage opener and then hitting the Sync button on the Rat firmware page.

its-pronounced-fee, 2024-04-17:
as it turns out it was my wall-switch. i was fortunate enough to qualify for a replacement through Chamberlain and got a free replacement - the new wall-switch works perfectly. I've called CS and they're sending me a 2nd one, it was the frigg'n wall-switch all a long.