ratgdo/esphome-ratgdo #225: 2 RATGDOs; one works perfectly - the other seem to be stuck OPEN, with no status changes

swobrien, 2024-02-11:
**Summary of problem**
I have 2 RATGDOs on similar Chamberlain GDOs. One works well, the other does not show the correct status of the GDO and hence does not control the opener correctly. They are wired identically. 

I tried switching the working RATGDO board between openers and it continued to work on the other GDO, so I can eliminate wiring mistakes, I think.

Does the fact that the one RATGDO works on both GDOs, mean that I have a hardware problem with the second RATGDO? The only difference I can see, is very slight differences in Firmware compile time

**Details**

GDO1 and GDO2
	Chamberlain
	Yellow button

Both RATGDOs
	V2.53i
	Flashed using ESPHome installer

**RATGDO1** - works with both GDOs
	[12:41:29][I][app:102]: ESPHome version 2023.12.9 compiled on Feb  4 2024, 21:59:37
	[12:41:29][I][app:104]: Project ratgdo.esphome version 2.5i
	
	Everything working from ESPHome and in Home Assistant
	
	Status - always correct
	Toggle - works
	Up/Down - works 
	Light - works
	Query Status
		12:56:28	[D]	[button:010]	
		'Query status' Pressed.
		12:56:28	[D]	[number:012]	
		'Rolling code counter': Sending state 906.000000
		12:56:28	[D]	[ratgdo:095]	
		Door state=CLOSED
		12:56:28	[D]	[ratgdo:204]	
		Light state=OFF
		12:56:29	[D]	[ratgdo:210]	
		Lock state=UNLOCKED
		12:56:29	[D]	[ratgdo:189]	
		Learn state=INACTIVE
	
	
**RATGDO2** - No status with GDO
	[12:44:39][I][app:102]: ESPHome version 2023.12.9 compiled on Feb 11 2024, 09:09:37
	[12:44:39][I][app:104]: Project ratgdo.esphome version 2.5i
	
	Status - always Open
	Toggle - works in ESPhome
	Up/Down - does NOT work
	Light - does NOT work
	Query Status
		12:52:36	[D]	[button:010]
		'Query status' Pressed.
		12:52:36	[D]	[number:012]	
		'Rolling code counter': Sending state 6541.000000
		
I tried re-flashing it from scratch - same behavior.
I tried putting the good RATGDO on the GDO that didn't work - it worked!
All manual controls work.
MyQ app works

It seems like everything is wired and toggle is working, but the RATGDO is not receiving anything from the GDO.

Any ideas on what I should try next?

rlowens, 2024-02-11:
Sounds like pretty comprehensive testing. Probably a bad rx transistor or solder on the bad board. Unless you are up for surface-mount soldering, I suggest you contact @PaulWieland by replying to the order confirmation email about getting a replacement for the defective board.

swobrien, 2024-02-11:
Thanks for the fast response. I’ll do that if  no other suggestions as I don’t have the equipment for some soldering. Thanks again

swobrien, 2024-03-02:
I contacted Paul, and he sent me a new device as he suspected a faulty one. Great support.

The new device worked, just like my first one and I now have two perfectly functioning ratgdo controllers and all my old Home Assistant automations restored for the first time since MyQ went stupid. 

Thanks all and thanks Paul for a great device.

closed 2024-03-02 by swobrien