ratgdo/esphome-ratgdo #230: Can't find configuration URL

PlinyElderATX, 2024-02-19:
What is the configuration URL - when clicking "visit device" after flashing ESPHome I simply get the page for status, etc. I've tried guessing a few obvious URLs /config, /configuration, /settings but no dice. I really want to secure this thing a bit more but have no earthly idea how to get to the config URL

rlowens, 2024-02-19:
ESPHome doesn't have a separate configuration page on the device's webserver. Everything changeable at run-time is on the main page.

All configuration beyond that is done by editing the device .yaml file in the ESPHome Dashboard (or otherwise if using ESPHome via the command line) and compiling and uploading a new firmware binary.

Re:security: To enable authentication on the webserver, see https://esphome.io/components/web_server.html#configuration-variables and use the auth: option like this:

```
web_server:
  auth:
    username: somename
    password: somepassword

```

Or if you prefer to not have a web server on the device (so it it only controlled via Home Assistant or MQTT if enabled manually), you can remove it in your .yaml with this:

```
#custom modifications using https://esphome.io/guides/configuration-types.html#extend
web_server: !remove

```

PlinyElderATX, 2024-02-19:
Ah I see- thx