ratgdo/esphome-ratgdo #236: FirmUpdate Question.

C4Wiz, 2024-03-02:
im sure this has been covered before, but i just bought the device for use with control4. i got it up and running no problem with the web installer. but how do i update the firmware via the GUI of the device itself? i see a OTA button, but i does not say anywhere what the current firmware is nor do i see a place to download it from?

[image]

alrassia, 2024-03-08:
You can update the firmware from the esphome dashboard or through home assistant. To get the dashboard active check the installation manual for esphome in the documentation. It’s listed under bonus dashboard

C4Wiz, 2024-03-08:
> You can update the firmware from the esphome dashboard or through home assistant. To get the dashboard active check the installation manual for esphome in the documentation. It’s listed under bonus dashboard

Right, and where is the firmware found?

alrassia, 2024-03-08:
That is automatically compiled based on your configuration for your device. You can choose to update wireless or receive the file to do it manually.

C4Wiz, 2024-03-08:
> That is automatically compiled based on your configuration for your device. You can choose to update wireless or receive the file to do it manually.

how? without home assistant?

PaulWieland, 2024-03-08:
See here: https://github.com/ratgdo/esphome-ratgdo/issues/128#issuecomment-1929996365

PaulWieland, 2024-03-08:
@C4Wiz the comment I linked you to explains how to download the firmware file.  It even has a link to the .bin file for the v2.5 board / security + 2 protocol firmware.

C4Wiz, 2024-03-08:
> @C4Wiz the comment I linked you to explains how to find and download the firmware file. It even has a link to the .bin file for the v2.5 board / security + 2 protocol firmware.

there is no folder view no date updated, how do i know when it's newer then what i have installed? why not have the https://ratgdo.github.io/esphome-ratgdo/v25iboard-esp8266 folder accessable to see when the file was updated last and to download it?

currently without home assistant there is no way to see what firmware you are running or if a update is available!

PaulWieland, 2024-03-08:
Firmware Build Date has been added to the web ui in the latest build, which will give you some idea of how old the firmware is. I need to investigate github actions and see if there is a way to push the built binaries into the releases section for manual download.

C4Wiz, 2024-03-08:
> Firmware Build Date has been added to the web ui in the latest build, which will give you some idea of how old the firmware is. I need to investigate github actions and see if there is a way to push the built binaries into the releases section for manual download. 

Awesome, thank you. Is it possible to have a update available and update from the UI as well?

C4Wiz, 2024-03-31:
> I don’t think esp home supports that. The intent is to use the esphome dashboard to push updates.

if i update via the esp home dashboard i get tons of:

`[09:44:39][D][api.connection:145]: : Sending keepalive failed 2 time(s), will retry in 1000 ms
[09:44:40][D][api.connection:145]: : Sending keepalive failed 3 time(s), will retry in 1000 ms
[09:44:41][D][api.connection:145]: : Sending keepalive failed 4 time(s), will retry in 1000 ms
[09:44:42][D][api.connection:145]: : Sending keepalive failed 5 time(s), will retry in 1000 ms
[09:44:43][D][api.connection:145]: : Sending keepalive failed 6 time(s), will retry in 1000 ms
[09:44:44][D][api.connection:145]: : Sending keepalive failed 7 time(s), will retry in 1000 ms
[09:44:45][D][api.connection:145]: : Sending keepalive failed 8 time(s), will retry in 1000 ms
[09:44:46][D][api.connection:145]: : Sending keepalive failed 9 time(s), will retry in 1000 ms
[09:44:47][W][api.connection:142]: : Sending keepalive failed 10 time(s), will retry in 1000 ms`

but if i download the firmware from above and update via the device itself i do not see this