ratgdo/esphome-ratgdo #238: Lost Connection after Power Outage

jizhoul, 2024-03-06:
For some reason, the firmware will lose wifi connection when encountering power outage.

mwaterbu, 2024-05-25:
I have a related issue. After the outage, my ratgdo came back online, but the wall remote no longer works. Also I can only send commands from the ratgdo intermittently. Sometimes I get a message in the log about the api taking too long for an operation.
I had been using this project for 6+ months with no issue, but when I updated to a newer version (which also switched to this official repo from the original one which seems to no longer exist), something got messed up. My rolling code on the ratgdo reset, and while it still works, other behaviors of the system (especially with the wall control) have been strange, making me think this is somehow related and the codes are out of sync or something.
For instance, #188 and #267 are both things I only noticed after the update, and with previous power outages, I never had this sort of issue.

billmoseley, 2024-08-15:
I'm seeing the same issue. I assume the device is failing to connect to wifi after power failure because the ratgdo is up before my wifi comes back online. 

Will the ratgdo continue to try and connect?

There's the [reboot_timeout](https://esphome.io/components/wifi.html#manual-ips) option on ESP home, but I don't see that in the YAML. 

I have not adopted my devices in Home Assistant.