ratgdo/esphome-ratgdo #239: Keep getting is closing then closed status with the door always closed

EEkedOut, 2024-03-08:
[image]

Again, the door is not operating and is closed when it says is closing. Also, the calibration didn’t work. It set way too long of open and closing times thus making the slider inop.

tomkessler, 2024-04-09:
I'm having something similar, but only when the door is closed via an automation.  The bottom set of open/close is from the wall button.  The top set is from an open-delay-close automation.  You can see the "is closing" triggered by the automation and then an "is closed" but then there is a duplicate "is closing" and "was closed" which follow.

[image]

thurstonsand, 2026-11-10:
fwiw, I've been running #701 for a bit now and it's resolved the problem. hoping to get it merged.