ratgdo/esphome-ratgdo #240: Door randomly shows wrong status

smarthomeaf, 2024-03-08:
Is is possible to use a 3rd party tilt or door sensor to always show correct status of door.   @PaulWieland, I believe you said on a previous thread the plan was to add this for dry contact installation, but would it be possible to the sec1 install as well.   I can work around the issue by creating a cover template in home assistant, but it would be cleaner to do it all in esphome.

I know the best option would be for me to go purchase a 889LN to solve this, but don't want to fork up $100 to get 2 of them.

Any help would be appreciated.

cgbale, 2024-03-09:
If you have 888LM keypads, you are eligible for free replacement with an 889LM. 

https://cloud.info.liftmaster.com/888LMReplacement

smarthomeaf, 2024-03-09:
great to know but unfortunately I have the 78LM