ratgdo/esphome-ratgdo #241: Won't get a sync between ratgdo 2.53i board and Chamberlain ML700EV

Ben-x86, 2024-03-09:
Hi, I'm having trouble to get the ratgdo board to work with my garage door opener. 

ratgdo 2.53i board successfully flashed with ESPHome
Chamberlain ML700EV (should be Security + 2.0)

At first i did the wiring as it is mentioned on the official diagram. Red cable to red terminal, white cable to the right terminal of red terminal and black cable to gray terminal. Result was a random opening and closing of the garage door. I was able to send commands from esphome but they did not to what they were supposed to.

Second try. After researching the github regarding such issues i realised i have to apply different wiring because of another terminal setup ([link](https://github.com/ratgdo/esphome-ratgdo/issues/122#issuecomment-1869532644)). But that way i wont get any kind of communication between the opnener and the board. 

Maybe someone can help me with that. Pressing sync button ends in what you can see in log screenshot. Probably the wiring is still a problem but i can't find any more hints what to change. I also powered off and on the opener without any other result. The original setup without ratgdo works just fine.

Original setup without ratgdo (connected wall button and obstruction sensor:
[image]

With ratgdo connected to terminals 0=red cable, 2=white cable, 4=black cable:
[image]

Log:
[image]

esphome (state of door says open while it's actually closed and it won't change the state):
[image]

Ben-x86, 2024-03-10:
Update: without wall button and obstruction sensor beeing connected to the ratgdo board it works. I can switch light on and off,  use the toggle button and in Home Assistant i get a Status (open/close, is opening/closing in %). The buttons open and close won't do anything but i can live with the toggle.

Maybe my journey helps someone else.

PaulWieland, 2024-03-10:
ROW openers that have the wall panel connected to the red terminal cannot be rewired wired using the pass through terminals on the ratgdo control board since they are not eserial devices. By doing that you effectively switched the wall panel wiring from the red terminal to the green eserial terminal.

closed 2024-03-10 by PaulWieland

torandreroland, 2024-08-01:
@PaulWieland: I have a ROW-opener (Liftmaster LM60) and a [Liftmaster 75EV wall button](https://www.liftmaster.eu/en-gb/produktkatalog/75ev-en). Does your previous comment here state that this wall button is incompatible with RatGDO, or is it some other way to connect it?

I'm asking because I have problems with controlling my garage door opener, and based on that comment and [this comment in issue 146](https://github.com/ratgdo/esphome-ratgdo/issues/146#issuecomment-1913618616), I'm thinking me connecting the wall button directly to the red terminal of the garage door opener is to blame.

I can live without the wall button, just trying to get to the bottom of my troubleshooting.

PaulWieland, 2024-08-01:
@torandreroland connect your wall button to your GDO as it was before. Don't use the pass through terminals.
Connect the ratgdo ctrl red wire to the eserial terminal (usually green, labeled 0).

torandreroland, 2024-08-01:
Thank you for your very quick reply. This was the way I had hooked it up, so I guess that my problem with controlling the garage door opener is not due to this then.

I just got back from the garage door and uncoupling the wall button didn't do much difference either.

But at least I have crossed that off my troubleshooting list. Thank you very much for your help.

traxeon, 2024-08-02:
I dont know this model that well, but it looks like wires are missing.   The wires on the outside edge of the block should be for the door button.  The wires to the inside should be the motion sensors.  On my door opener, i have 2 sets of wires running into those, not one set.   The grounds are twisted together for each sensor and the positives are twisted together so the sensors are powered.   This images helped me during my install today.,
https://imgur.com/wXQE0To
Also look at this post and BOTH pictures for clues.
https://www.reddit.com/r/ratgdo/comments/19bkz95/ratgdo_253_wiring_details/