ratgdo/esphome-ratgdo #243: RATGDO not working with a Liftmaster 8355

jrfernandes, 2024-03-10:
Hello all.

I got a RATGDO v2.5 board and flashed it with ESPHome using the [official web flash tool](https://ratgdo.github.io/esphome-ratgdo/). I selected "Security + 2.0" + "ratgdo v2.5x".

Then, connected to my Liftmaster 8355 (yellow learn button) according to the diagram on the website.

None of the controls work, and when opening ESPHome webpage, I see a lot of these errors:
```
19:37:37 | [W] | [ratgdo_secplus1:262] | [602273] Discard incomplete packet: [39 ...]
19:37:37 | [W] | [ratgdo_secplus1:262] | [602532] Discard incomplete packet: [38 ...]
19:37:37 | [W] | [ratgdo_secplus1:262] | [602792] Discard incomplete packet: [3A ...]
19:37:37 | [W] | [ratgdo_secplus1:262] | [603050] Discard incomplete packet: [39 ...]
 ...
```

Opening/closing the door, or turning on the light does not work. The statuses are not updating either, with the exception of the obstruction status, which does work.

What am I missing?

My ESPHome yaml:
```yaml
substitutions:
  name: "garage-door"
  friendly_name: garage-door
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: !secret api_key
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

Pictures of the GDO:
[image]
[image]

rlowens, 2024-03-10:
> I selected "Security + 2.0" + "ratgdo

Must not have, since it has Security + 1.0 firmware installed.

Change this:

`packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@main`

to

`packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main`

Then compile and flash OTA and see if you have any errors then.

C4Wiz, 2024-03-10:
looks like you installed sec+v1 firmware (purple button) by mistake?

TerryFrench, 2024-03-24:
@jrfernandes if the answers from the 2 other contributors solved your problem, please reply here, and mark the issue as resolved. This will help others with the same issue. Thanks!