ratgdo/esphome-ratgdo #244: Request: Add "Delay before Closing (seconds)" variable to allow "door close warning" automations to be triggered and completed by external Home Automation System

cgbale, 2024-03-16:
My home automation system triggers "Warning Garage Door is Closing" sequences when the "Close Garage Door" command is triggered. Unfortunately, the garage door is already 1/3 closed before this happens, which is not nearly enough time for someone to react (to prevent a car from backing into a closing garage door.) This delay / warning is required for UL325 Safety Certification Compliance.

Can you please add a variable to the UI for "Delay before Closing (seconds)". This will allow sufficient time for the home automation system to complete any audible / flashing light notifications before RATGDO starts closing the garage door.

Thank you!

restauffer, 2024-03-16:
Are you using Home Assistant?  If, yes, why not build the delay into your automation? The sequence would look like this:

- Trigger
- Announcement
- Delay
- Do open command

Ricke

cgbale, 2024-03-16:
I’m using Control4 and not Home Assistant. The C4 ratgdo driver integrates with a native C4 UI driver to control the door through the UI. It’s not possible to add a delay through that combination of drivers and retain the Native UI for the garage door. (It could be done through a separate menu/experience item, but that’s a non-intuitive workaround from a UI perspective.) 

I’ve requested the ratgdo C4 driver developer to add a delay function within the drivet, but received no response. Native capability in ratgdo may be a more flexible solution anyway.

restauffer, 2024-03-16:
Thanks for the explanation. I never heard of Control4 before.

rlowens, 2024-03-16:
> I’m using Control4 and not Home Assistant.

You've created this issue in the esphome-ratgdo firmware, is that what you are using?

cgbale, 2024-03-16:
Yes, the Chowmain ratgdo Control4 driver uses the esphome API. 

https://chowmain.software/drivers/control4-ratgdo

cwagdev, 2024-09-06:
> Are you using Home Assistant? If, yes, why not build the delay into your automation? The sequence would look like this:
> 
> * Trigger
> * Announcement
> * Delay
> * Do open command
> 
> Ricke

Any chance you can elaborate on this? I am new to Home Assistant and installed both it and ratgdo yesterday. For personal reasons I came to like the warning+delay that comes with MyQ. My ideal would be that when anyone uses HA to close the garage it doesn't do so immediately but instead flashes the light and delays 5s. 

Am I on the right track here? 

[image]

I am hoping to avoid a lot of opening and closing to stumble my way through this, does this alone delay the original action and then do what I fill in after or will the original action still happen and then the process I defined kicks off?

restauffer, 2024-09-07:
In your example I suspect  your automation trigger is caused by “pushing “ the Toogle icon in the Ratgdo HA UI. Pushing this icon will start the open/close of the door immediately AND also trigger your automation. You want a separate trigger that when pushed will only trigger the automation but not start opening/closing the door. A good choice might be a toggle helper. Once you trigger the automation by toggling your helper switch the first action(s) should be to flash the GDO light on and off as many times as you want. The next action may be a delay and the third action would be to use the toggle you currently have as your trigger. I said the second action MAY be a delay but I recommend flashing the light enough times to cover the number of seconds seat you want. A separate delay will then not be needed.Flashing the lights may not be as easy as it seems.  I would try a sequence of action like this:Toogle lightDelay 1 secondToggle lightDelay 1 second      Etc for desired length. Turn off light You may have to play with the delay as a series of fast back-to-back commands sometimes does not produce desired results.There is a drawback to this rather simple automation. Since you are using a toggle switch the flashing will occur whether you are opening or closing the door. If you only want this to occur if the door is closing, add a condition that says the door must be open for the actions to run.Ricke Sent from my iPhone

cwagdev, 2024-09-07:
Thanks for the reply even after I had deleted my question! As you stated yes I soon confirmed the action takes place immediately when using the actual toggle. Which makes sense. 

Having a "fake" toggle that ultimately calls the real toggle makes sense. Thanks.

restauffer, 2024-09-08:
Chris,

Once you post in Github it sends out an email. Even if you later delete the
post, it can't recall the emails. I was replying to the email and never
even looked in Github.  Good luck with your automation.

Ricke Stauffer