ratgdo/esphome-ratgdo #248: cover.toggle service stops working intermittently

smarthomeaf, 2024-03-24:
cover.toggle stops working randomly on ratgdo (works for all other devices)  There is no error displayed anywhere.  Using the entity toggle seems to work fine.

**Not working**
[image]

**working**

[image]

PaulWieland, 2024-03-24:
And what happens when you try it from the devices web ui?

alexonbandit, 2024-04-19:
Confirmed! I do have the same issue!
I run the ESP version and have to use the toggle button in my automations in Home Assistant since I don’t have an obstruction sensor.

smarthomeaf, 2024-04-20:
> And what happens when you try it from the devices web ui?

This seems to be fine now.   Maybe a recent ESP update solved the issue.