ratgdo/esphome-ratgdo #250: wireless remote requires 2 presses to activate gdo

Juggler00, 2024-04-10:
ratgdo v2.5 running esphome-ratgdo Security + 1.0
gdo: Chamberlain 3800C
wall button: 889LM MyQ

I have connected my ratgdo and flashed it with ESPHome-ratgdo Security + 1.0. I am able to control the door and see status through Homeassistant. The wall button continues to work as expected.

Since connecting the ratgdo, I now have to press the button on my wireless remote twice to activate the door. This is required ~80% of the time. 

I have looked in the logs, but don't see anything obvious. Is anyone else experiencing this? Any suggestion for what I should be looking for to diagnosis/fix the issue?

restauffer, 2024-04-10:
I had a similar problem.  Another piece of info I had was, the remote worked ok if it was close to the gdo.  In my case the problem was the wall wart I was using was putting out RFI that was interfering with the receiver.  My solution was to move the Ratgdo and wall wart about 15 ft from the GDO and run longer wires from the Ratgdo to the GDO. YMMV.

4vanger, 2024-04-16:
Same for me, Firmware: 2024.3.2 (Apr 8 2024, 04:18:40), Hardware: 2.5i, Security+1.0

ScottinOkla, 2024-07-02:
I am also experiencing this, I did not know if it was due to HomeAssistant or Ratgdo.  I was here looking into another issue when I stumbled upon this.

PaulWieland, 2024-07-02:
Mount the control board further away from your GDO antenna.

closed 2024-07-02 by PaulWieland

4vanger, 2025-02-24:
Still happens for me - first click - wall control starts flashing, second click - door opens.  ratgdo control board is about 10" from door opener