ratgdo/esphome-ratgdo #251: Sec+ 1.0: light wall button not working

neheb, 2024-04-13:
I wired up ratgdo to my Liftmaster and found out that I can control everything in Home Assistant, including the light. However the light button on the wall no longer works. I assume this is a firmware issue as ratgdo can control the light just fine.

rlowens, 2024-04-13:
From the FAQ https://paulwieland.github.io/ratgdo/09_faq.html

> **My Security + 1.0 wall panel’s light and lock buttons don’t work after connecting ratgdo**
> You have an analog control wall panel. ratgdo in Security + 1.0 control mode requires a digital communication with the garage door opener in order to both control the door and to get the door’s status. When the door opener is in digital communication mode, the analog light and lock controls are ignored.
> 
> If this is a deal breaker for you, you have two choices:
> 
> Use ratgdo’s dry contact control and reed switches to get the door status
> Buy a digital wall control which is myQ compatible such as the 889LM / 041A7928-3MC.

neheb, 2024-04-14:
hrm that's unfortunate. Guess I'll just install a smart switch.

closed 2024-04-14 by neheb

restauffer, 2024-04-14:
Don’t dismiss Paul’s suggestion of the. 889LM wall panel too quickly . It replaces your current panel and allows both Ratgdo and regular control of the door.  I am happy with mine.

neheb, 2024-04-14:
rather pricey. The big button works just fine. Since I have switches lying around, easier to just use that.

neheb, 2024-04-14:
@restauffer I see there's a cheaper 888LM. I assume that wouldn't work?

doctorkb, 2024-04-15:
It appears the 889lm replaced the 888lm, the latter of which has been discontinued.

I suspect they both use the needed signal type.

doctorkb, 2024-04-15:
Note, however: https://www.cpsc.gov/Recalls/2023/Chamberlain-Group-Recalls-LiftMaster-myQ-Garage-Door-Control-Panels-Due-to-Entrapment-Hazard-Recall-Alert

That may be why it was replaced...

restauffer, 2024-04-15:
> @restauffer I see there's a cheaper 888LM. I assume that wouldn't work?

doctorkb has the correct answer.  The 889lm replaced the 888lm due to problems with capacitors in the 888lm.

neheb, 2024-05-17:
So on recommendation of people here, I bought the 889lm. Now my garage door randomly opens and closes. This thing is pure garbage.

neheb, 2024-05-17:
Oh this is a ratgdo bug. Lovely.