ratgdo/esphome-ratgdo #253: Wall Control Issues (“Press the push bar to activate control see user manual”)

shadhry, 2024-04-16:
Hello, I’ve installed my ratgdo on a garage door (LiftMaster) and I’ve since had issues with my wall control. Almost every time I use the ratgdo to control the garage door, I get a message that says “press the push bar to activate control” and “see user manual” on my wall controls.

[image]

I am wiring it through the board (meaning it’s bypassed I think), and I’ve had no issues with the wall panel prior to the installation. I also have no issues when I use the wall control panel itself to toggle the garage door state. It reconnects after a quick click, but it is frustrating to have to reconnect it every time the ratgdo runs and loosing the ability to use the motion sensor in automations until reconnected. 

I will say, I have noticed that it looks like every few seconds while the garage door is moving per a ratgdo request, the wall control panel goes dark for a moment and seems to reboot as the display stays blank for a few moments until the error comes up. I realize this may be a low voltage issue, but I’m unsure how I could fix that since it was working prior and all the wires seem snug.

Other than this it’s been so amazing to have and I’m very happy with it.

rlowens, 2024-04-16:
Duplicate of https://github.com/ratgdo/esphome-ratgdo/issues/140

Also:
> I am wiring it through the board (meaning it’s bypassed I think)
It is not bypassed, the ratgdo's connections are just passthroughs and are electrically the same. You are always connecting the ratgdo in parallel with the wall panel.

shadhry, 2024-04-16:
Sorry this is my first post so I assumed since all other threads are marked as closed despite not having a solution I might try bringing it up again. 

Also now that I’m thinking about it, not sure why I thought that it was bypassed that doesn't really make sense. I think initially I meant to imply its passing through the board just chose the wrong wording.

doctorkb, 2024-04-16:
Before I got my ratgdo, my GDO had developed some intermittent issues with the button and other opener device.  Turned out to have just been a poor connection at the GDO.

I'd suggest re-stripping the wires to the correct length and ensuring they're straight when inserted.  Even a previously-twisted solid-core copper wire had issues in those crappy terminals that Chamberlain uses on their GDO.

greggitter, 2024-04-17:
Same problem, with two openers and two different model wall controls.  Never occurred until about a week or two after the RATGDO was installed on each.  I searched on this a while ago and found a comment that sounded like the controls get de-sync'd from the openers.  I left it at that and just press the button when it says to (on the other control without a display, the amber light flashes).  Mildly annoying but for all the benefits the RATGDO brings, I can live with it.

hacsjalano, 2024-04-18:
I've had this exact same issue, using both the ESPHome as well as the MQTT official RATGDO firmware.

Unfortunately the conclusion I've reached is that the RATGDO occasionally pulls the voltages on the communication line so low that that wall control loses power so it basically is getting the equivalent of being unplugged and plugged in again, resetting itself and asking to be re-synced.

I don't think there is anything that can be done about this; it's a design flaw/feature that the wall opener uses the communication lines for power.

For what it's worth, the ESPHome firmware seems to get itself into situations where it resets my wall panel quite often; the offiical MQTT firmware only does this whenever it can't reach the MQTT server, which for me is much less often, pretty much only whenever I upgrade Home Assistant or reboot the server that my MQTT server is running on.

PaulWieland, 2024-04-18:
> Unfortunately the conclusion I've reached is that the RATGDO occasionally pulls the voltages on the communication line so low that that wall control loses power so it basically is getting the equivalent of being unplugged and plugged in again.

So ratgdo only pulls the line low when sending commands to the door opener. This is also the same thing that happens when the GDO sends updates on its current position/status. The wall panel losing power and rebooting points to the internal capacitors failing, and so there isn't enough power to keep the processor running.

Capacitor failures are a running theme with the OEM wall controls. These "smart" wall panels that have backlit LCD screens, motion sensors, temp sensors, etc consume more power and are more likely to suffer from brown outs due to aging capacitors when compared with the simpler "dumb" wall panels which don't have a screen.

PaulWieland, 2024-04-18:
> Can these capacitors be replaced? Or do I need to buy a new wall unit?

Anything is possible if you have the tools and skill to do it.

hacsjalano, 2024-04-18:
I've ordered some capacitors, but I do think this issue should be added the a Frequently Asked Questions section of your website because while it is *not* necessarily the fault of ratgdo, it is very common that many of have garage door openers that are more than a few years old and so we will have wall units that have older capacitors and will encounter this issue.

I think it's important to note this because although the capacitors are too old to handle the extra communication the ratgdo introduces, they are often not too old to function in the original manner that the manufacturer intended which is why people are posting questions thinking that something is wrong, since their garage door wall control panels seem to work just fine when the ratgdo is not connected or doing anything in particular.

It's just a suggestion so you don't have to keep closing tickets here on github.com.  Or at least, if you have this on your FAQ you can post a link to it directly before you close the ticket.

greggitter, 2024-04-18:
@hacsjalano If/when you're successful....would love to have some details on the how-to.

Thanks!

hacsjalano, 2024-04-23:
So just a follow up  - I ended up buying a new 880LMW to replace my 880LM, just in case I botched the capacitor replacement job.  With the new 880LMW, which has a manufacture date of 2 months ago, so it's a new device with fresh capacitors, the problem happens less often but it still happens.

In the past I would see this issue whenever my WiFi router was rebooted or the Home Assistant server was rebooted.  In the past this would happen almost every time, now it's about 1 in every 4 times.  I was doing some software updates these last 2 days and so I got a chance to test this out a lot more.

I'm keeping the new 880LMW because my previous 880LM's screen was getting a little faded but at this point I'm not sure getting a new wall module if this issue bothers you.  Perhaps replacing the capacitors with something beefier than what was originally on the circuit board might work, but I'm not an electrical engineer so I don't know how to safely measure how much bigger you could go without messing something up.

greggitter, 2024-04-23:
Appreciate the follow-up.  My opener which is years old...probably about 10 now is not occurring 1 in 4 times...it's actually very rare, maybe once per 1 - 2 or even 3 weeks.  I'll have to watch for a pattern as to when it happens.  Cheers.

shadhry, 2024-05-15:
Thank you all for your support! Here's my update so far: I've re-stripped the wires to essentially the EXACT length as depicted on the instructions on the GDO. I reinserted the freshly cut wires and unfortunately, the issue persists. I do have some observations that might help narrow this down. It doesn't normally happen upon the garage door opening for some reason, however almost every time the garage door is closed the wall panel can be seen rebooting once the garage door has stopped moving. Im not sure why it's at that specific point in time but that's all I've really gathered so far.

inittabayanami, 2024-11-14:
Having what seems like the same issue with 3 Chamberlain B2211TMC openers, each with a ratgdo v2.5 kit and 041A7367-3 wall controls (open/close button, light/learn button, no display, security 2.0 (yellow learn button)). 

All openers were purchased new along with ratgdo's. See issue #[339](https://github.com/ratgdo/esphome-ratgdo/issues/339). I will usually have to press the wall control twice to get a response/door movement with the ratgdo's installed, all 3 doors are experiencing the same issue. wall controls do have a yellow light on them but they have been a solid light with no blinking. 

This issue usually happens after awhile, if I stand there and attempt to replicate the issue by using home assistant to open/close the door then hit the wall opener it appears to work every time, but if I go to walk out of the house in the morning and try to use the wall opener I will see this issue, or when the wife comes home, uses her car control to open the door, then attempts to use the wall control to close the door.

inittabayanami, 2024-11-24:
> A couple weeks ago I replaced my smart wall control panel with a newer one manufactured a few months ago - from Chamberlain. Same model. I've yet to have the wall control go dark and require a reboot so far. Before this it would happen daily. I'm going to take the old smart control and replace the two decoupling caps with maybe double the ratings and give it a test.

Do you have the model # for the wall control panel you're using? 

I was also thinking if there was a way we could use one of the sets of contact sensors as a button trigger instead that would probably solve a lot of this problem as long as you don't require your button being smart. pressing a simple normally open opener button would trigger the ratgdo to open the door.

jdranchman, 2024-11-25:
> Do you have the model # for the wall control panel you're using?
>
> I was also thinking if there was a way we could use one of the sets of contact sensors as a button trigger instead that would [...]

https://www.chamberlain.com/wall-control-panel/p/041A7305-1

jdranchman, 2024-11-25:
>

A better, simpler approach would be to convert from the too smart wall panel to the security 2.0 single button opener without light, motion and smarts then it wouldn't matter anymore. You would have to roll your own motion/presence sensor and light. I wanted to maybe try using the light dry contact with a shelly with my own motion sensor... Maybe...

inittabayanami, 2024-11-25:
> A better, simpler approach would be to convert from the too smart wall panel to the security 2.0 single button opener without light, motion and smarts then it wouldn't matter anymore. You would have [...]

https://www.chamberlain.com/push-button-door-control/p/041A7367-
This one? Rhats what I currently have and having the same desync issue where i have to press twice to get it to sync and actually close/open the door

jdranchman, 2024-11-25:
@inittabayanami The model number hasn't changed only the manufacturing date. I'm going assume that the capacitors haven't aged well and that most people don't have some type of UPS or surge suppression on their garage door systems. Getting a brand new wall controller isn't cheap but it may fix it for you - for a while anyway. The electrolytic caps on this board are liquid/aluminum and are prone to degrade over time due to several issues - the most damaging is just drying out and slowly stop working

inittabayanami, 2024-11-25:
> @inittabayanami The model number hasn't changed only the manufacturing date. I'm going assume that the capacitors haven't aged well and that most people don't have some type of UPS or surge [...]

Yeah, that shouldnt be an issue in at least my case, i have the more dumbed down buttons and had this issue out the gate with three brand new installed openers

dbavatar, 2024-11-25:
> @inittabayanami The model number hasn't changed only the manufacturing date. I'm going assume that the capacitors haven't aged well and that most people don't have some type of UPS or surge [...]

I have a chamberlain opener with internal lead-acid backup, so the wall panels continue to work with no power (although they seem to know to turn off their backlight on battery power). Went from and old panel with the issue to a new panel which doesn't show the error message, but mostly still doesn't work on first button press. Meanwhile I have another (old) panel on the same system that's never shown an issue.

BTW another strange observance - I hear "coil whine" from the "bad" panel which shows no error, when the ratgdo is talking.

I just assume design tolerances are aligning badly.

HeroicMercenary, 2024-11-27:
> A couple weeks ago I replaced my smart wall control panel with a newer one manufactured a few months ago - from Chamberlain. Same model. I've yet to have the wall control go dark and require a [...]

Please keep us posted on whether this does anything.  I took my two caps off the board, tested them with an ESR meter, and both tested good even though I get this happening VERY often.  By "doubling the ratings", I'm assuming you're referring to capacitance.

jdranchman, 2024-12-01:
It's a bit early to be a diffinative solution but after 3 days with the old wall controller and new double rated caps the wall control unit hasn't gone into a rebootme event. This unit was made in 2018 and used to get stuck once or twice a day or more depending if I noticed it. This was a very tight fit because the new caps are obviously bigger. I don't recommend doing this if you aren't a good electronics tech with a well equipped bench.

HeroicMercenary, 2024-12-02:
> It's a bit early to be a diffinative solution but after 3 days with the old wall controller and new double rated caps the wall control unit hasn't gone into a rebootme event. This unit was made in [...]

Appreciate the update either way! Do you happen to have the specs for the caps you installed?

jdranchman, 2024-12-02:
I pulled the replacements out of my parts bin - I'm sure the originals were 220uf 25v and 100uf 50v and changed to 470/25 and 220/50. The difficult part of this task is reassembly of the PCB into the case as there is not much space to fold over the radial mounts and make it fit back in place.

hartmms, 2025-03-02:
I have brand new 885LM wall buttons.  These have no fancy features except a LED that stays on all the time.  I too have noticed it will take 2 presses to make the button work.  I've also noticed the LED will turn of even when I'm not pressing the button. It seems to be the same issue of the wall panel micro controller loosing power when ratgdo is communicating on the wire. 

I'm not sure I'll be able to replace the capacitors. They are such small SMD, I'm not sure I can measure their C or replace them with larger ones.  Maybe I can piggy back new ones on top of old to double up the capacitance. 

I had thought perhaps I can just drop a capacitor between the screw terminals, but I suspect that would not allow the ratgdo to communcate properly.  I might have to trace the circuit board manually to see how the micro controller gets it's power. 

@PaulWieland - do you have any electrical details you can share of the "one wire" signaling that's used?

hartmms, 2025-03-22:
I had a 1000uF electrolytic cap in my parts bin and soldered that in parallel to the existing one. I have verified the cap is electrically connected as I can disconnect the wires and the LDE will stay lit for another second or 2.   Also, I can measure 11.7v on the cap leads.  

Unfortunately it did not fix the issue. I still have to push the wall button twice to get it to register with the opener again and operate the door.

jdranchman, 2025-03-22:
I ended up swapping in the control panel without the LCD. https://www.chamberlain.com/motion-detecting-wall-control-panel/p/041A7327-1

I think I can live without the display. So far I haven't had any issues. I also upgraded to the 32 disco because of lasers!

greggitter, 2025-03-23:
> I ended up swapping in the control panel without the LCD. https://www.chamberlain.com/motion-detecting-wall-control-panel/p/041A7327-1
> 
> I think I can live without the display. So far I haven't had any issues. I also upgraded to the 32 disco because of lasers!

Just for the record, I have that model as well as the digital one and they both do it.  Not frequently, but both do it.  The more I use the remote ability to operate the light, the more often it happens.  I don't know what the trigger is but I'm guessing (at this point) it's not a hardware issue.

PaulWieland, 2025-07-07:
Duplicate of #140 

The consensus is that this issue is caused by a bad wall control. I cannot reproduce this with any of the wall controls that I have for testing, so this is really an educated guess.

When the capacitors fail (due to age, heat, etc), they slowly lose their ability to hold a charge, or enough charge to keep the wall control's microprocessor running while data is being transmitted over the power line. Essentially, the wall control is experiencing a brownout which causes it to reboot.

closed 2025-07-07 by PaulWieland

tfcarlin, 2025-08-08:
To control the doors from a different location I implemented an ESPHOME switch device using these:

https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html 
https://www.amazon.com/dp/B0B8QL8J69?th=1
https://www.amazon.com/LeMotech-Plastic-Electrical-Project-Junction/dp/B07W6SJGJM?th=1

Not as simple as a physical switch but works with HA seamlessly. I soldered the wires to the board which is taped down in the box. The button is passed through the cover.

jdranchman, 2025-08-08:
I'm pretty sure you can just replace the intelligent S2 wall control with a
simple push button switch. The only thing wrong with that approach for me
is I really like the motion control that comes in the smart S2 switches.
I'm prob going to backdoor that through HA though with much better PIR
sensor (external Hue maybe).

Jim Davis

PaulWieland, 2025-08-08:
> It would also be nice if we could use one of the inputs on the ratgdo for a simple contact switch and use dumb butons instead.

You can. That’s what the trigger open, trigger close inputs are for.

hartmms, 2026-03-08:
I bought a new 882LMW from Amazon hoping a newer version would not have this issue, like @HeroicMercenary experienced.  Unfortunately, it didn't fix the problem.  There must be something with the firmware of the 882LMW that just doesn't like the ratgdo on the circuit.