ratgdo/esphome-ratgdo #254: Critical - door randomly opened randomly

4vanger, 2024-04-16:
I came outside yesterday and to my surprise garage door was fully opened but nobody at home did that. I freshly reflashed ratgdo just few days ago with esphome because mqtt was becoming unresponsive. Because I freshly updated it I didn't had a chance to update my automations so I don't think this is one of them opening the door.
esphome by ratgdo
Firmware: 2024.3.2 (Apr 8 2024, 04:18:40)
Hardware: 2.5i
MAC: AC:0B:FB:E4:38:A1
Home Assistant 
[image]

PaulWieland, 2024-07-12:
Please update to latest release

closed 2024-07-12 by PaulWieland

mikecarr, 2026-03-17:
I have been having this issue lately, I downloaded latest from GitHub and it is part of esphome.