ratgdo/esphome-ratgdo #256: ESPHome 2024.4.0 build fails

doctorkb, 2024-04-17:
I've tried cleaning build files and going again, but I'm still getting this when trying to build ratgdo under ESPHome 2024.4.0:
```
Linking .pioenvs/ratgdov25i-a855e0/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/ratgdov25i-a855e0/src/esphome/components/ratgdo/secplus2.cpp.o:(.rodata._ZTVN7esphome6ratgdo8secplus28Secplus2E[_ZTVN7esphome6ratgdo8secplus28Secplus2E]+0x18): undefined reference to `_ZN7esphome6ratgdo8protocol8Protocol14set_open_limitEb'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/ratgdov25i-a855e0/src/esphome/components/ratgdo/secplus2.cpp.o:(.rodata._ZTVN7esphome6ratgdo8secplus28Secplus2E[_ZTVN7esphome6ratgdo8secplus28Secplus2E]+0x1c): undefined reference to `_ZN7esphome6ratgdo8protocol8Protocol15set_close_limitEb'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/ratgdov25i-a855e0/src/esphome/components/ratgdo/secplus2.cpp.o:(.rodata._ZTVN7esphome6ratgdo8secplus28Secplus2E[_ZTVN7esphome6ratgdo8secplus28Secplus2E]+0x20): undefined reference to `_ZN7esphome6ratgdo8protocol8Protocol21set_discrete_open_pinEPNS_15InternalGPIOPinE'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/ratgdov25i-a855e0/src/esphome/components/ratgdo/secplus2.cpp.o:(.rodata._ZTVN7esphome6ratgdo8secplus28Secplus2E[_ZTVN7esphome6ratgdo8secplus28Secplus2E]+0x24): undefined reference to `_ZN7esphome6ratgdo8protocol8Protocol22set_discrete_close_pinEPNS_15InternalGPIOPinE'
collect2: error: ld returned 1 exit status
*** [.pioenvs/ratgdov25i-a855e0/firmware.elf] Error 1
```

I have it working correctly under 2024.2.2, so this isn't the end of the world for me, but it might affect new users.

doctorkb, 2024-04-17:
Appears to be resolved by #257 .

closed 2024-04-17 by doctorkb

ccutrer, 2024-04-21:
This is a problem again.