ratgdo/esphome-ratgdo #260: Both ratgdo devices offline after updating to 2024.4.0 of ESPHome

gokou340, 2024-04-17:
I ran the update for 2024.4.0 on my Ratgdo devices, and both of them show offline. I can ping them, but they don't do anything. I can run the install again and validate, but it still doesn't bring them online.

```INFO ESPHome 2024.4.0
INFO Reading configuration /config/esphome/maingarage-ca04cd.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
Validation
INFO Detected timezone 'America/Denver'
INFO Generating C++ source...
INFO Compiling app...
Processing maingarage-ca04cd (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 6.18.5
|-- Secplus @ 1.0.0+sha.f98c322
|-- EspSoftwareSerial @ 8.1.0+sha.57f1aa7
|-- noise-c @ 0.1.4
|-- Improv @ 1.2.3
RAM:   [=====     ]  46.2% (used 37880 bytes from 81920 bytes)
Flash: [======    ]  60.4% (used 630713 bytes from 1044464 bytes)
========================= [SUCCESS] Took 4.10 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 10.9.201.48
INFO Uploading /data/build/maingarage-ca04cd/.pioenvs/maingarage-ca04cd/firmware.bin (634864 bytes)
INFO Compressed to 424269 bytes
Uploading: [============================================================] 100% Done...

INFO Upload took 5.84 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 10.9.201.48 using esphome API
WARNING Can't connect to ESPHome API for maingarage-ca04cd @ 10.9.201.48: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.9.201.48', port=6053))]: [Errno 111] Connect call failed ('10.9.201.48', 6053) (SocketAPIError)
INFO Trying to connect to maingarage-ca04cd @ 10.9.201.48 in the background

gokou340, 2024-04-17:
Restored from a Veeam Backup and was able to reinstall the YAML on 2024.3.2. Both Ratgdo's are up and running again.

edrikk, 2024-04-23:
@gokou340 I see 4.1 was released with this specfic fix:  https://github.com/esphome/esphome/pull/6598
May be why you experienced the issue so thought to share (I've stayed at .3.2 until a confirmation that .4.X is working again).

Worth a try...

gokou340, 2024-04-24:
> @gokou340 I see 4.1 was released with this specfic fix: [esphome/esphome#6598](https://github.com/esphome/esphome/pull/6598) May be why you experienced the issue so thought to share (I've stayed at .3.2 until a confirmation that .4.X is working again).
> 
> Worth a try...

I will take a snapshot of my Home Assistant instance and test this new release out and report back.

gokou340, 2024-04-24:
> @gokou340 I see 4.1 was released with this specfic fix: [esphome/esphome#6598](https://github.com/esphome/esphome/pull/6598) May be why you experienced the issue so thought to share (I've stayed at .3.2 until a confirmation that .4.X is working again).
> 
> Worth a try...
@edrikk I can confirm that 4,1 did indeed resolve the issue and I was able to update my Home Assistant and my Ratgdo devices and they reconnected without issue.

Thanks for heads up!

closed 2024-04-24 by gokou340