ratgdo/esphome-ratgdo #263: Failure to Compile with ESP-IDF Framework

ccutrer, 2024-04-22:
When I have the following snippet in my config (which I use to enable connecting to MQTT over TLS on my other ESPHome devices):

```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
```

compilation fails with the following:

```
Compiling .pioenvs/john-deere-gdo/src/esphome/components/ratgdo/dry_contact.o
In file included from src/esphome/components/ratgdo/dry_contact.cpp:2:
src/esphome/components/ratgdo/dry_contact.h:3:10: fatal error: SoftwareSerial.h: No such file or directory

************************************************************************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SoftwareSerial.h"
* Web  > https://registry.platformio.org/search?q=header:SoftwareSerial.h
*
************************************************************************

 #include "SoftwareSerial.h" // Using espsoftwareserial https://github.com/plerup/espsoftwareserial
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/john-deere-gdo/src/esphome/components/ratgdo/dry_contact.o] Error 1
src/esphome/components/ratgdo/cover/ratgdo_cover.cpp: In member function 'void esphome::ratgdo::RATGDOCover::on_door_state(esphome::ratgdo::DoorState, float)':
src/esphome/components/ratgdo/cover/ratgdo_cover.cpp:52:28: warning: this statement may fall through [-Wimplicit-fallthrough=]
             this->position = position;
             ~~~~~~~~~~~~~~~^~~~~~~~~~
src/esphome/components/ratgdo/cover/ratgdo_cover.cpp:53:9: note: here
         case DoorState::UNKNOWN:
         ^~~~
```

ccutrer, 2024-04-22:
I was able to get it to compile (for ESP32/esp-idf/secplus2) with https://github.com/ccutrer/esphome-ratgdo/commit/778b6c9dc173f76f9552007218bf3697d4e7f126, but communication with the GDO failed, so I just reverted to Arduino framework without TLS on these devices for now.

zeel01, 2026-01-30:
Similar situation, I found this project today: https://github.com/francescopace/espectre, and wanted to see if I could get this component to run on my ratgdo to provide motion detection in my garage.

However it requires esp-idf as the framework, and if I choose that ratgdo fails to compile with that same error:
```
In file included from src/esphome/components/ratgdo/ratgdo.cpp:22:
src/esphome/components/ratgdo/secplus1.h:7:10: fatal error: SoftwareSerial.h: No such file or directory

******************************************************************************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SoftwareSerial.h"
* Web  > https://registry.platformio.org/search?q=header:%1B%5Bm%1B%5BKSoftwareSerial.h
*
******************************************************************************

    7 | #include "SoftwareSerial.h" // Using espsoftwareserial https://github.com/plerup/espsoftwareserial
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```
Is this the only thing that gets in the way of IDF support? It would be awesome if that could be remedied.